Xilinx/Vitis-AI #249: Error building docker GPU image in Vitis-AI-master

YOKO-XIA, 2021-01-08:
Hello!

I was trying to use Vitis-AI-master.
But I'm faced with problem to build docker image of GPU.
The error message is below:

 ---> e3a5ce28a3e4
Step 22/45 : RUN cd /tmp; wget --progress=dot:mega https://codeload.github.com/google/protobuf/zip/v3.4.0     && unzip v3.4.0     && cd protobuf-3.4.0     && ./autogen.sh     && ./configure     && make -j     && make install     && ldconfig     && rm -fr /tmp/*
 ---> Running in e39cf49c90aa
--2021-01-08 10:42:54--  https://codeload.github.com/google/protobuf/zip/v3.4.0
Resolving codeload.github.com (codeload.github.com)... 13.250.162.133
Connecting to codeload.github.com (codeload.github.com)|13.250.162.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [application/zip]
Saving to: ‘v3.4.0’

     0K ........ ........ ........ ........ ........ ........ 8.56K
  3072K ........ ........ ......                              4.91K=10m55s

2021-01-08 10:53:50 (6.91 KB/s) - ‘v3.4.0’ saved [4631826]

Archive:  v3.4.0
  End-of-central-directory signature not found.  Either this file is not
  a zipfile, or it constitutes one disk of a multi-part archive.  In the
  latter case the central directory and zipfile comment will be found on
  the last disk(s) of this archive.
unzip:  cannot find zipfile directory in one of v3.4.0 or
        v3.4.0.zip, and cannot find v3.4.0.ZIP, period.
The command '/bin/bash -c cd /tmp; wget --progress=dot:mega https://codeload.github.com/google/protobuf/zip/v3.4.0     && unzip v3.4.0     && cd protobuf-3.4.0     && ./autogen.sh     && ./configure     && make -j     && make install     && ldconfig     && rm -fr /tmp/*' returned a non-zero code: 9

How can I fix it?

hanxue, 2021-01-13:
This seems to be a temporary network problem. I have just tested the following command to build protobuf 3.4.0 and it runs fine

```
cd /tmp; wget --progress=dot:mega https://codeload.github.com/google/protobuf/zip/v3.4.0 && unzip v3.4.0 && cd protobuf-3.4.0 && ./autogen.sh && ./configure && make -j 
```

Could you please try building the GPU docker image again?

qianglin-xlnx, 2022-02-20:
Hi @YOKO-XIA 
Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-02-20 by qianglin-xlnx