Xilinx/Vitis-AI #253: Looking for a detection model avarable with Alveo-U250

Hikaru-Furuta, 2021-01-13:
I looked for a object-detection model, not a classification model, for Alveo-U250 at https://github.com/Xilinx/Vitis-AI/tree/master/models/AI-Model-Zoo/model-list only to find trese's not.
If anyone know a possible detection-model on Alveo-U250, please tell me about it.

sumitn-xilinx, 2021-01-13:
Check yolo and face_detect in following directory:

https://github.com/Xilinx/Vitis-AI/tree/master/examples/DPUCADX8G

Hikaru-Furuta, 2021-01-13:
@sumitn-xilinx
Any other way?
I've never implemented with caffe
Do you know any ways out with Tensorflow or Pytorch?

jimheaton, 2021-01-13:
You could use darknet to train a yolo model, and then use the flow described in https://github.com/Xilinx/Vitis-Tutorials/blob/master/Machine_Learning/Design_Tutorials/07-yolov4-tutorial/README.md to convert the model to keras or use the caffe flow. 

This tutorial covers Yolov4, but will also work for Yolov3 and Tiny Yolov3.

This tutorial was developed for Vitis-A1.2, and  Vitis-1.3 now has TensorFlow2 support. If convert the darknet model to keras, then you can now directly bring the model .h5 into Vitis-AI and do not need to first convert to a TensorFlow.pb file as covered in the tutorial.

Hikaru-Furuta, 2021-01-13:
I'll try, thanks

Hikaru-Furuta, 2021-01-13:
@jimheaton
As is written "The following tutorials cover how to train, evaluate, convert, quantize, compile, and deploy Yolov4 on the Xilinx ZCU102 and ZCU104 evaluation boards." in tutorials, Alveo-U250 may not be supported.

Hikaru-Furuta, 2021-01-20:
I've decided to implement with Caffe. Thank you!

closed 2021-01-20 by Hikaru-Furuta

Hikaru-Furuta, 2021-02-12:
Package                Version
---------------------- -------------------
absl-py                0.11.0
aiohttp                3.6.3
appdirs                1.4.4
argon2-cffi            20.1.0
arrow                  0.17.0
astunparse             1.6.3
async-generator        1.10
async-timeout          3.0.1
attrs                  20.3.0
backcall               0.2.0
bleach                 3.2.1
blinker                1.4
brotlipy               0.7.0
cachetools             4.1.1
certifi                2020.11.8
cffi                   1.14.0
chardet                3.0.4
click                  7.1.2
cpools                 0.0.0
cryptography           3.2.1
cycler                 0.10.0
Cython                 3.0a6
dataclasses            0.8
decorator              4.4.2
defusedxml             0.6.0
docopt                 0.6.2
easydict               1.9
entrypoints            0.3
gast                   0.3.3
google-auth            1.23.0
google-auth-oauthlib   0.4.2
google-pasta           0.2.0
graphviz               0.14.2
grpcio                 1.31.0
h5py                   2.10.0
idna                   2.10
idna-ssl               1.1.0
importlib-metadata     2.0.0
inform                 1.23.0
iniconfig              1.1.1
ipykernel              5.4.2
ipython                7.16.1
ipython-genutils       0.2.0
ipywidgets             7.5.1
jedi                   0.17.2
Jinja2                 2.11.2
jsonschema             3.2.0
jupyter                1.0.0
jupyter-client         6.1.7
jupyter-console        6.2.0
jupyter-core           4.7.0
jupyterlab-pygments    0.1.2
Keras-Preprocessing    1.1.0
kiwisolver             1.3.1
llvmlite               0.34.0
Markdown               3.3.3
MarkupSafe             1.1.1
matplotlib             3.3.3
mistune                0.8.4
mkl-fft                1.2.0
mkl-random             1.1.1
mkl-service            2.3.0
multidict              4.7.6
nbclient               0.5.1
nbconvert              6.0.7
nbformat               5.0.8
nest-asyncio           1.4.3
nngen                  1.3.0
notebook               6.1.5
numba                  0.51.2
numpy                  1.19.2
oauthlib               3.1.0
olefile                0.46
onnx                   1.8.0
onnx-simplifier        0.2.19
onnxoptimizer          0.1.1
onnxruntime            1.6.0
opencv-python          4.4.0.46
opt-einsum             3.1.0
packaging              20.8
pandocfilters          1.4.3
parso                  0.7.1
pexpect                4.8.0
pickleshare            0.7.5
Pillow                 8.0.1
pip                    20.2.4
pluggy                 0.13.1
ply                    3.11
progress               1.5
progressbar            2.5
prometheus-client      0.9.0
prompt-toolkit         3.0.8
protobuf               3.13.0
ptyprocess             0.6.0
py                     1.10.0
pyasn1                 0.4.8
pyasn1-modules         0.2.8
pycocotools            2.0
pycparser              2.20
Pygments               2.7.3
PyJWT                  1.7.1
pyOpenSSL              19.1.0
pyparsing              3.0.0b1
pyrsistent             0.17.3
PySocks                1.7.1
pystan                 2.19.1.1
pytest                 6.2.1
pytest-pythonpath      0.7.3
python-dateutil        2.8.1
pyverilog              1.2.1
pyxir                  0.0.1
pyzmq                  20.0.0
qtconsole              5.0.1
QtPy                   1.9.0
quantiphy              2.13.0
requests               2.24.0
requests-oauthlib      1.3.0
rsa                    4.6
scipy                  1.5.4
Send2Trash             1.5.0
setuptools             50.3.1.post20201107
six                    1.15.0
tensorboard            2.3.0
tensorboard-plugin-wit 1.6.0
tensorflow             2.2.0
tensorflow-estimator   2.2.0
termcolor              1.1.0
terminado              0.9.1
testpath               0.4.4
tf2onnx                1.7.2
toml                   0.10.2
torch                  1.4.0
torchvision            0.5.0
tornado                6.1
tqdm                   4.56.0
traitlets              4.3.3
tvm                    1.0.0
typing-extensions      3.7.4.3
urllib3                1.25.11
veriloggen             1.8.2
wcwidth                0.2.5
webencodings           0.5.1
Werkzeug               1.0.1
wheel                  0.35.1
widgetsnbextension     3.5.1
wrapt                  1.12.1
yarl                   1.6.2
zipp                   3.4.0

quandangk54, 2021-03-17:
> You could use darknet to train a yolo model, and then use the flow described in https://github.com/Xilinx/Vitis-Tutorials/blob/master/Machine_Learning/Design_Tutorials/07-yolov4-tutorial/README.md [...]

quandangk54, 2021-03-17:
Hi @jimheaton,

This tutorial will also work with Tiny yolov4 or not?
https://github.com/Xilinx/Vitis-Tutorials/blob/master/Machine_Learning/Design_Tutorials/07-yolov4-tutorial/README.md

jimheaton, 2021-03-17:
@quandangk54 
Have not tried the tutorial for the U250, only the U50 and ZCU102 ZCU104.

The  VIA tools do not currently have support for the strided slice layer used in Tiny yolov4. You would have to change that to something else first, or it will be implemented as  a separate cpu kernel.

Other than that the tutorial should work for Tiny yolov4.
The tutorial should also work for Tiny Yolov3. If you use the Darknet to Caffe flow for this, you will first have to comment out the max_pooling layer with stride 1 to be compatible with Caffe.

jimheaton, 2021-03-17:
@quandangk54
In order to avoid the srtrided slice issue, change all the route_layers with a group_id to a 1x1 conv. I believe there are 2 of these in the yolov4-tiny.cf file.

Also recommend that you use the conversion from Darknet to TensorFlow instead of Caffe. There seems to be some issues with the Caffe conversion for tiny yolov4.