Xilinx/Vitis-AI #254: ERROR Compiling XIR Source Code of the VART

HSqure, 2021-01-14:
Hi, here's an ERROR while compiling xir of the VART. Does someone have an idea? Need help.

### Operation describe

**1.** Entering sdk environment: `source opt/petelinux/2020.2/petalinux_sdk/environment-setup-aarch64-xilinx-linux`

**2.** According to [https://github.com/Xilinx/Vitis-AI/blob/master/tools/Vitis-AI-Runtime/VART/quick_start_for_edge.md](url) , I finished setting up the host. 

**3.** According to [https://github.com/Xilinx/Vitis-AI/tree/v1.1/Vitis-AI-Library#setting-up-the-host](url) to set up the `glog-0.4.0`.

**4.** start to compile the VART

- `unilog`  
Running `./cmake.sh` completed, smoothing.

- `xir`
After running `./cmake.sh`, ERROR showing as the blow ERROR log.
 
**e.g. Whatever I do `step1` or not (cross-compile for the target or compile for the host), it still gets the same situation and error.**

### ERROR log
```
 xxx@ubuntu:~/Vitis-ai-suit-v1.3/Vitis-AI-master/tools/Vitis-AI-Runtime/VART/xir$ ./cmake.sh 
Cross-platform building...
Found target sysroot /opt/petalinux/2020.2/petalinux_sdk/sysroots/aarch64-xilinx-linux
cd /home/xxx/build/build.linux.2020.2.aarch64.Debug/xir
cmake -DBUILD_TEST=ON -DBUILD_CONTRIB=OFF -DBUILD_DOC=OFF -DCMAKE_BUILD_TYPE=Debug -DCMAKE_EXPORT_COMPILE_COMMANDS=ON -DBUILD_PYTHON=OFF -DINSTALL_HOME=OFF -DINSTALL_USER=OFF -DCMAKE_TOOLCHAIN_FILE=/opt/petalinux/2020.2/petalinux_sdk/sysroots/x86_64-petalinux-linux/usr/share/cmake/OEToolchainConfig.cmake -DCMAKE_PREFIX_PATH=/install/Debug -DCMAKE_INSTALL_PREFIX=/opt/petalinux/2020.2/petalinux_sdk/sysroots/aarch64-xilinx-linux/install/Debug /home/xxx/Vitis-ai-suit-v1.3/Vitis-AI-master/tools/Vitis-AI-Runtime/VART/xir
-- The C compiler identification is GNU 9.2.0
-- The CXX compiler identification is GNU 9.2.0
-- Check for working C compiler: /opt/petalinux/2020.2/petalinux_sdk/sysroots/x86_64-petalinux-linux/usr/bin/aarch64-xilinx-linux/aarch64-xilinx-linux-gcc
-- Check for working C compiler: /opt/petalinux/2020.2/petalinux_sdk/sysroots/x86_64-petalinux-linux/usr/bin/aarch64-xilinx-linux/aarch64-xilinx-linux-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /opt/petalinux/2020.2/petalinux_sdk/sysroots/x86_64-petalinux-linux/usr/bin/aarch64-xilinx-linux/aarch64-xilinx-linux-g++
-- Check for working CXX compiler: /opt/petalinux/2020.2/petalinux_sdk/sysroots/x86_64-petalinux-linux/usr/bin/aarch64-xilinx-linux/aarch64-xilinx-linux-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found importable target unilog::unilog: /opt/petalinux/2020.2/petalinux_sdk/sysroots/aarch64-xilinx-linux/install/Debug/lib/libunilog.so.1.3.0
-- Found OpenSSL: /opt/petalinux/2020.2/petalinux_sdk/sysroots/aarch64-xilinx-linux/usr/lib/libcrypto.so (found version "1.1.1d")  
-- Found Boost: /opt/petalinux/2020.2/petalinux_sdk/sysroots/aarch64-xilinx-linux/usr/lib/cmake/Boost-1.71.0/BoostConfig.cmake (found version "1.71.0")  
-- Found Protobuf: /opt/petalinux/2020.2/petalinux_sdk/sysroots/aarch64-xilinx-linux/usr/lib/libprotobuf.so;-lpthread (found version "3.9.2") 
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
-- Configuring done
-- Generating done
-- Build files have been written to: /home/xxx/build/build.linux.2020.2.aarch64.Debug/xir
[  3%] Running cpp protocol buffer compiler on /home/xxx/Vitis-ai-suit-v1.3/Vitis-AI-master/tools/Vitis-AI-Runtime/VART/xir/src/xir/proto/graph_proto_v2.proto
[  3%] Running cpp protocol buffer compiler on /home/xxx/Vitis-ai-suit-v1.3/Vitis-AI-master/tools/Vitis-AI-Runtime/VART/xir/src/xir/proto/graph_proto_v1.proto
Scanning dependencies of target xir
[  5%] Building CXX object src/xir/CMakeFiles/xir.dir/graph_proto_v1.pb.cc.o
[  6%] Building CXX object src/xir/CMakeFiles/xir.dir/util/data_type.cpp.o
[  8%] Building CXX object src/xir/CMakeFiles/xir.dir/graph_proto_v2.pb.cc.o
[ 10%] Building CXX object src/xir/CMakeFiles/xir.dir/util/error_code.cpp.o
[ 11%] Building CXX object src/xir/CMakeFiles/xir.dir/util/internal_util.cpp.o
[ 13%] Building CXX object src/xir/CMakeFiles/xir.dir/op/built_in_ops.cpp.o
[ 15%] Building CXX object src/xir/CMakeFiles/xir.dir/attrs/attr_expander_imp.cpp.o
[ 16%] Building CXX object src/xir/CMakeFiles/xir.dir/attrs/attrs.cpp.o
[ 28%] Building CXX object src/xir/CMakeFiles/xir.dir/graph/elf2xir.cpp.o
[ 18%] Building CXX object src/xir/CMakeFiles/xir.dir/op/op_def.cpp.o
[ 30%] Building CXX object src/xir/CMakeFiles/xir.dir/op/op_imp.cpp.o
[ 20%] Building CXX object src/xir/CMakeFiles/xir.dir/attrs/attrs_imp.cpp.o
[ 32%] Building CXX object src/xir/CMakeFiles/xir.dir/graph/graph_template.cpp.o
[ 23%] Building CXX object src/xir/CMakeFiles/xir.dir/attrs/attr_def.cpp.o
[ 33%] Building CXX object src/xir/CMakeFiles/xir.dir/graph/subgraph_imp.cpp.o
[ 23%] Building CXX object src/xir/CMakeFiles/xir.dir/graph/graph_template_imp.cpp.o
[ 25%] Building CXX object src/xir/CMakeFiles/xir.dir/util/tool_function.cpp.o
[ 27%] Building CXX object src/xir/CMakeFiles/xir.dir/tensor/tensor.cpp.o
[ 35%] Building CXX object src/xir/CMakeFiles/xir.dir/graph/graph_imp.cpp.o
[ 37%] Building CXX object src/xir/CMakeFiles/xir.dir/op/shape_inference.cpp.o
[ 38%] Building CXX object src/xir/CMakeFiles/xir.dir/graph/serialize_v2.cpp.o
[ 40%] Building CXX object src/xir/CMakeFiles/xir.dir/graph/graph.cpp.o
[ 42%] Building C object src/xir/CMakeFiles/xir.dir/version.c.o
[ 44%] Building CXX object src/xir/CMakeFiles/xir.dir/op/op_def_factory_imp.cpp.o
[ 45%] Building CXX object src/xir/CMakeFiles/xir.dir/tensor/tensor_imp.cpp.o
/home/xxx/Vitis-ai-suit-v1.3/Vitis-AI-master/tools/Vitis-AI-Runtime/VART/xir/src/xir/util/tool_function.cpp: In function ‘const string xir::get_lib_id()’:
/home/xxx/Vitis-ai-suit-v1.3/Vitis-AI-master/tools/Vitis-AI-Runtime/VART/xir/src/xir/util/tool_function.cpp:71:32: error: ‘PROJECT_GIT_COMMIT_ID’ was not declared in this scope
   71 |   const auto ret = std::string{PROJECT_GIT_COMMIT_ID};
      |                                ^~~~~~~~~~~~~~~~~~~~~
/home/xxx/Vitis-ai-suit-v1.3/Vitis-AI-master/tools/Vitis-AI-Runtime/VART/xir/src/xir/util/tool_function.cpp:71:53: error: no matching function for call to ‘std::__cxx11::basic_string<char>::basic_string(<brace-enclosed initializer list>)’
   71 |   const auto ret = std::string{PROJECT_GIT_COMMIT_ID};
      |                                                     ^

... ...
... ...
... ...

src/xir/CMakeFiles/xir.dir/build.make:143: recipe for target 'src/xir/CMakeFiles/xir.dir/util/tool_function.cpp.o' failed
make[2]: *** [src/xir/CMakeFiles/xir.dir/util/tool_function.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:117: recipe for target 'src/xir/CMakeFiles/xir.dir/all' failed
make[1]: *** [src/xir/CMakeFiles/xir.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2

```

qianglin-xlnx, 2021-01-14:
Hi @HSqure 
I can't reproduce the issue. But it seems that the shell on your system is not bash. Could you try the following command.
`/bin/bash cmake.sh`
BTW, there is no need to do the step3 (glog-0.4.0)  you mentioned.
The following is the xir building log for your reference.

`Cross-platform building...
Found target sysroot /proj/xsjhdstaff6/linqiang/vai1.3/integrate_test_vai1.3/petalinux_sdk/sysroots/aarch64-xilinx-linux
cd /home/linqiang/build/build.linux.2020.2.aarch64.Debug/xir
cmake -DBUILD_TEST=ON -DBUILD_CONTRIB=OFF -DBUILD_DOC=OFF -DCMAKE_BUILD_TYPE=Debug -DCMAKE_EXPORT_COMPILE_COMMANDS=ON -DBUILD_PYTHON=OFF -DINSTALL_HOME=OFF -DINSTALL_USER=OFF -DCMAKE_TOOLCHAIN_FILE=/proj/xsjhdstaff6/linqiang/vai1.3/integrate_test_vai1.3/petalinux_sdk/sysroots/x86_64-petalinux-linux/usr/share/cmake/OEToolchainConfig.cmake -DCMAKE_PREFIX_PATH=/install/Debug -DCMAKE_INSTALL_PREFIX=/proj/xsjhdstaff6/linqiang/vai1.3/integrate_test_vai1.3/petalinux_sdk/sysroots/aarch64-xilinx-linux/install/Debug /proj/xsjhdstaff6/linqiang/vai1.3/integrate_test_vai1.3/Vitis-AI/tools/Vitis-AI-Runtime/VART/xir
-- The C compiler identification is GNU 9.2.0
-- The CXX compiler identification is GNU 9.2.0
-- Check for working C compiler: /proj/xsjhdstaff6/linqiang/vai1.3/integrate_test_vai1.3/petalinux_sdk/sysroots/x86_64-petalinux-linux/usr/bin/aarch64-xilinx-linux/aarch64-xilinx-linux-gcc
-- Check for working C compiler: /proj/xsjhdstaff6/linqiang/vai1.3/integrate_test_vai1.3/petalinux_sdk/sysroots/x86_64-petalinux-linux/usr/bin/aarch64-xilinx-linux/aarch64-xilinx-linux-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /proj/xsjhdstaff6/linqiang/vai1.3/integrate_test_vai1.3/petalinux_sdk/sysroots/x86_64-petalinux-linux/usr/bin/aarch64-xilinx-linux/aarch64-xilinx-linux-g++
-- Check for working CXX compiler: /proj/xsjhdstaff6/linqiang/vai1.3/integrate_test_vai1.3/petalinux_sdk/sysroots/x86_64-petalinux-linux/usr/bin/aarch64-xilinx-linux/aarch64-xilinx-linux-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found importable target unilog::unilog: /proj/xsjhdstaff6/linqiang/vai1.3/integrate_test_vai1.3/petalinux_sdk/sysroots/aarch64-xilinx-linux/install/Debug/lib/libunilog.so.1.3.0
-- Found OpenSSL: /proj/xsjhdstaff6/linqiang/vai1.3/integrate_test_vai1.3/petalinux_sdk/sysroots/aarch64-xilinx-linux/usr/lib/libcrypto.so (found version "1.1.1d")
-- Found Boost: /proj/xsjhdstaff6/linqiang/vai1.3/integrate_test_vai1.3/petalinux_sdk/sysroots/aarch64-xilinx-linux/usr/lib/cmake/Boost-1.71.0/BoostConfig.cmake (found version "1.71.0")
-- Found Protobuf: /proj/xsjhdstaff6/linqiang/vai1.3/integrate_test_vai1.3/petalinux_sdk/sysroots/aarch64-xilinx-linux/usr/lib/libprotobuf.so;-lpthread (found version "3.9.2")
-- Configuring done
-- Generating done
-- Build files have been written to: /home/linqiang/build/build.linux.2020.2.aarch64.Debug/xir
[  3%] Running cpp protocol buffer compiler on /proj/xsjhdstaff6/linqiang/vai1.3/integrate_test_vai1.3/Vitis-AI/tools/Vitis-AI-Runtime/VART/xir/src/xir/proto/graph_proto_v1.proto
[  3%] Running cpp protocol buffer compiler on /proj/xsjhdstaff6/linqiang/vai1.3/integrate_test_vai1.3/Vitis-AI/tools/Vitis-AI-Runtime/VART/xir/src/xir/proto/graph_proto_v2.proto
Scanning dependencies of target xir
[ 11%] Building CXX object src/xir/CMakeFiles/xir.dir/graph_proto_v1.pb.cc.o
[ 11%] Building CXX object src/xir/CMakeFiles/xir.dir/util/data_type.cpp.o
[ 11%] Building CXX object src/xir/CMakeFiles/xir.dir/util/internal_util.cpp.o
[ 11%] Building CXX object src/xir/CMakeFiles/xir.dir/graph_proto_v2.pb.cc.o
[ 13%] Building CXX object src/xir/CMakeFiles/xir.dir/util/error_code.cpp.o
[ 13%] Building CXX object src/xir/CMakeFiles/xir.dir/util/tool_function.cpp.o
[ 16%] Building CXX object src/xir/CMakeFiles/xir.dir/attrs/attr_def.cpp.o
[ 16%] Building CXX object src/xir/CMakeFiles/xir.dir/attrs/attr_expander_imp.cpp.o
[ 18%] Building CXX object src/xir/CMakeFiles/xir.dir/attrs/attrs.cpp.o
[ 23%] Building CXX object src/xir/CMakeFiles/xir.dir/op/built_in_ops.cpp.o
[ 23%] Building CXX object src/xir/CMakeFiles/xir.dir/attrs/attrs_imp.cpp.o
[ 23%] Building CXX object src/xir/CMakeFiles/xir.dir/op/op_def.cpp.o
[ 27%] Building CXX object src/xir/CMakeFiles/xir.dir/op/op_imp.cpp.o
[ 28%] Building CXX object src/xir/CMakeFiles/xir.dir/op/op_def_factory_imp.cpp.o
[ 32%] Building CXX object src/xir/CMakeFiles/xir.dir/op/shape_inference.cpp.o
[ 37%] Building CXX object src/xir/CMakeFiles/xir.dir/graph/graph_imp.cpp.o
[ 38%] Building CXX object src/xir/CMakeFiles/xir.dir/graph/subgraph_imp.cpp.o
[ 45%] Building CXX object src/xir/CMakeFiles/xir.dir/tensor/tensor.cpp.o
[ 45%] Building CXX object src/xir/CMakeFiles/xir.dir/graph/serialize_v2.cpp.o
[ 45%] Building CXX object src/xir/CMakeFiles/xir.dir/graph/graph.cpp.o
[ 45%] Building C object src/xir/CMakeFiles/xir.dir/version.c.o
[ 38%] Building CXX object src/xir/CMakeFiles/xir.dir/graph/elf2xir.cpp.o
[ 42%] Building CXX object src/xir/CMakeFiles/xir.dir/tensor/tensor_imp.cpp.o
[ 42%] Building CXX object src/xir/CMakeFiles/xir.dir/graph/graph_template_imp.cpp.o
[ 45%] Building CXX object src/xir/CMakeFiles/xir.dir/graph/graph_template.cpp.o
[ 47%] Linking CXX shared library libxir.so
[ 47%] Built target xir
[... cut ...]
[ 47%] Built target xir
[ 62%] Built target xir_util
[ 66%] Built target test_graph_impl.bin
[ 69%] Built target test_deserialize_model.bin
[ 72%] Built target test_graph_iso.bin
[ 76%] Built target test_graph_template.bin
[ 79%] Built target test_subgraph_iso.bin
[ 83%] Built target test_deserialize.bin
[ 86%] Built target demo.bin
[ 89%] Built target test_c_api.bin
[ 93%] Built target test_subgraph.bin
[ 96%] Built target test_subgraph_topo.bin
[100%] Built target test_c_api
Install the project...
`

HSqure, 2021-01-15:
@qianglin-xlnx Thank you for the quick reply！

I try to compile it again following your solution, but I found the error still remains. Details are as follows:

**1.** Actually I don't need `glog` from **step2**. Thanks.

**2.** I find that my system is true **bash**:
```
xxx@ubuntu:~/Vitis-ai-suit-v1.3/Vitis-AI-master/tools/Vitis-AI-Runtime/VART/xir$ echo $SHELL
/bin/bash
```
Then I still use `/bin/bash cmake.sh` to compile **xir**. But unfortunately, there is the same old error.

### Interim solution
According to the error, I just modify `PROJECT_GIT_COMMIT_ID` to something random constant like `10`, then it worked! I passed the whole VART compilation. But sadly I still don't know where the problem comes from.

### ERROR loaction

1. Here's where the error occurs.
[https://github.com/Xilinx/Vitis-AI/blob/master/tools/Vitis-AI-Runtime/VART/xir/src/xir/util/tool_function.cpp](url)
 `line 71` `const auto ret = std::string{PROJECT_GIT_COMMIT_ID};`

2. Here's a definition interface, which waits for value from cmakelists.
[https://github.com/Xilinx/Vitis-AI/blob/master/tools/Vitis-AI-Runtime/VART/xir/src/xir/config/config.hpp.in](url)
`line 23` `fine PROJECT_GIT_COMMIT_ID "@PROJECT_GIT_COMMIT_ID@"`

3. Here's cmakelists which gives value to `PROJECT_GIT_COMMIT_ID`
[https://github.com/Xilinx/Vitis-AI/blob/master/tools/Vitis-AI-Runtime/VART/xir/src/xir/CMakeLists.txt](url)
`line 26` `OUTPUT_VARIABLE PROJECT_GIT_COMMIT_ID`

Then I have no idea how it goes and what `OUTPUT_VARIABLE` means about.

qianglin-xlnx, 2021-01-15:
@HSqure 
I'm sorry I still can't reproduce the issue you mentioned. I notice the following error from your log
[image]

Do you git clone the whole Vitis-AI? or Do you remove the .git folder.
[image]
If possible, could you git clone Vitis-AI and try it again.

HSqure, 2021-01-15:
@qianglin-xlnx It worked! Thanks for your help!

I got the Vitis-AI-master by just download `.zip` file before, maybe it's the reason why it lacks `.git` folder.

I will close this issure#254.

At last, Thank you again!

closed 2021-01-15 by HSqure