Xilinx/Vitis-AI #256: Not available ethernet on ZCU 102

suhyung-jang, 2021-01-15:
The board which was flashed using "xilinx-zcu102-dpu-v2020.2-v1.3.0.img.gz" has some issue about an ethernet.
When I use ifconfig command on the board through UART, bellow message was shown.

`eth0      Link encap:Ethernet  HWaddr 00:0A:35:00:22:01  
          inet6 addr: fe80::20a:35ff:fe00:2201/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1 errors:0 dropped:0 overruns:0 frame:0
          TX packets:57 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:46 (46.0 B)  TX bytes:16050 (15.6 KiB)
          Interrupt:32 

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:26 errors:0 dropped:0 overruns:0 frame:0
          TX packets:26 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:1820 (1.7 KiB)  TX bytes:1820 (1.7 KiB)
`

After flasing the board, i did nothing. Should I do something?

andyluo7, 2021-01-15:
@suhyung-jang I connected ZCU102 Ethernet to my Ethernet switch. It can get the IP address automatically.

suhyung-jang, 2021-01-15:
@andyluo7 Thanks for your reply. I also tried to connect using working ethernet cable to my ethernet switch. But As you see, it can not get IP. Did you do anything extra for the ethernet connection like setting jumper ?

qianglin-xlnx, 2021-02-02:
> 
> 
> @andyluo7 Thanks for your reply. I also tried to connect using working ethernet cable to my ethernet switch. But As you see, it can not get IP. Did you do anything extra for the ethernet connection like setting jumper ?

In this case, you have to set the network manually. You can modify the  /etc/network/interfaces file. Or Use the ifconfig command via the UART interface to set the IP address of the board.

suhyung-jang, 2021-02-17:
@qianglin-xlnx Thanks you for your kind help.

closed 2021-02-17 by suhyung-jang