Xilinx/Vitis-AI #257: Is it possible to deploy my own QNN?

Kelvinyu1117, 2021-01-16:
I would like to train the network with my own quantizer, it is possible to deploy such a model onto the board?
The weights of the network are quantized in low precision but stored in fp32. All the model development will be done in using Pytorch.

niuxjxlnx, 2021-01-18:
In general, Vitis-AI quantizer does not support customized quantization strategy because it is tightly related with hardware design.
And I suggest you check our QAT(Quantization Aware Training) part in user guide, that maybe meet your requirement.

Kelvinyu1117, 2021-01-30:
I am trying to quantize the ResNet in Pytorch. What long time does it take typically? 
I am not sure that is it normal to wait for a long time on this stage? 
[image]

niuxjxlnx, 2021-02-01:
According to the network throughput and layers number, the stage upper you shows will need minutes to several hours. 
The stage is in fast finetuning. You can limit the data items number in fast finetuning, for example, 1000, to accelerate the process.

Kelvinyu1117, 2021-02-04:
I would like to ask how can I fix the following warning?

```DPU prefers xir::Op{name = CifarResNet__CifarResNet_Conv2d_conv1__input_2, type = conv2d-fix}'s input batch to be 1, but it's 128 now. So it will be assigned to CPU.```

It seems that all the layer are assigned to CPU.

Kelvinyu1117, 2021-02-06:
I can compile the model successfully. However, I encounter an error when I try to deploy to Alveo U50.
I follow the example to write my code: 
https://github.com/Xilinx/Vitis-In-Depth-Tutorial/tree/master/Machine_Learning/Design_Tutorials/09-mnist_pyt-master

and I actually used get_child_subgraph_dpu() in the example directly: 
https://github.com/Xilinx/Vitis-In-Depth-Tutorial/blob/master/Machine_Learning/Design_Tutorials/09-mnist_pyt-master/files/application/app_mt.py

[image]
 but it gives me the following error
[image]

It seems the error thrown by ```cs.get_attr("device")```, I have used the command ```xir dump_txt <xmodel> <txt _file>``` to dump the model information to a text file and I can see there is a "device" attribute for the operators.

I have also tried to run the example directly instead of modifying it and writing my own code. For the example code mention above, I got the following error:
[image]

I'm not sure how to debug this problem, can you give me some ideas?

lishixlnx, 2021-02-22:
Use one classic data as the test data, 
then compare the input layer of your U50 network and the input layer of your qantized network, make sure they are same (or almost same),  it not, please improve your deploy code to make it same.
then compare the output layer of your U50 network and the output layer of your qantized network, check if they are same or with much difference.  
if they are same (or alost same), then check your deploy code of the post-process part.
if they are different, you should check your network.

Kelvinyu1117, 2021-03-07:
Thank you for your comments, I have successfully deployed the model on U50.
However, is it possible to know the model size before/after quantization? As the model is quantized into 8 bit, there should be some reduction in model size?

niuxjxlnx, 2021-03-15:
Hi, @Kelvinyu1117 :

Model size/resource statistic summary will be supported in next release. 
Model size should be reduced by 8 bits quantization, but may not be proportional.

Kelvinyu1117, 2021-03-18:
So, is it not possible to get those information at this moment? For example, can I write some script to calculate the result?

niuxjxlnx, 2021-03-18:
Yes, it is possible.

Kelvinyu1117, 2021-03-18:
Can you give some advice for writing those scripts to get the resource utilization?

niuxjxlnx, 2021-03-22:
You can use some modules to do this? such as thop/torchstat/torchsummary.

Kelvinyu1117, 2021-03-22:
How about the quantized model statistics? Can I use the same approach?
As the weight has been quantized into lower bit width.

niuxjxlnx, 2021-03-22:
In quantizer, weight are still float value but the value were changed to (quantization step * a integer value in [-128, 127)). So nothing special for quantizatized weights on size among quantization, you can get float model weights count and calculated quantized weights size if weights values are integer.

Kelvinyu1117, 2021-03-22:
So you mean the model size = number of total trainable params * quantized weights size (8 bits/4bits/2bits)?

The quantization is done by the AI quantizer (8bits), I think the weights values are integer right?