Xilinx/Vitis-AI #258: [VAI 1.3] Compilation failed for cf_resnet50_imagenet_224_224_7.7G_1.3

hill19850213, 2021-01-16:
Hi,

I have downloaded the cf_resnet50_imagenet_224_224_7.7G_1.3 and have tried to compile it.
There was an error occurred during compilation.
**************************************************
* VITIS_AI Compilation - Xilinx Inc.`
**************************************************
[libprotobuf ERROR google/protobuf/text_format.cc:309] Error parsing text-format caffe.NetParameter: 33:22: Message type "caffe.FixedParameter" has no field named "follow_data_layer".`
[VAI_C][Error] Open prototxt file failed: ./cf_resnet50_imagenet_224_224_7.7G_1.3/quantized/deploy.prototxt

Here is my script:
> #!/bin/sh
>
>TARGET=ZCU102
>NET_NAME=resnet50
>DEPLOY_MODEL_PATH=vai_q_output
>CF_NETWORK_PATH=./cf_resnet50_imagenet_224_224_7.7G_1.3
>
>ARCH=./arch.json
>
>vai_c_caffe --prototxt ${CF_NETWORK_PATH}/quantized/deploy.prototxt \
>           --caffemodel ${CF_NETWORK_PATH}/quantized/deploy.caffemodel \
>           --arch ${ARCH} \
>	    --output_dir ${CF_NETWORK_PATH}/vai_c_output_${TARGET}/ \
>	    --net_name ${NET_NAME} \
>            --options  "{'save_kernel':''}"

How to fix it?

Sitting

qianglin-xlnx, 2021-01-17:
Hi @hill19850213 
Could you check if you have download the latest cf_resnet50_imagenet_224_224_7.7G_1.3.zip package, we have updated it last week. The md5 of cf_resnet50_imagenet_224_224_7.7G_1.3.zip package is: 
4ef5c8c680a52dc22252a9d82417d0bb  cf_resnet50_imagenet_224_224_7.7G_1.3.zip
When I use the script to compile the model, it shows normal, as shown below.

(vitis-ai-caffe) Vitis-AI /workspace/model_test > cat compiler.sh
#!/bin/sh

TARGET=ZCU102
NET_NAME=resnet50
DEPLOY_MODEL_PATH=vai_q_output
CF_NETWORK_PATH=./cf_resnet50_imagenet_224_224_7.7G_1.3

ARCH=/opt/vitis_ai/compiler/arch/DPUCZDX8G/ZCU102/arch.json

vai_c_caffe --prototxt ${CF_NETWORK_PATH}/quantized/deploy.prototxt     \
    --caffemodel ${CF_NETWORK_PATH}/quantized/deploy.caffemodel         \
    --arch ${ARCH}                                                      \
    --output_dir ${CF_NETWORK_PATH}/vai_c_output_${TARGET}/             \
    --net_name ${NET_NAME}                                              \
    --options "{'save_kernel':''}"

(vitis-ai-caffe) Vitis-AI /workspace/model_test > bash compiler.sh
**************************************************
* VITIS_AI Compilation - Xilinx Inc.
**************************************************
[INFO] Namespace(inputs_shape=None, layout='NCHW', model_files=['./cf_resnet50_imagenet_224_224_7.7G_1.3/quantized/deploy.caffemodel'], model_type='caffe', out_filename='./cf_resnet50_imagenet_224_224_7.7G_1.3/vai_c_output_ZCU102//resnet50_org.xmodel', proto='./cf_resnet50_imagenet_224_224_7.7G_1.3/quantized/deploy.prototxt')
[INFO] caffe model: cf_resnet50_imagenet_224_224_7.7G_1.3/quantized/deploy.caffemodel
[INFO] caffe model: cf_resnet50_imagenet_224_224_7.7G_1.3/quantized/deploy.prototxt
[INFO] parse raw model     :100%|████████████████████████████████████████| 194/194 [00:08<00:00, 22.23it/s]
[INFO] infer shape (NCHW)  :100%|████████████████████████████████████████| 196/196 [00:00<00:00, 5807.73it/s]
[INFO] infer shape (NHWC)  :100%|████████████████████████████████████████| 196/196 [00:00<00:00, 6962.97it/s]
[INFO] generate xmodel     :100%|████████████████████████████████████████| 196/196 [00:00<00:00, 326.99it/s]
[INFO] generate xmodel: /workspace/model_test/cf_resnet50_imagenet_224_224_7.7G_1.3/vai_c_output_ZCU102/resnet50_org.xmodel
[UNILOG][INFO] The compiler log will be dumped at "/tmp/vitis-ai-user/log/xcompiler-20210116-232643-147"
[UNILOG][INFO] Target architecture: DPUCZDX8G_ISA0_B4096_MAX_BG2
[UNILOG][INFO] Compile mode: dpu
[UNILOG][INFO] Debug mode: function
[UNILOG][INFO] Target architecture: DPUCZDX8G_ISA0_B4096_MAX_BG2
[UNILOG][INFO] Graph name: deploy, with op num: 412
[UNILOG][INFO] Begin to compile...
[UNILOG][INFO] Total device subgraph number 3, DPU subgraph number 1
[UNILOG][INFO] Compile done.
[UNILOG][INFO] The meta json is saved to "/workspace/model_test/./cf_resnet50_imagenet_224_224_7.7G_1.3/vai_c_output_ZCU102/meta.json"
[UNILOG][INFO] The compiled xmodel is saved to "/workspace/model_test/./cf_resnet50_imagenet_224_224_7.7G_1.3/vai_c_output_ZCU102//resnet50.xmodel"
[UNILOG][INFO] The compiled xmodel's md5sum is ef9de04ada7fd0189696e8b0f3abd0d6, and been saved to "/workspace/model_test/./cf_resnet50_imagenet_224_224_7.7G_1.3/vai_c_output_ZCU102/md5sum.txt"

hill19850213, 2021-01-17:
Hi @qianglin-xlnx ,

I have checked the md5sum and it was same with latest md5sum.

(vitis-ai-caffe) root@hill213-pc:/workspace/Tool-Example/zu6eg_custom# md5sum cf_resnet50_imagenet_224_224_7.7G_1.3.zip 
4ef5c8c680a52dc22252a9d82417d0bb  cf_resnet50_imagenet_224_224_7.7G_1.3.zip
(vitis-ai-caffe) root@hill213-pc:/workspace/Tool-Example/zu6eg_custom# bash custom_resnet50.sh 
**************************************************
* VITIS_AI Compilation - Xilinx Inc.
**************************************************
[libprotobuf ERROR google/protobuf/text_format.cc:309] Error parsing text-format caffe.NetParameter: 33:22: Message type "caffe.FixedParameter" has no field named "follow_data_layer".
[VAI_C][Error] Open prototxt file failed: ./cf_resnet50_imagenet_224_224_7.7G_1.3/quantized/deploy.prototxt
(vitis-ai-caffe) root@hill213-pc:/workspace/Tool-Example/zu6eg_custom#

qianglin-xlnx, 2021-01-17:
@hill19850213 
When I use the other docker, I can reproduce the issue, as shown in the following figure.
[image]

But when I use the xilinx/vitis-ai-cpu:latest docker, the compiling shows normal. So could you please check which docker do you use? Is it the same with the following?
[image]
[image]

hill19850213, 2021-01-18:
Hi @qianglin-xlnx ,

After I changed to xilinx/vitis-ai-cpu:latest, and there was new error as below:
(vitis-ai-caffe) Vitis-AI /workspace/Tool-Example/zu6eg_custom > bash custom_resnet50.sh 
**************************************************
* VITIS_AI Compilation - Xilinx Inc.
**************************************************
[INFO] Namespace(inputs_shape=None, layout='NCHW', model_files=['./cf_resnet50_imagenet_224_224_7.7G_1.3/quantized/deploy.caffemodel'], model_type='caffe', out_filename='./cf_resnet50_imagenet_224_224_7.7G_1.3/vai_c_output_ZCU102//resnet50_org.xmodel', proto='./cf_resnet50_imagenet_224_224_7.7G_1.3/quantized/deploy.prototxt')
[INFO] caffe model: cf_resnet50_imagenet_224_224_7.7G_1.3/quantized/deploy.caffemodel
[INFO] caffe model: cf_resnet50_imagenet_224_224_7.7G_1.3/quantized/deploy.prototxt
[INFO] parse raw model     :100%|█| 194/194 [00:04<00:00, 44.12it/s]            
[INFO] infer shape (NCHW)  :100%|█| 196/196 [00:00<00:00, 16431.48it/s]         
[INFO] infer shape (NHWC)  :100%|█| 196/196 [00:00<00:00, 14503.94it/s]         
[INFO] generate xmodel     :100%|█| 196/196 [00:00<00:00, 1202.09it/s]          
Traceback (most recent call last):
  File "/opt/vitis_ai/conda/envs/vitis-ai-caffe/bin/xnnc-run", line 33, in <module>
    sys.exit(load_entry_point('xnnc==1.3.0', 'console_scripts', 'xnnc-run')())
  File "/opt/vitis_ai/conda/envs/vitis-ai-caffe/lib/python3.6/site-packages/xnnc/__main__.py", line 194, in main
    normal_run(args)
  File "/opt/vitis_ai/conda/envs/vitis-ai-caffe/lib/python3.6/site-packages/xnnc/__main__.py", line 185, in normal_run
    serialize_as_file(graph, out_filename)
  File "/opt/vitis_ai/conda/envs/vitis-ai-caffe/lib/python3.6/site-packages/xnnc/__main__.py", line 88, in serialize_as_file
    filename.parent.mkdir(parents=True)
  File "/opt/vitis_ai/conda/envs/vitis-ai-caffe/lib/python3.6/pathlib.py", line 1248, in mkdir
    self._accessor.mkdir(self, mode)
  File "/opt/vitis_ai/conda/envs/vitis-ai-caffe/lib/python3.6/pathlib.py", line 387, in wrapped
    return strfunc(str(pathobj), *args)
PermissionError: [Errno 13] Permission denied: 'cf_resnet50_imagenet_224_224_7.7G_1.3/vai_c_output_ZCU102'
(vitis-ai-caffe) Vitis-AI /workspace/Tool-Example/zu6eg_custom > 

Sitting

qianglin-xlnx, 2021-01-18:
@hill19850213 
It seems that /workspace/Tool-Example/zu6eg_custom folder don't have the write permission. Could you please check the properties of /workspace/Tool-Example/zu6eg_custom folder. 
[image]

hill19850213, 2021-01-18:
Hi @qianglin-xlnx ,

I don't know what happened

(vitis-ai-caffe) Vitis-AI /workspace/Tool-Example/zu6eg_custom > bash custom_resnet50.sh 
**************************************************
* VITIS_AI Compilation - Xilinx Inc.
**************************************************
[INFO] Namespace(inputs_shape=None, layout='NCHW', model_files=['./cf_resnet50_imagenet_224_224_7.7G_1.3/quantized/deploy.caffemodel'], model_type='caffe', out_filename='./cf_resnet50_imagenet_224_224_7.7G_1.3/vai_c_output_ZCU102//resnet50_org.xmodel', proto='./cf_resnet50_imagenet_224_224_7.7G_1.3/quantized/deploy.prototxt')
[INFO] caffe model: cf_resnet50_imagenet_224_224_7.7G_1.3/quantized/deploy.caffemodel
[INFO] caffe model: cf_resnet50_imagenet_224_224_7.7G_1.3/quantized/deploy.prototxt
[INFO] parse raw model     :100%|█████████████████████████████████████████████████████████████████| 194/194 [00:03<00:00, 54.72it/s]                
[INFO] infer shape (NCHW)  :100%|█████████████████████████████████████████████████████████████████| 196/196 [00:00<00:00, 20438.66it/s]             
[INFO] infer shape (NHWC)  :100%|█████████████████████████████████████████████████████████████████| 196/196 [00:00<00:00, 18860.75it/s]             
[INFO] generate xmodel     :100%|█████████████████████████████████████████████████████████████████| 196/196 [00:00<00:00, 1607.19it/s]              
[INFO] generate xmodel: /workspace/Tool-Example/zu6eg_custom/cf_resnet50_imagenet_224_224_7.7G_1.3/vai_c_output_ZCU102/resnet50_org.xmodel
[UNILOG][INFO] The compiler log will be dumped at "/tmp/vitis-ai-user/log/xcompiler-20210118-060201-145"
[UNILOG][INFO] Target architecture: DPUCZDX8G
[UNILOG][FATAL][TARGET_FACTORY_UNREGISTERED_TARGET][Unregistered target!] Cannot find target with name DPUCZDX8G, valid names are: {DPUCAHX8H_ISA2=>0x20200000000002a,DPUCAHX8H_ISA2_ELP2=>0x20200000000002e,DPUCAHX8L_ISA0=>0x30000000000001d,DPUCVDX8G_ISA0_B16384C64B1=>0x600000076080812,DPUCVDX8G_ISA0_B8192C32B1=>0x600000076080811,DPUCVDX8G_ISA0_B8192C32B1_ELP4=>0x600000076040411,DPUCVDX8G_ISA0_B8192C32B3=>0x600000076080831,DPUCVDX8G_ISA0_B8192C32B3_DW=>0x6000000f6088831,DPUCVDX8G_ISA0_B8192C32B3_I4W8B2=>0x600000276080831,DPUCVDX8G_ISA0_B8192C32B3_I8W4B2=>0x600000376080831,DPUCVDX8G_ISA0_B8192C32B3_I8W8B2=>0x600000176080831,DPUCVDX8H_ISA0=>0x5000000000007ee,DPUCZDI4G_ISA0_B4096_DEMO_SSD=>0x400002003220206,DPUCZDI4G_ISA0_B8192D8_DEMO_SSD=>0x400002003220207,DPUCZDX8G_ISA0_B1024_MAX=>0x1000020f7014402,DPUCZDX8G_ISA0_B1024_MIN=>0x100002022010102,DPUCZDX8G_ISA0_B1152_MAX=>0x1000020f7012203,DPUCZDX8G_ISA0_B1152_MIN=>0x100002022010103,DPUCZDX8G_ISA0_B1600_MAX=>0x1000020f7014404,DPUCZDX8G_ISA0_B1600_MIN=>0x100002022010104,DPUCZDX8G_ISA0_B2304_MAX=>0x1000020f7014405,DPUCZDX8G_ISA0_B2304_MAX_BG2=>0x1000020f6014405,DPUCZDX8G_ISA0_B2304_MIN=>0x100002022010105,DPUCZDX8G_ISA0_B3136_MAX=>0x1000020f7014406,DPUCZDX8G_ISA0_B3136_MAX_BG2=>0x1000020f6014406,DPUCZDX8G_ISA0_B3136_MIN=>0x100002022010106,DPUCZDX8G_ISA0_B4096_MAX=>0x1000020f7014407,DPUCZDX8G_ISA0_B4096_MAX_BG2=>0x1000020f6014407,DPUCZDX8G_ISA0_B4096_MAX_EM=>0x1000030f7014407,DPUCZDX8G_ISA0_B4096_MIN=>0x100002022010107,DPUCZDX8G_ISA0_B512_MAX=>0x1000020f7012200,DPUCZDX8G_ISA0_B512_MIN=>0x100002022010100,DPUCZDX8G_ISA0_B800_MAX=>0x1000020f7012201,DPUCZDX8G_ISA0_B800_MIN=>0x100002022010101}
*** Check failure stack trace: ***
This program has crashed!
Aborted (core dumped)

hill19850213, 2021-01-18:
Hi @qianglin-xlnx ,

Dose the arch.json need to update to vitis 1.3?
I still use vitis 1.2  version as below:

```
{
    "target"   : "DPUCZDX8G",
    "dcf"      : "./dpu-11-02-2020-15-15.dcf",
    "cpu_arch" : "arm64"
}

```

hill19850213, 2021-01-18:
After I update the arch.json as below, it works fine for compilation
```
{
    "target"   : "DPUCZDX8G_ISA0_B4096_MAX_BG2",
    "dcf"      : "./dpu-11-02-2020-15-15.dcf",
    "cpu_arch" : "arm64"
}
```

closed 2021-01-18 by hill19850213