Xilinx/Vitis-AI #263: fingerprint mismatch

hill19850213, 2021-01-19:
Hi,

when I ran the resnet50 , and I got fingerprint mismatch error.
```
W0110 11:13:48.316934  1383 dpu_runner_base_imp.cpp:692] CHECK fingerprint fail ! model_fingerprint 0x1000020f6014407 dpu_fingerprint 0x1000000f6014407
F0110 11:13:48.317059  1383 dpu_runner_base_imp.cpp:664] fingerprint check failure.
```
I have checked the generated arch json which is "0x1000020f6014407 ", but I don't know why the fingerprint was changed to "0x1000000f6014407" when I ran xmodel on the board.

Is the fingerprint got from dpu.xclbin?

qianglin-xlnx, 2021-01-20:
Hi @hill19850213 
Yes.
0x1000020f6014407 stands for "DPUCZDX8G_ISA0_B4096_MAX_BG2".  The model's fingerprint should be the same with the target's fingerprint. Otherwise, you can't run the model on the target.
You can use the following command to check the target's information, including fingerprint.
`/usr/bin/xlnx/xilinx_show_devices`

hill19850213, 2021-01-20:
Hi @qianglin-xlnx ,

How to view the xilinx_show_devices?

Thanks
Sitting

hill19850213, 2021-01-20:
Hi @qianglin-xlnx ,

When I ran the xilinx_show_devices, there is an error as below:

root@xilinx-zu6eg-2020_2:/usr/bin/xlnx# ./xilinx_show_devices
[   74.611394] [drm] Pid 1246 opened device
[   74.615344] [drm] Pid 1246 closed device
[   74.619403] [drm] Pid 1246 opened device
[   74.623326] [drm] Pid 1246 closed device
[   74.627483] [drm] Pid 1246 opened device
[   74.632300] [drm] Pid 1246 closed device
[   74.684687] [drm] Pid 1246 opened device
[   74.688626] [drm] Pid 1246 closed device
[   74.692623] [drm] Pid 1246 opened device
[   74.701829] [drm] sched_reset_exec: stop scheduler
[   74.701889] [drm] get section DEBUG_IP_LAYOUT err: -22
[   74.706682] [drm] get section AIE_METADATA err: -22
[   74.711980] [drm] zocl_xclbin_read_axlf ae58ff81-475e-4c58-adc0-4d1bae03ca45 ret: 0
[   74.721902] [drm] bitstream ae58ff81-475e-4c58-adc0-4d1bae03ca45 locked, ref=1
[   74.729577] [drm] No ERT scheduler on MPSoC, using KDS
[   74.741946] Internal error: synchronous external abort: 96000010 [#1] SMP
[   74.748719] Modules linked in: iptable_nat nf_nat iptable_filter dmaproxy(O) dpu(O) mali(O) zocl(    O)
[   74.757766] CPU: 3 PID: 406 Comm: zocl-scheduler- Tainted: G           O      5.4.0-xilinx-v2020.    2 #1
[   74.766971] Hardware name: ZynqMP ZCU104 RevC (DT)
[   74.771748] pstate: 40000005 (nZcv daif -PAN -UAO)
[   74.776533] pc : zocl_cu_init+0xbc/0x278 [zocl]
[   74.781058] lr : zocl_cu_init+0xb8/0x278 [zocl]
[   74.785572] sp : ffff800012fbbb80
[   74.788871] x29: ffff800012fbbb80 x28: ffff0009735d5080
[   74.794165] x27: 0000000000000000 x26: 00000000a0020000
[   74.799460] x25: 0000000000000048 x24: 0000000000000001
[   74.804755] x23: 0000000000000000 x22: ffff800011815000
[   74.810050] x21: 00000000a0020000 x20: 0000000000000001
[   74.815345] x19: ffff800011806000 x18: ffff800011203df8
[   74.820639] x17: 0000000000000000 x16: 0000000000000000
[   74.825934] x15: 00000000a0020000 x14: ffff000971c6de00
[   74.831229] x13: ffff800011e75000 x12: ffff000971338e80
[   74.836524] x11: ffff800011e41000 x10: ffff800011424a74
[   74.841819] x9 : ffff8000114023e0 x8 : ffff00096fe2ea80
[   74.847113] x7 : 0140000000000000 x6 : 0040000000000001
[   74.852408] x5 : ffff000971c54bd0 x4 : 0000000000010000
[   74.857703] x3 : ffff800011e30000 x2 : 00000000a0020000
[   74.862998] x1 : ffff800011e30000 x0 : 0000000000000000
[   74.868293] Call trace:
[   74.870732]  zocl_cu_init+0xbc/0x278 [zocl]
[   74.874906]  configure.isra.0+0x374/0x8f0 [zocl]
[   74.879515]  scheduler+0x87c/0xc88 [zocl]
[   74.883514]  kthread+0x12c/0x130
[   74.886732]  ret_from_fork+0x10/0x1c
[   74.890293] Code: f0000041 9127e021 95e3cfe3 f94006c1 (b9400021)
[   74.896375] ---[ end trace 9c2914902fa9a9d7 ]---

hill19850213, 2021-01-21:
Hi @qianglin-xlnx 
I solved this issue. Thanks you

closed 2021-01-21 by hill19850213

vaibhav810, 2021-05-07:
Facing same error..How do you solve it ?I have checked the generated arch json which is "0x1000020f6014407 ", but I don't know why the fingerprint was changed to "0x1000000f6014407" when I ran xmodel on the board.

aiqiangbrave, 2021-11-28:
> 

you can replace xclbin file，than try again！

aiqiangbrave, 2021-11-28:
you can replace xclbin file， later try again！

aiqiangbrave, 2021-11-28:
> 

you can replace xclbin file， later try again！

AfifaIshtiaq, 2022-04-21:
Faced this:
root@versal-rootfs-common-2021_2:~/demo/VART/resnet50# ./resnet50 resnet50.xmodel                            WARNING: Logging before InitGoogleLogging() is written to STDERR
I0421 07:35:39.625972   746 main.cc:292] create running for subgraph: subgraph_conv1
XAIEFAL: INFO: Resource group Avail is created.
XAIEFAL: INFO: Resource group Static is created.
XAIEFAL: INFO: Resource group Generic is created.
W0421 07:35:48.376049   746 dpu_runner_base_imp.cpp:676] CHECK fingerprint fail ! model_fingerprint 0x602001036088231 dpu_fingerprint 0x602001036088252
F0421 07:35:48.376101   746 dpu_runner_base_imp.cpp:648] fingerprint check failure.
*** Check failure stack trace: ***
Aborted
How to change fingerprint on target
root@versal-rootfs-common-2021_2:/usr/bin/xlnx# ./xilinx_show_devices
-sh: ./xilinx_show_devices: No such file or directory

qianglin-xlnx, 2022-04-21:
Hi @AfifaIshtiaq 
You need to recompile your model according to your target's bit fingerprint.
If you want to change the target's bit fingerprint, you need to config the dpu design and integrate it. Then, recompile your model according to your new target's bit fingerprint.

AfifaIshtiaq, 2022-04-21:
Hi @qianglin-xlnx Thanks for quick response I downloaded model from
wget https://www.xilinx.com/bin/public/openDownload?filename=resnet50-vck190-r2.0.0.tar.gz -O resnet50-vck190-r2.0.0.tar.gz but I am having issue in recompiling the xmodel.
How can I change fingerprint in  /opt/vitis_ai/compiler/arch/<DPU type> folder.

[image]

qianglin-xlnx, 2022-04-21:
Hi @AfifaIshtiaq 
The model you download is the compiled xmodel. You need to download the quantized model in : https://www.xilinx.com/bin/public/openDownload?filename=cf_resnet50_imagenet_224_224_7.7G_2.0.zip 
find the quantized model under "cf_resnet50_imagenet_224_224_7.7G_2.0/quantized" folder. Then compile it in Vitis-AI docker. 
vai_c_caffe

AfifaIshtiaq, 2022-04-21:
Vitis-AI /opt/vitis_ai/compiler/arch/DPUCVDX8G/VCK190 > ls
arch_edit.json  arch.json
how to edit the fingerprint in this json file with my DPU finger print as this .json is protected?

root@versal-rootfs-common-2021_2:~#  xdputil query
XAIEFAL: INFO: Resource group Avail is created.
XAIEFAL: INFO: Resource group Static is created.
XAIEFAL: INFO: Resource group Generic is created.
{   
    "DPU IP Spec":{
        "DPU Core Count":1,
        "DPU Target Version":"v1.4.1",
        "IP version":"v1.1.0",
        "generation timestamp":"2022-04-19 15-45-00",
        "git commit id":"ae7356b",
        "git commit time":2112061226,
        "regmap":"1toN version"
    },
    "VAI Version":{
        "libvart-runner.so":"Xilinx vart-runner Version: 2.0.0-c34b0446ca2c0a49345bf041d996eabf612b8614 64 2022-01-13-21:42:10 ",
        "libvitis_ai_library-dpu_task.so":"Xilinx vitis_ai_library dpu_task Version: 2.0.0-887576f58e2c191c1abb75c757af12f37fa1571e 64 2022-01-13 13:46:11 [UTC] ",
        "libxir.so":"Xilinx xir Version: xir-9120e08ab9769fc285c8a2b8c0856232de0627c2 2022-01-13-21:40:30",
        "target_factory":"target-factory.2.0.0 0228accac8740ee81bd172a487c61a04273f5ab1"
    },
    "kernels":[
        {   
            "DPU Arch":"DPUCVDX8G_ISA2_C64B5_0602001036088252",
            "DPU Batch Number":5,
            "DPU Frequency (MHz)":333,
            "IP Type":"DPU",
            "Load Parallel":7,
            "Load augmentation":"enable",
            "Load minus mean":"disable",
            "Save Parallel":2,
            "XRT Frequency (MHz)":150,
            "cu_addr":"0xa4050000",
            "cu_handle":"0xaaab15183080",
            "cu_idx":0,
            "cu_mask":1,
            "cu_name":"DPUCVDX8G:DPUCVDX8G_1",
            "device_id":0,
            "fingerprint":"0x602001036088252",
            "name":"DPU Core 0"
        }
    ]
}

aiqiangbrave, 2022-04-21:
please use: sudo chmod 777 filename.jason

then use: vim filename.json to edit file

AfifaIshtiaq, 2022-04-21:
I have to replace the target in json file of docker with one mentioned on
target as DPu_arch is it right?

In docker i read json file as
{
Target: DPUCVX8G_ISA2_C32B3
}

On versal board
DPU Arch":"DPUCVDX8G_ISA2_C64B5_0602001036088252"

dextroza, 2022-11-30:
Hi,
I also have problem with model fingerprint. 
The DPU fingerprint is: "0x101000016010407".

So, I compile pytorch model facerec_resnet20 (downloaded from model_zoo) with arch.json { "fingerprint":"0x101000016010407"}.

Compile cmd: 
`vai_c_xir -x quantized/Resnet_int.xmodel -a arch.json -o facerec_pretrained_resnet20 -n facerec_pretrained_resnet20
`
But, when I ran this .xmodel, I get the error: 

```
dpu_runner_base_imp.cpp:676] CHECK fingerprint fail ! model_fingerprint 0x1000020f6014407 dpu_fingerprint 0x101000016010407 
F1130 12:58:53.213528  1400 dpu_runner_base_imp.cpp:648] fingerprint check failure.

```
So, the question is why model_fingeprint is 0x1000020f6014407 instead 0x101000016010407?

My environment: 
Docker Image Version: 2.5.0.1260   (GPU) (tried and CPU version as well)
Vitis AI Git Hash: c80e859e3 
Build Date: 2022-09-02

OS: Petalinux 2022.1, ZCU104, custom board image

`root@zcu104_custom_plnx:/mnt# xdputil query
{
    "DPU IP Spec":{
        "DPU Core Count":1,
        "IP version":"v4.0.0",
        "enable softmax":"False"
    },
    "VAI Version":{
        "libvart-runner.so":"Xilinx vart-runner Version: 2.5.0-c26eae36f034d5a2f9b2a7bfe816b8c43311a4f8  2022-11-23-17:44:35 ",
        "libvitis_ai_library-dpu_task.so":"Xilinx vitis_ai_library dpu_task Version: 2.5.0-c26eae36f034d5a2f9b2a7bfe816b8c43311a4f8  2022-06-15 07:33:00 [UTC] ",
        "libxir.so":"Xilinx xir Version: xir-c26eae36f034d5a2f9b2a7bfe816b8c43311a4f8 2022-11-23-17:08:58",
        "target_factory":"target-factory.2.5.0 c26eae36f034d5a2f9b2a7bfe816b8c43311a4f8"
    },
    "kernels":[
        {
            "DPU Arch":"DPUCZDX8G_ISA1_B4096",
            "DPU Frequency (MHz)":200,
            "cu_idx":0,
            "fingerprint":"0x101000016010407",
            "is_vivado_flow":true,
            "name":"DPU Core 0"
        }
    ]
}
`

Thanks!

aiqiangbrave, 2022-12-14:
> > 
> 
> you can replace xclbin file， later try again！

The fingerprint of the board does not match that of the DPU model，replace it with the same

dextroza, 2022-12-14:
Hi,
I solved this issue few days ago with the correct fingerprint.
Thanks!

aiqiangbrave, 2022-12-14:
congratulations~

| |
ln
|

---- Replied Message ----
| From | ***@***.***> |
| Date | 12/14/2022 21:09 |
| To | ***@***.***> |
| Cc | ***@***.***>***@***.***> |
| Subject | Re: [Xilinx/Vitis-AI] fingerprint mismatch (#263) |

Hi,
I solved this issue few days ago with the correct fingerprint.
Thanks!

davidbenncsiro, 2023-09-22:
Okay, so where to replace the fingerprint seems unclear from the foregoing exchanges. I see this error when running:
~~~
WARNING: Logging before InitGoogleLogging() is written to STDERR
W1119 17:22:09.076498   855 dpu_runner_base_imp.cpp:733] CHECK fingerprint fail! model_fingerprint 0x101000016010407 is un-matched with actual dpu_fingerprint 0x101000056010407. Please re-compile xmodel with dpu_fingerprint 0x101000056010407 and try again.
F1119 17:22:09.076589   855 dpu_runner_base_imp.cpp:695] fingerprint check failure.
*** Check failure stack trace: ***
~~~
I look in `/opt/vitis_ai/compiler/arch/DPUCZDX8G/ZCU102/arch.json` and see:
~~~
{
    "target": "DPUCZDX8G_ISA1_B4096"
}
~~~
There is no hint of a fingerprint there.

Then, on zcu102 board:
~~~
root@xilinx-zcu102-20222:/run/media/mmcblk0p1/target_zcu102#  xdputil query
WARNING: Logging before InitGoogleLogging() is written to STDERR
E1119 17:27:11.488224   871 xdputil_query.cpp:196] Unsupported platform fingerprint: 0, cu_idx: 3
{
    "DPU IP Spec":{
        "DPU Core Count":4,
        "IP version":"v4.1.0",
        "generation timestamp":"2022-12-06 16-30-00",
        "git commit id":"8a36f42",
        "git commit time":2022120616,
        "regmap":"1to1 version"
    },
    "VAI Version":{
        "libvaip-core.so":"Xilinx vaip Version: 1.0.0-a176db67b19f94b0a31f9d24ef80322efe4494ad  2022-12-26-14:51:25 ",
        "libvart-runner.so":"Xilinx vart-runner Version: 3.0.0-2efa5fe1e56c2b2c8a7e71e9fc1636242dd50a9f  2022-12-26-14:29:49 ",
        "libvitis_ai_library-dpu_task.so":"Xilinx vitis_ai_library dpu_task Version: 3.0.0-1cccff04dc341c4a6287226828f90aed56005f4f  20 22-12-20 10:29:01 [UTC] ",
        "libxir.so":"Xilinx xir Version: xir-9204ac72103092a7b253a0c23ec7471481656940 2022-12-26-14:21:57",
        "target_factory":"target-factory.3.0.0 860ed0499ab009084e2df3004eeb9ae710c26351"
    },
    "kernels":[
        {
            "DPU Arch":"DPUCZDX8G_ISA1_B4096",
            "DPU Frequency (MHz)":300,
            "IP Type":"DPU",
            "Load Parallel":2,
            "Load augmentation":"enable",
            "Load minus mean":"disable",
            "Save Parallel":2,
            "XRT Frequency (MHz)":300,
            "cu_addr":"0x80000000",
            "cu_handle":"0xaaaad3cd9170",
            "cu_idx":0,
            "cu_mask":2,
            "cu_name":"DPUCZDX8G:DPUCZDX8G_1",
            "device_id":0,
            "fingerprint":"0x101000056010407",
            "name":"DPU Core 0"
        },
...
~~~
So I can see `"fingerprint":"0x101000056010407"` but where in `arch.json` should this go before recompiling the model?

Thanks.

quentonh, 2023-09-22:
Your technique is not wrong, however there remains a conflict.

You can see that the model was compiled using Vitis AI 2.5 by comparing the Fingerprint found here:  https://github.com/Xilinx/Vitis-AI/blob/2.5/src/Vitis-AI-Runtime/VART/target_factory/targets/DPUCZDX8G_ISA1_B4096.prototxt

By contrast, it appears that the board image that you are targeting has a newer version of the DPU IP (Vitis AI 3.0):
https://github.com/Xilinx/Vitis-AI/blob/3.0/src/vai_runtime/target_factory/targets/DPUCZDX8G_ISA1_B4096.prototxt

Ideally, you should be compiling your model using Vitis AI 3.0 given the version of the DPU that is running on the target.  The differences are slight, and you might get away with disabling the Fingerprint check, but for the moment I cannot find the documented method to do that.

https://xilinx.github.io/Vitis-AI/3.5/html/docs/reference/version_compatibility.html#all-releases
https://support.xilinx.com/s/article/DPU-fingerprint-ERROR?language=en_US

davidbenncsiro, 2023-09-25:
Thanks @quentonh.

I confirm that my Vitis-AI docker image version is V2.5 (**Docker Image Version: 2.5.0.1260**) and my zcu102 OS image is v3.0 (**xilinx-zcu102-dpu-v2022.2-v3.0.0.img**) via https://www.xilinx.com/member/forms/download/design-license-xef.html?filename=xilinx-zcu102-dpu-v2022.2-v3.0.0.img.gz

So, one approach is to use to use a xilinx-zcu102-dpu image that's V2.5. I can't see a full list of such images. Do you have a pointer to a list?

The other approach is to install a different docker (V3.0).

I may try both although flashing a new SD card with a V2.5 image may be faster.