Xilinx/Vitis-AI #265: libnccl.so.1 is missed in vitis-ai-caffe conda.

lllkiller-jlll, 2021-01-20:
Hi.

I'm trying to use Caffe in Vitis AI v1.3.
I built GPU docker image, and I successfully turned on Vitis AI docker.
But I received the message below when I use caffe.

**_caffe: error while loading shared libraries: libnccl.so.1: cannot open shared object file: No such file or directory_**

How can I fix it?

Thanks.

wangxd-xlnx, 2021-01-21:
@lllkiller-jlll Hi, we checked this library and confirm that it has been compiled in docker. So could you provide some more details about your operation? such as the models or your environment version.

lllkiller-jlll, 2021-01-25:
Hi @wangxd-xlnx.

I followed below steps:
1) Build Vitis AI v1.3 GPU docker.
2) Turn on the Vitis AI docker.
3) Activate Caffe conda environment.
4) Trying to train a model.
  - model: resnet50
  - command: caffe train --solver <path_to_solver>
5) I received above error message.

wangxd-xlnx, 2021-01-25:
Hi @lllkiller-jlll ,

We tried to reproduce this error but cannot achieve it. Could you please provide your Nvidia driver version, and CUDA/CUDNN version?

NessAupera, 2021-01-27:
I faced similar error and the environment is as follows:
NVIDIA-SMI 450.80.02    Driver Version: 450.80.02    CUDA Version: 11.0  
CUDA Version 10.0.130

NessAupera, 2021-01-27:
` conda install -c anaconda nccl
`
solved the error temporarily.

hanxue, 2021-01-28:
Hi @NessAupera ，

NCCL is one of the dependencies for building caffe in Vitis AI. The caffe package and the vitis-ai-caffe environment should not even be created successfully if nccl is not installed. 

I have extracted the content of the [GPU caffe conda meta.yaml here](https://gist.github.com/hanxue/2953a044f18ef32bababdfd440bb58a5) 

Did you see any errors in the build log?

code-junki, 2021-01-28:
I have the exact same problem while running Caffe. 
`ImportError: libnccl.so.1: cannot open shared object file: No such file or directory`

NessAupera, 2021-01-28:
> Hi @NessAupera ，
>
> NCCL is one of the dependencies for building caffe in Vitis AI. The caffe package and the vitis-ai-caffe environment should not even be created successfully if nccl is not [...]

no error was found while building docker image, and other than that it's working fine for quantization and compilation.

hanxue, 2021-01-29:
I would suggest rebuilding the vitis-ai-gpu Docker image. 

During the build process, the caffe conda environment is created here https://github.com/Xilinx/Vitis-AI/blob/master/setup/docker/docker/DockerfileGPU#L243 

I have re-built the GPU docker image internally a few times, and I see this:

```
nccl-2.8.3.1         | 23.8 MB   |            |   0% 
nccl-2.8.3.1         | 23.8 MB   | ##1        |  22% 
nccl-2.8.3.1         | 23.8 MB   | ####9      |  50% 
nccl-2.8.3.1         | 23.8 MB   | ########4  |  85% 
nccl-2.8.3.1         | 23.8 MB   | ########## | 100% 
```

When I run the Docker container and enter the `vitis-ai-caffe`, nccl is found

```
(vitis-ai-caffe) Vitis-AI /workspace > conda list | grep nccl
nccl                      2.8.3.1              h91b26fc_0  
```

code-junki, 2021-02-01:
when i run ` conda list | grep nccl`

I get `nccl                      2.8.3.1              h91b26fc_0'
however I still get the error 
`caffe: error while loading shared libraries: libnccl.so.1: cannot open shared object file: No such file or directory
`

dviettu134, 2021-05-31:
Any ideas on this problem?

I checked the shared object files in this directory and see that no libnccl.so.1 is available:

> cd /opt/vitis_ai/conda/envs/vitis-ai-caffe/lib
> ls -l | grep nccl

lrwxrwxrwx  1 root root        16 May 27 22:39 libnccl.so -> libnccl.so.2.8.3*
lrwxrwxrwx  1 root root        16 May 27 22:39 libnccl.so.2 -> libnccl.so.2.8.3*
-rwxrwxr-x  2 root root  87810288 Jan  7 14:06 libnccl.so.2.8.3*
-rw-rw-r--  2 root root 125211384 Jan  7 14:06 libnccl_static.a

It looks like we need nccl version 1.* to run Caffe?

NessAupera, 2022-11-14:
> `conda install -c anaconda nccl` solved the error temporarily.

you can do conda install -c anaconda nccl==1.* -y to make sure nccl version 1.* is install.

closed 2023-01-16 by qianglin-xlnx