Xilinx/Vitis-AI #268: ERROR in Compiling the Pre-processing Acceleration Kernel (.xo File) from WWA-TRD

HSqure, 2021-01-20:
@hanxue Hi! 

There's an error while compiling the pre-processing Acceleration kernel (.xo file generation process) from project `Vitis-AI/dsa/WAA-TRD/accel/detection-pre/`. Seems like a weird project name problem, but I have no idea how to fix it. Need help!

### Envirment
`Vitis-AI V1.3`
`XRT 2020.2`
`Vitis 2020.2`

### Operation Describe

1. Change Makefile in path `Vitis-AI/dsa/WAA-TRD/accel/detection-pre/`

`line 70`                *`TOOL_VERSION ?= 2019.2` -> `TOOL_VERSION ?= 2020.2`

2. Since I just want to generate preprocess kernel (.xo) instead of SD card image, so I wrote a script and run `source` it on`Vitis-AI/dsa/WAA-TRD/accel/detection-pre/` as below:

script: `pp_xo_gen.sh`
```
#!/bin/bash

export CUR_DIR=$PWD

VITIS_AI_V1_3=/home/xxx/Vitis-ai-suit-v1.3/Vitis-AI-master
TRD_HOME=${VITIS_AI_V1_3}/dsa/WAA-TRD

SDX_PLATFORM_DIR=${VITIS_AI_V1_3}/../platform/xilinx_${BOARD_NAME}_base_202020_1

# Envirment Setting
cd ${VITIS_AI_V1_3}/script
source settings64-vitis.sh    # Vitis Envirment Setting
source settings64-xrt.sh    # XRT Envirment Setting
cd ${CUR_DIR}

export EDGE_COMMON_SW=/home/xxx/Vitis-ai-suit-v1.3/pkg/xilinx-zynqmp-common-v2020.2
export SDX_PLATFORM=${SDX_PLATFORM_DIR}/*.xpfm

make cleanall
make xo TARGET=hw BOARD=Zynq ARCH=aarch64
```

###  ERROR Log

```
source pp_xo_gen.sh
```

```
... ...

INFO: [v++ 200-1470] Pipelining result : Target II = 1, Final II = 1, Depth = 4, loop 'L1'
INFO: [v++ 204-61] Pipelining loop 'VITIS_LOOP_164_1'.
INFO: [v++ 200-1470] Pipelining result : Target II = 1, Final II = 1, Depth = 3, loop 'VITIS_LOOP_164_1'
INFO: [v++ 200-790] **** Loop Constraint Status: All loop constraints were satisfied.
INFO: [v++ 200-789] **** Estimated Fmax: 384.62 MHz
ERROR: [v++ 213-28] Failed to generate IP.
ERROR: [v++ 60-300] Failed to build kernel(ip) pp_pipeline_accel, see log for details: /home/xxx/Vitis-ai-suit-v1.3/Vitis-AI-master/dsa/WAA-TRD/accel/detection-pre/_x_*_hw/pp_pipeline_accel/pp_pipeline_accel/vitis_hls.log
ERROR: [v++ 60-773] In '/home/xxx/Vitis-ai-suit-v1.3/Vitis-AI-master/dsa/WAA-TRD/accel/detection-pre/_x_*_hw/pp_pipeline_accel/pp_pipeline_accel/vitis_hls.log', caught Tcl error: ERROR: [ProjectBase 2-104] Project name '/home/xxx/Vitis-ai-suit-v1.3/Vitis-AI-master/dsa/WAA-TRD/accel/detection-pre/_x_*_hw/pp_pipeline_accel/pp_pipeline_accel/pp_pipeline_accel/solution/impl/ip/tmp.xpr' is illegal.
ERROR: [v++ 60-773] In '/home/xxx/Vitis-ai-suit-v1.3/Vitis-AI-master/dsa/WAA-TRD/accel/detection-pre/_x_*_hw/pp_pipeline_accel/pp_pipeline_accel/vitis_hls.log', caught Tcl error: ERROR: [IMPL 213-28] Failed to generate IP.
ERROR: [v++ 60-599] Kernel compilation failed to complete
ERROR: [v++ 60-592] Failed to finish compilation
INFO: [v++ 60-1653] Closing dispatch client.
Makefile:299: recipe for target '/home/xxx/Vitis-ai-suit-v1.3/Vitis-AI-master/dsa/WAA-TRD/accel/detection-pre/pp_pipeline_accel.xo' failed
make: *** [/home/xxx/Vitis-ai-suit-v1.3/Vitis-AI-master/dsa/WAA-TRD/accel/detection-pre/pp_pipeline_accel.xo] Error 1
```

### ERROR Trace
**1.** According to the log for details, I found `vitis_hls.log`  in path:
```
/home/xxx/Vitis-ai-suit-v1.3/Vitis-AI-master/dsa/WAA-TRD/accel/detection-pre/_x_*_hw/pp_pipeline_accel/pp_pipeline_accel/vitis_hls.log
```

```

... ...
INFO: [RTMG 210-285] Implementing FIFO 'start_for_xfMat2hlsStrm_128_9_608_608_1_46208_U0_U(pp_pipeline_accel_start_for_xfMat2hlsStrm_128_9_608_608_1_46208_U0)' using Shift Registers.
INFO: [RTMG 210-285] Implementing FIFO 'start_for_Loop_VITIS_LOOP_52_1_proc6_U0_U(pp_pipeline_accel_start_for_Loop_VITIS_LOOP_52_1_proc6_U0)' using Shift Registers.
INFO: [HLS 200-111] Finished Generating all RTL models: CPU user time: 6.14 seconds. CPU system time: 0.35 seconds. Elapsed time: 6.5 seconds; current allocated memory: 650.714 MB.
INFO: [VHDL 208-304] Generating VHDL RTL for pp_pipeline_accel.
INFO: [VLOG 209-307] Generating Verilog RTL for pp_pipeline_accel.
INFO: [HLS 200-1603] Design has MAXI bursts, see Vitis HLS GUI synthesis summary report for detailed information.
INFO: [HLS 200-790] **** Loop Constraint Status: All loop constraints were satisfied.
INFO: [HLS 200-789] **** Estimated Fmax: 384.62 MHz
INFO: [HLS 200-111] Finished Command csynth_design CPU user time: 53.69 seconds. CPU system time: 2.63 seconds. Elapsed time: 55.56 seconds; current allocated memory: 655.149 MB.
INFO: [HLS 200-1510] Running: export_design 
INFO: [IMPL 213-8] Exporting RTL as a Vivado IP.

****** Vivado v2020.2 (64-bit)
  **** SW Build 3064766 on Wed Nov 18 09:12:47 MST 2020
  **** IP Build 3064653 on Wed Nov 18 14:17:31 MST 2020
    ** Copyright 1986-2020 Xilinx, Inc. All Rights Reserved.

source run_ippack.tcl -notrace
ERROR: [ProjectBase 2-104] Project name '/home/xxx/Vitis-ai-suit-v1.3/Vitis-AI-master/dsa/WAA-TRD/accel/detection-pre/_x_*_hw/pp_pipeline_accel/pp_pipeline_accel/pp_pipeline_accel/solution/impl/ip/tmp.xpr' is illegal.
INFO: [Common 17-206] Exiting Vivado at Wed Jan 20 10:34:28 2021...
ERROR: [IMPL 213-28] Failed to generate IP.
INFO: [HLS 200-111] Finished Command export_design CPU user time: 13.58 seconds. CPU system time: 1.31 seconds. Elapsed time: 13.11 seconds; current allocated memory: 665.125 MB.
command 'ap_source' returned error code
    while executing
"source pp_pipeline_accel.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel \#0 [list source $arg] "
```
**2.** And command `export_design ` from `pipeline_accel.tcl` in path:
```
/home/huahua/Vitis-ai-suit-v1.3/Vitis-AI-master/dsa/WAA-TRD/accel/detection-pre/_x_*_hw/pp_pipeline_accel/pp_pipeline_accel/pp_pipeline_accel.tcl
```
```
catch {::common::set_param -quiet hls.xocc.mode csynth};
# 
# HLS run script generated by v++ compiler
# 

open_project pp_pipeline_accel
set_top pp_pipeline_accel
# v++ -g, -D, -I, --advanced.prop kernel.pp_pipeline_accel.kernel_flags
add_files "/home/huahua/Vitis-ai-suit-v1.3/Vitis-AI-master/dsa/WAA-TRD/accel/detection-pre/xf_pp_pipeline_accel.cpp" -cflags " -g -D __SDSVHLS__ -D HLS_NO_XIL_FPO_LIB -I /home/huahua/Vitis-ai-suit-v1.3/Vitis-AI-master/dsa/WAA-TRD/accel/detection-pre -I /home/huahua/Vitis-ai-suit-v1.3/Vitis-AI-master/dsa/WAA-TRD/accel/detection-pre/include -I /home/huahua/Vitis-ai-suit-v1.3/Vitis-AI-master/dsa/WAA-TRD/accel/detection-pre "
open_solution -flow_target vitis solution
set_part xczu15eg-ffvb1156-2-i
# v++ --hls.clock or --kernel_frequency
create_clock -period 300MHz -name default
# v++ -g or --profile_kernel stall
config_rtl -kernel_profile
# v++ --advanced.param compiler.hlsDataflowStrictMode
config_dataflow -strict_mode warning
# v++ -g
config_debug -enable
# v++ --advanced.param compiler.deadlockDetection
config_export -disable_deadlock_detection=true
# v++ --advanced.param compiler.axiDeadLockFree
config_rtl -m_axi_conservative_mode=1
config_interface -m_axi_addr64
# v++ --hls.max_memory_ports
config_interface -m_axi_auto_max_ports=0
config_export -format ip_catalog -ipname pp_pipeline_accel
csynth_design
export_design
close_project
puts "HLS completed successfully"
exit
```

It seems like this command `export_design` from HLS gave `Project name` a weird illegal name`/home/xxx/Vitis-ai-suit-v1.3/Vitis-AI-master/dsa/WAA-TRD/accel/detection-pre/_x_*_hw/pp_pipeline_accel/pp_pipeline_accel/pp_pipeline_accel/solution/impl/ip/tmp.xpr`

I don't know what happened.

hanxue, 2021-01-20:
Hi @HSqure , 

That's right, the path name has special character `*` and that is causing the problem

```
ERROR: [ProjectBase 2-104] Project name '/home/xxx/Vitis-ai-suit-v1.3/Vitis-AI-master/dsa/WAA-TRD/accel/detection-pre/_x_*_hw/pp_pipeline_accel/pp_pipeline_accel/pp_pipeline_accel/solution/impl/ip/tmp.xpr' is illegal.
```

Do you mean Vivado created the `/home/xxx/Vitis-ai-suit-v1.3/Vitis-AI-master/dsa/WAA-TRD/accel/detection-pre/_x_*_hw/` directory?

Can you rename it to something different such as

```
/home/xxx/Vitis-ai-suit-v1.3/Vitis-AI-master/dsa/WAA-TRD/accel/detection-pre/x_hw/
```

?

HSqure, 2021-01-20:
@hanxue Thanks for the quick reply.

After I do some modification, It worked ! Thank you!

I found that `*` is taken from `export SDX_PLATFORM=${SDX_PLATFORM_DIR}/*.xpfm`, which comes from in my script file `pp_xo_gen.sh`. After I change `*` to `xilinx_axu15eg_base_202020_1` , that folder name becomes `_x_xilinx_axu15eg_base_202020_1_hw`. Then I got `.xo` file, finally.

Acturely, I compiled the source file by HLS (GUI version) and got the .xo file successfully. What's the difference between GUI Flow and Script Flow in compiling?

mahadurkarm, 2021-01-20:
@HSqure 
In the  `pp_xo_gen.sh`, `export` instruction was incorrect. `export` takes exact file name, '*' operator won't work. 
GUI Flow & script flow will do same thing in the compilation.

HSqure, 2021-01-20:
@mahadurkarm Got it! Thanks for your answer!

I will close this issue.

closed 2021-01-20 by HSqure