Xilinx/Vitis-AI #269: Why are the four DPU models testing FPS on video on vitis-ai only 17? What's the problem?

Songxinlei, 2021-01-20:
Why are the four DPU models testing FPS on video on vitis-ai only 17? What's the problem?

qianglin-xlnx, 2021-01-21:
hi @Songxinlei 
Can you describe the problem you encountered in detail? Which model, target and how do you test it? Thank you.

Songxinlei, 2021-01-22:
@qianglin-xlnx I am on the ZCU102 board, according to VART /samples/adas-detection, to make modifications. Test my own model on the board of FPS. The FPS was found to be very low. I was wondering if I had done something wrong, which led to the result. I would appreciate it if you could tell me why.Thank you

lishixlnx, 2022-02-18:
@Songxinlei 
Is this issue solved?  If so, please close this issue.

qianglin-xlnx, 2022-02-23:
HI @Songxinlei 
Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-02-23 by qianglin-xlnx