Xilinx/Vitis-AI #270: pt_pointpillars_kitti_12000_100_10.8G_1.3

eerobert, 2021-01-20:
hi, 
i was trying to go thru the quantization process of mode pt_pointpillars_kitti_12000_100_10.8G_1.3 inside docker/conda environment.

when running code/test/prepare_data.sh, it reports an error of the following

_/tmp/tmph6upbitx.cc:2:22: fatal error: cuda_runtime.h: No such file or directory_

inside docker, the cuda_runtime.h is not there, although outside docker those files are there.

my question is, is it recommended that to run thru the model pt_pointpillars_kitti_12000_100_10.8G_1.3 inside xilinx docker
or it suppose to run on independent setup of pytorch?

thanks

qianglin-xlnx, 2021-01-22:
Hi @eerobert 
It seems your environment lacks some header file for GPU. Please confirm that your host has a GPU and cuda/cuda_toolkit are well installed.
You can check if "cuda_runtime.h"  exists in your "/usr/local/cuda/" directory. If not, try to install nvidia-cuda-toolkit.
If you meet some issues which could not solved, you could check this public link (https://github.com/traveller59/second.pytorch/issues) to see if someone met the same problem before.

910560, 2021-01-25:
hi,

yes, my host system do have cuda_runtime.h

but inside xilinx docker, it is not there

so again, my question is
can we run the example inside xilinx docker?

eerobert, 2021-01-25:
hi,

yes, my host system do have cuda_runtime.h

but inside xilinx docker, it is not there

so again, my question is
can we run the example inside xilinx docker?

qianglin-xlnx, 2021-01-25:
@eerobert 
Yes, we have tested the example inside Xilinx docker successfully.
When you enter the docker, remember to follow instruction on README.md to prepare the environment.
If you execute 'sudo apt-get update && sudo apt-get install libboost-all-dev && sudo apt-get install nvidia-cuda-toolkit' following the instruction, you could find "/usr/include/cuda_runtime.h" under the docker environment.

eerobert, 2021-01-26:
i see. so the docker and conda is not for all model

or in other words, each model will have their own environment need to be setup (inside docker)

thanks

closed 2021-01-27 by eerobert