Xilinx/Vitis-AI #272: Where is fix_train_test.prototxt ?how to fine-tune my model

Songxinlei, 2021-01-22:
Where is fix_train_test.prototxt  and what file it refers to. I want to fine-tune the accuracy of my model, but I can’t find where the fix_train_test.prototxt  is and I don’t know how to do it. If someone could tell me how to fine-tune the calibration accuracy, I would be very grateful.

dingguodong-826, 2021-01-23:
I have the same confusion

wangxd-xlnx, 2021-01-25:
Hi @Songxinlei @dingguodong-826 

You could find the quantize_train_test.prototxt (that is the fix_train_test.prototxt) and it's used for quantized-point training and testing with quantize_train_test.caffemodel on GPU when datalayer modified to your own data path.   
More details, you could refer to this user guide [https://www.xilinx.com/support/documentation/sw_manuals/vitis_ai/1_3/ug1414-vitis-ai.pdf] and this release page [https://github.com/Xilinx/Vitis-AI/tree/master/models/AI-Model-Zoo#model-directory-structure](url)

Songxinlei, 2021-01-26:
@wangxd-xlnx  Thank you very much . I have tried this method,but I got the error. 

`F0125 18:37:54.044116 16581 insert_splits.cpp:30] Unknown bottom blob 'data' (layer 'data_fixed', bottom index 0)
*** Check failure stack trace: ***
`
Do you  know how to solve this problem？

dextroza, 2021-11-29:
any updates on this? I would like to finetune the quantized caffe model, but I am getting the same error, "Unknown bottom blob 'data' (layer 'data_fixed', bottom index 0)".

Model prototxt looks like this:

layer {
  name: "data"
  type: "ImageData"
  top: "data"
  top: "label"
  include {
    phase: TRAIN
  }
  transform_param {
    mirror: false
    yolo_height: 416
    yolo_width: 416
  }
  image_data_param {
    source: "full_path_to_calib.txt"
    batch_size: 1
    shuffle: false
    root_folder: "full_path_to_images_dir/"
  }
}
layer {
  name: "data_fixed"
  type: "FixedNeuron"
  bottom: "data"
  top: "data"
  param {
    lr_mult: 0
    decay_mult: 0
  }
  fixed_param {
    fixed_method: OVER_FLOW
    bit_width: 8
    follow_data_layer: true
  }
}

Any help would be appreciated!

Best regards,
dextroza

Songxinlei, 2021-11-29:
Should there be a difference here？

wangxd-xlnx, 2022-02-23:
Hi @Songxinlei @dextroza 

Has your problem been resolved? This type of error is often caused by the mismatch between the your data and the parameters in prototxt. 
You could check whether the path of data directory is correct and whether the data format matches the definition in quantize_train_test.prototxt.

qianglin-xlnx, 2022-03-07:
Hi @Songxinlei 
Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-03-07 by qianglin-xlnx