Xilinx/Vitis-AI #274: SERVICE_INVALID error with U50 using 6E300M DPU engine

zohourih, 2021-01-23:
I am trying to test the latest Vitis AI v1.3 with a U50 board and the provided docker image (v1.3.411) by Xilinx. I had previously successfully ran multiple of the existing networks on the same machine using Vitis AI v1.2 and the 6E300M DPU engine. However, with Vitis AI v1.3, I reliably get the following error regardless of the network:

```/workspace/tools/Vitis-AI-Library/overview/samples/classification > ./test_performance_classification resnet50 test_performance_classification.list -t 1 -s 10
loading xclbin: /opt/xilinx/overlaybins/dpu.xclbin
errCode: errCode: 22
errCode String: SERVICE_INVALID
myHandle: 0
valid: 1

terminate called after throwing an instance of 'std::runtime_error'
  what():  Error: could not acquire device handle
Aborted (core dumped)
```

Restarting XButler from outside the docker or running `xbutil reset` does not make any difference. The 1E333M DPU engine works just fine in the same docker instance.

wilderfield, 2021-01-23:
See if 

sudo journalctl -u xbutler

Gives you any clues.

Also check that this file exists:

> loading xclbin: /opt/xilinx/overlaybins/dpu.xclbin

zohourih, 2021-01-25:
@wilderfield This is the log:

```Jan 26 07:26:55 Edge-tower xbutler[8741]: -------------------------------------
Jan 26 07:26:55 Edge-tower xbutler[8741]:       New Connection From Client
Jan 26 07:26:55 Edge-tower xbutler[8741]: -------------------------------------
Jan 26 07:26:55 Edge-tower xbutler[8741]: Server: got connection on Tue Jan 26 07:26:55 2021
Jan 26 07:26:55 Edge-tower xbutler[8741]: Server: got connection from 127.0.0.1 port 53730
Jan 26 07:26:55 Edge-tower xbutler[8741]: ServerBooklet: Add socket: 4.
Jan 26 07:26:55 Edge-tower xbutler[8741]: ServerBooklet: Add port: 58065.
Jan 26 07:26:55 Edge-tower xbutler[8741]: Servicing PID 637
Jan 26 07:26:55 Edge-tower xbutler[8741]: -------------------------------------
Jan 26 07:26:55 Edge-tower xbutler[8741]: Starting Ping...
Jan 26 07:26:55 Edge-tower xbutler[8741]: Ping: Receiving user name...
Jan 26 07:26:55 Edge-tower xbutler[8741]: Ping: user name is: vitis-ai-user
Jan 26 07:26:55 Edge-tower xbutler[8741]: ServerBooklet: Add socket: 4.
Jan 26 07:26:55 Edge-tower xbutler[8741]: ServerBooklet: Add username: vitis-ai-user.
Jan 26 07:26:55 Edge-tower xbutler[8741]: Ping: Sending response...
Jan 26 07:26:55 Edge-tower xbutler[8741]: Ping: Response is: SUCCESS
Jan 26 07:26:55 Edge-tower xbutler[8741]: Finishing Ping.
Jan 26 07:26:55 Edge-tower xbutler[8741]: -------------------------------------
Jan 26 07:26:55 Edge-tower xbutler[8741]:         New Message From Client
Jan 26 07:26:55 Edge-tower xbutler[8741]: -------------------------------------
Jan 26 07:26:55 Edge-tower xbutler[8741]: Server: got connection on Tue Jan 26 07:26:55 2021
Jan 26 07:26:55 Edge-tower xbutler[8741]: Server: got connection from 127.0.0.1 port 53730
Jan 26 07:26:55 Edge-tower xbutler[8741]: ServerBooklet: Already exist socket: 4.
Jan 26 07:26:55 Edge-tower xbutler[8741]: ServerBooklet: Already exist port: 58065.
Jan 26 07:26:55 Edge-tower xbutler[8741]: Servicing PID 637
Jan 26 07:26:55 Edge-tower xbutler[8741]: -------------------------------------
Jan 26 07:26:55 Edge-tower xbutler[8741]: Starting AcquireCU...
Jan 26 07:26:55 Edge-tower xbutler[8741]: AcquireCU: Receiving user name...
Jan 26 07:26:55 Edge-tower xbutler[8741]: AcquireCU: user name is: vitis-ai-user
Jan 26 07:26:55 Edge-tower xbutler[8741]: ServerBooklet: Already exist socket: 4.
Jan 26 07:26:55 Edge-tower xbutler[8741]: ServerBooklet: Already exist username: vitis-ai-user.
Jan 26 07:26:55 Edge-tower xbutler[8741]: AcquireCU: Receiving kernel name...
Jan 26 07:26:55 Edge-tower xbutler[8741]: AcquireCU: Kernel name is: DPUCAHX8L
Jan 26 07:26:55 Edge-tower xbutler[8741]: AcquireCU: Receiving xclbin batch...
Jan 26 07:26:55 Edge-tower xbutler[8741]: AcquireCU: Non-UDF Mode.
Jan 26 07:26:55 Edge-tower xbutler[8741]: AcquireCU: executing algo.
Jan 26 07:26:55 Edge-tower xbutler[8741]: AcquireCU: Validating UDF result...
Jan 26 07:26:55 Edge-tower xbutler[8741]: ERROR: AcquireCU: FAILED!
Jan 26 07:26:55 Edge-tower xbutler[8741]: AcquireCU: Sending response...
Jan 26 07:26:55 Edge-tower xbutler[8741]: AcquireCU: Response is: ASSIGNMENT_UNFEASIBLE
Jan 26 07:26:55 Edge-tower xbutler[8741]: AcquireCU: Sending handle...
Jan 26 07:26:55 Edge-tower xbutler[8741]: AcquireCU: Handle is: 0
Jan 26 07:26:55 Edge-tower xbutler[8741]: AcquireCU: Sending result...
Jan 26 07:26:55 Edge-tower xbutler[8741]: Finishing AcquireCU.
Jan 26 07:26:55 Edge-tower xbutler[8741]: -------------------------------------
Jan 26 07:26:55 Edge-tower xbutler[8741]: -------------------------------------
Jan 26 07:26:55 Edge-tower xbutler[8741]:         New Message From Client
Jan 26 07:26:55 Edge-tower xbutler[8741]: -------------------------------------
Jan 26 07:26:55 Edge-tower xbutler[8741]: Server: got connection on Tue Jan 26 07:26:55 2021
Jan 26 07:26:55 Edge-tower xbutler[8741]: Server: got connection from 127.0.0.1 port 53730
Jan 26 07:26:55 Edge-tower xbutler[8741]: ServerBooklet: Already exist socket: 4.
Jan 26 07:26:55 Edge-tower xbutler[8741]: ServerBooklet: Already exist port: 58065.
Jan 26 07:26:55 Edge-tower xbutler[8741]: Servicing PID 637
Jan 26 07:26:55 Edge-tower xbutler[8741]: -------------------------------------
Jan 26 07:26:55 Edge-tower xbutler[8741]: Starting AcquireCU...
Jan 26 07:26:55 Edge-tower xbutler[8741]: AcquireCU: Receiving user name...
Jan 26 07:26:55 Edge-tower xbutler[8741]: AcquireCU: user name is: vitis-ai-user
Jan 26 07:26:55 Edge-tower xbutler[8741]: ServerBooklet: Already exist socket: 4.
Jan 26 07:26:55 Edge-tower xbutler[8741]: ServerBooklet: Already exist username: vitis-ai-user.
Jan 26 07:26:55 Edge-tower xbutler[8741]: AcquireCU: Receiving kernel name...
Jan 26 07:26:55 Edge-tower xbutler[8741]: AcquireCU: Kernel name is: DPUCAHX8H_3ENGINE
Jan 26 07:26:55 Edge-tower xbutler[8741]: AcquireCU: Receiving xclbin batch...
Jan 26 07:26:55 Edge-tower xbutler[8741]: AcquireCU: Non-UDF Mode.
Jan 26 07:26:55 Edge-tower xbutler[8741]: AcquireCU: executing algo.
Jan 26 07:26:55 Edge-tower xbutler[8741]: AcquireCU: Validating UDF result...
Jan 26 07:26:55 Edge-tower xbutler[8741]: ERROR: AcquireCU: FAILED!
Jan 26 07:26:55 Edge-tower xbutler[8741]: AcquireCU: Sending response...
Jan 26 07:26:55 Edge-tower xbutler[8741]: AcquireCU: Response is: ASSIGNMENT_UNFEASIBLE
[... 44 lines cut ...]
Jan 26 07:26:55 Edge-tower xbutler[8741]: SubscribeService: user name is: vitis-ai-user
Jan 26 07:26:55 Edge-tower xbutler[8741]: ServerBooklet: Already exist socket: 4.
[... cut ...]
Jan 26 07:26:55 Edge-tower xbutler[8741]: ServerBooklet: Already exist port: 58065.
Jan 26 07:26:55 Edge-tower xbutler[8741]: -------------------------------------
Jan 26 07:26:55 Edge-tower xbutler[8741]:       Detected Dead Connection
Jan 26 07:26:55 Edge-tower xbutler[8741]: Removing socket: 4
Jan 26 07:26:55 Edge-tower xbutler[8741]: ServerBooklet: Removing socket: 4.
Jan 26 07:26:55 Edge-tower xbutler[8741]: -------------------------------------```

The bitstream file certainly exists in the aforementioned location.

hanxue, 2021-04-09:
Hi @zohourih ,

Is there a possibility that mis-matched / wrong xclbin has been used? Seems like V3E xclbin is used to run a V3ME model, from this error message

```
AcquireCU: Kernel name is: DPUCAHX8H_3ENGINE
```

SuZhan0322, 2022-09-01:
Hi @zohourih , Please reopen this issue if you still have any other questions. thanks!

closed 2022-09-01 by SuZhan0322