Xilinx/Vitis-AI #275: Lower batch-1 performance with 1E333 compared to 6E300

zohourih, 2021-01-23:
I am trying to compare batch-1 performance between the 1E333 and 6E300 DPU variants for U50 by setting `-t 1` in the `test_performance_classification` benchmark. My understanding is that the 1E333 variant has one engine running at 333 MHz and is optimized for batch-1, while the 6E300 variant runs the lower frequency of 300 MHz but has multiple (6?) DPU engines and is optimized for batched inference. However, I am measuring ~200 FPS for resnet50_pt for 6E300 with `-t 1`, while I measure only 75 FPS for 1E333 with `-t 1`. Other networks seem to behave similarly as well with 6E300 still being faster even with `-t 1`. This raises the question as to what exactly is the point of the 1E333 variant (apart from supporting depth-wise convolution) since it doesn't actually seem to have better latency than 6E300 either? Maybe I am interpreting `-t 1` incorrectly and it doesn't actually mean batch = 1?

qianglin-xlnx, 2021-01-25:
@zohourih 
`-t 1` doesn't actually mean batch =1 for 6E300 DPU. 
When you run the `test_performance_classification` for U50 with 6E300M DPU, the batch will be 3. So the latency should be 1000ms/(200/3)= 15ms.

zohourih, 2021-01-25:
@qianglin-xlnx Okay, thank you for the clarification. Is there any way to test batch-1 performance on 6E300 DPU with `test_performance_classification`?

qianglin-xlnx, 2021-02-02:
> 
> 
> @qianglin-xlnx Okay, thank you for the clarification. Is there any way to test batch-1 performance on 6E300 DPU with `test_performance_classification`?

Since Vitis-AI-Library is fully open source, you can modify the code to test batch-1 performance on 6E300 DPU if you need.
You can modify the following code to set batch = 1, and test it again.
https://github.com/Xilinx/Vitis-AI/blob/master/tools/Vitis-AI-Library/benchmark/include/vitis/ai/benchmark.hpp#L70

zohourih, 2021-02-05:
Great, thank you.

closed 2021-02-05 by zohourih