Xilinx/Vitis-AI #276: Fingerprint check failure on ZCU102

sanjeewk, 2021-01-25:
We're trying to test the the PointPillars model on the ZCU102 board, but ran into this error.

```
W0703 07:40:06.588299  2318 dpu_runner_base_imp.cpp:692] CHECK fingerprint fail ! model_fingerprint 0x1000020f6014407 dpu_fingerprint 0x1000000f6014407 
F0703 07:40:06.589164  2318 dpu_runner_base_imp.cpp:664] fingerprint check failure.

```

Model links were obtained from link for ZCU102 from here:
https://github.com/Xilinx/Vitis-AI/blob/master/models/AI-Model-Zoo/model-list/pt_pointpillars_kitti_12000_100_10.8G_1.3/model.yaml

The model was copied onto the board, and the files from https://github.com/Xilinx/Vitis-AI/tree/master/demo/Vitis-AI-Library/samples/pointpillars were copied onto the board and then compiled with no errors. 

Running xilinx_show_devices gives the following result
```
I0702 09:03:19.701958  1972 test_xrt_device_handle.cpp:24] cu[0]: device_id 0 full_name DPUCZDX8G:DPUCZDX8G_1 kernel_name DPUCZDX8G instance_name DPUCZDX8G_1 fingerprint 0x1000000f6014407 cu_handle 0x55c39b3e70 cu_mask 2 cu_addr 0x80040000 
I0702 09:03:19.701997  1972 test_xrt_device_handle.cpp:24] cu[1]: device_id 0 full_name DPUCZDX8G:DPUCZDX8G_2 kernel_name DPUCZDX8G instance_name DPUCZDX8G_2 fingerprint 0x1000000f6014407 cu_handle 0x55c39ddc80 cu_mask 1 cu_addr 0x80050000 
I0702 09:03:19.702054  1972 test_xrt_device_handle.cpp:24] cu[2]: device_id 0 full_name DPUCZDX8G:DPUCZDX8G_3 kernel_name DPUCZDX8G instance_name DPUCZDX8G_3 fingerprint 0x1000000f6014407 cu_handle 0x55c39bdba0 cu_mask 4 cu_addr 0x80060000 
I0702 09:03:19.702083  1972 test_xrt_device_handle.cpp:24] cu[3]: device_id 0 full_name sfm_xrt_top:sfm_xrt_top_1 kernel_name sfm_xrt_top instance_name sfm_xrt_top_1 fingerprint 0x0 cu_handle 0x55c39d92e0 cu_mask 8 cu_addr 0x80070000
```

What are possible solutions to this? Just starting off with the ZCU102 so a bit confused about dpu fingerprints

qianglin-xlnx, 2021-01-25:
@sanjeewk 
1 Do you use the official vai1.3 zcu102 system image? If not, you can download the image from https://www.xilinx.com/bin/public/openDownload?filename=xilinx-zcu102-dpu-v2020.2-v1.3.0.img.gz 
In this case, you can test the PointPillars sample with the model we described in https://github.com/Xilinx/Vitis-AI/blob/master/models/AI-Model-Zoo/model-list/pt_pointpillars_kitti_12000_100_10.8G_1.3/model.yaml 

2 If you integrate a different DPU in your custom system image, you have to recompile the model with the new arch.json generated by new DPU.  Then the fingerprints mismatch problem does not occur.  
https://github.com/Xilinx/Vitis-AI/tree/master/dsa/DPU-TRD/prj/Vitis#522-get-json-file

In a word, the fingerprint of the model and the fingerprint of the DPU should match.  If not, change either of them.

sanjeewk, 2021-01-25:
We're using the official image, for vitis v1.2. However the model from the given link is not working, it's giving the fingerprint check failure.

 We attempted to recompile the model using the xmodel files in https://www.xilinx.com/bin/public/openDownload?filename=pt_pointpillars_kitti_12000_100_10.8G_1.3.zip and the docker image. We then used the following command to convert to the dpu given by using vai_c_xir

```
vai_c_xir -i VoxelNet_0_int.xmodel -a /opt/vitis_ai/compiler/arch/DPUCZDX8G/arch.json -o . -n PointPillar
```
But end up with the same error

qianglin-xlnx, 2021-01-25:
Hi @sanjeewk 
The VAI1.2 zcu102 image does not support Pointpillars model. Please use the official vai1.3 zcu102 system image and try it again.

sanjeewk, 2021-01-26:
@qianglin-xlnx Yep that fixed it. Thanks!

closed 2021-01-26 by sanjeewk