Xilinx/Vitis-AI #279: XCOM_OPERATION_FAILED - PyTorch model compilation crashes

fenyesb, 2021-01-26:
I have a model based on the Inception_v3 network with a few layers added or changed. The model runs fine in PyTorch CUDA. I can quantize it, but I can't compile it using the Vitis AI toolchain. The arch.json is for the pre-built ZCU102 image.

```
(vitis-ai-pytorch) Vitis-AI /workspace/psr-ui > vai_c_xir -x q/Inception3_int.xmodel -a arch.json -o c -n Inception3_int
**************************************************
* VITIS_AI Compilation - Xilinx Inc.
**************************************************
[UNILOG][INFO] The compiler log will be dumped at "/tmp/vitis-ai-user/log/xcompiler-20210126-111943-153"
[UNILOG][INFO] Target architecture: DPUCZDX8G_ISA0_B4096_MAX_BG2
[UNILOG][INFO] Compile mode: dpu
[UNILOG][INFO] Debug mode: function
[UNILOG][INFO] Target architecture: DPUCZDX8G_ISA0_B4096_MAX_BG2
[UNILOG][INFO] Graph name: Inception3, with op num: 781
[UNILOG][INFO] Begin to compile...
[UNILOG][WARNING] xir::Op{name = Inception3__Inception3_BasicConv2d_padding_conv_20__Conv2d_conv__input_128, type = conv2d-fix} has been assigned to CPU: [DPU only supports "pad_left"(1) less than "kernel_width"(1)].
[UNILOG][FATAL][XCOM_OPERATION_FAILED][The supposed operation is failed!] Failed to add smf node to bank!
*** Check failure stack trace: ***
Aborted (core dumped)
```

Also why does it say that the padding is too large compared to the kernel size, but it says here that this value should be supported: [Supported OPs and DPU Limitations](https://china.xilinx.com/html_docs/vitis_ai/1_3/compiling_model.html#lah1605503013889__section_a3t_f2y_ykb)

I attached the quantizer's outputs: quant_info.json and Inception3.py
[inception3_quant.zip](https://github.com/Xilinx/Vitis-AI/files/5873081/inception3_quant.zip)

fenyesb, 2021-01-27:
After replacing the convolutional layer which gave the warning with a deconvolutional layer, that issue was solved. But I still get a fatal error when compiling:

```
(vitis-ai-pytorch) Vitis-AI /workspace/psr-ui > vai_c_xir -x q/Inception3_int.xmodel -a arch.json -o c -n Inception3_int
**************************************************
* VITIS_AI Compilation - Xilinx Inc.
**************************************************
[UNILOG][INFO] The compiler log will be dumped at "/tmp/vitis-ai-user/log/xcompiler-20210127-082044-292"
[UNILOG][INFO] Target architecture: DPUCZDX8G_ISA0_B4096_MAX_BG2
[UNILOG][INFO] Compile mode: dpu
[UNILOG][INFO] Debug mode: function
[UNILOG][INFO] Target architecture: DPUCZDX8G_ISA0_B4096_MAX_BG2
[UNILOG][INFO] Graph name: Inception3, with op num: 781
[UNILOG][INFO] Begin to compile...
module_infer = 0, counter_m[module_infer] = 27, counter_p[module_infer] = 27
module_idx = 0, counter_m[module_idx] = 27, counter_p[module_idx] = 27
module_idx = 1, counter_m[module_idx] = 0, counter_p[module_idx] = 0
module_idx = 2, counter_m[module_idx] = 25, counter_p[module_idx] = 25
module_idx = 3, counter_m[module_idx] = 0, counter_p[module_idx] = 0
[UNILOG][FATAL][XCOM_PM_FAIL][The compiler occurs an error when generating instructions, please contact us.]
*** Check failure stack trace: ***
Aborted (core dumped)
```

I attached the quantizer's outputs: quant_info.json and Inception3.py: [inception3_quant.zip](https://github.com/Xilinx/Vitis-AI/files/5878665/inception3_quant.zip)

wanghong4compiler, 2022-03-02:
Hi @fenyesb ，
could you also provide the test script?

qianglin-xlnx, 2022-03-10:
Hi @fenyesb 
Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-03-10 by qianglin-xlnx