Xilinx/Vitis-AI #283: Why did the inference time jump when I was testing the model.

Songxinlei, 2021-02-01:
Why did the inference time jump when I was testing the model. Sometimes it can even be ten times worse. Why?

qianglin-xlnx, 2021-02-01:
Hi @Songxinlei  
Could you describe the details of the test you did? Thank you.

Songxinlei, 2021-02-01:
Hi @qianglin-xlnx ,
I tested it under ZCU102, vitis-ai 1.3. The modified code is VART/vitis-ai-library/samples/yolov3, the code is changed to test video, and it is single-threaded. It is found that the inference time jumps from 1.7ms to 20ms. I am very puzzled why this process time is unstable

yongshengxu-xlnx, 2021-02-01:
Hi @Songxinlei ,
Does inference time include only post-processing time or the time including reading video and display.
If it’s just post-processing time. In order to reproduce the problem, please provide pictures that takes 20ms, and the name of the yolov3 xmodel.
If the time includes reading video and displaying, the longer time is due to the reading and displaying of the video

Songxinlei, 2021-02-01:
hi  @yongshengxu-xlnx
The inference time only includes the code of "task ->run(0u);". The time for pre-treatment and post-treatment is not included.
And the time is not include reading video and displaying.
so , I confused .

yongshengxu-xlnx, 2021-02-01:
Hi, @Songxinlei 
Which model are you using? Our test program uses the yolov3_voc model by default. Please refer to the readme for the specific path and how to use it. Our yolov3_voc dpu time data: 65.4ms.  Please note that the correct model is used and the correct result is obtained.

Songxinlei, 2021-02-01:
hi @yongshengxu-xlnx
I modified the sample code of yolov3, and the model used is my own training model. And it ran out without any problems. It's just that my inference time keeps jumping. I am very confused about this, why use adasdetection.webm to test, the run time of each frame is not the same?
你好，我是修改了那个VART/vitis-ai-library/samples/yolov3的示例代码，并且使用的模型是自己的训练模型。跑出来是没有任何问题的。只是我打印出来看inference time  在不断的跳跃，FPS反复跳，不稳定。我对此很疑惑，为什么利用adasdetection.webm 视频来测试，每一帧的run time 还不一样呢。这个时间只针对 task->run  这句代码的时间。我找不到具体的原因是为何

yongshengxu-xlnx, 2021-02-02:
Hi @Songxinlei 
In order to reproduce the problem, is it convenient to provide us with the model?

Songxinlei, 2021-02-04:
不好意思 这个不太方便

yongshengxu-xlnx, 2021-02-04:
Hi @Songxinlei 
We cannot debug and analyze the cause of the problem without model.
If there is progress, discovery or other needs, please feel free to ask.
Thinks.

qianglin-xlnx, 2022-02-18:
Hi @Songxinlei 
Since we haven't received your reply for a long time, I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-02-18 by qianglin-xlnx