Xilinx/Vitis-AI #284: Is it possible to change the weights of network in xir.Graph?

dprzewlocka, 2021-02-02:
Let's say I deserialized the model using xir.Graph.deserialize(path_to_model) and I would like overwrite some weights (for example for the selected convolutional layer) - is it possible?

yongshengxu-xlnx, 2021-02-07:
It's not possible.  Weights cannot be modified after the model is compiled.

xingy16, 2021-05-10:
Actually, it is possible. I made an experiment on it.
`import numpy as np
import xir

g = xir.Graph("test")
g = g.deserialize("test.xmodel")

weights_op = g.get_op("weights")

origin_w = np.frombuffer(weights_op.get_attr("data"), np.float32)
print(origin_w.flatten()[0:5]) // [-36. -66. -95.  -8.  15.]

weights_op.set_attr("data", (origin_w*2).tobytes())
new_w = np.frombuffer(weights_op.get_attr("data"), np.float32)
print(new_w.flatten()[0:5]) // [ -72. -132. -190.  -16.   30.]`

I have an xmodel, which contains a weights op called "weights", it is a 'const' operation.
First i deserialize the xmodel and get this weights op, when i get the data in it, i got [-36. -66. -95.  -8.  15.].
Then i multiply the data by 2 and set the data back to this 'const' operation, the data became [ -72. -132. -190.  -16.   30.]

But you need to pay attention to the data type, get_attr("data") returns bytes, you need to use np.frombuffer to transform bytes into float32, int8 or something else, the datatype is specified by another attibute named "data_type" in const op. When you set_attr("data", data), the data should be in bytes as well.

yongshengxu-xlnx, 2022-02-22:
Hi @dprzewlocka 
Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-02-22 by qianglin-xlnx