Xilinx/Vitis-AI #288: Vitis AI Run TIme on ZCU102 error (libunilog)

dduartefernandes, 2021-02-05:
I have been trying to install the Vitis AI on ZYNQ ZCU102, however and i try to install the Vitis Run Time (following the instructions in Vitis AI User Guide) I get an error relating to the installation of the rpm files.

I adopted the the official linux system image, but changed the rootfs to the Ubuntu 18.04.5 (as I want to receive point cloud data from ROS), and when I execute alien –i libunilog-1.3.0-r*.aarch64.rpm or rpm -ivh --force libunilog-1.3.0-r*.aarch64.rpm I got an error message stating the the libunilog is just for aarch64. My uname -a result is  "aarch64 aarch64 aarch64 GNU/Linux".

Thanks in advance for your help.

qianglin-xlnx, 2021-02-06:
Hi @dduartefernandes 
For the different system that you need to port, it's better to recompile VART from source code to address potential dependencies and other issues.
Back to your question, when you install the RPM package, it will read the system's architecture name and refuses to install it if the name don't match.
If you still want to install the VART packages, you can unzip them from the host and copy the libraries to the corresponding path.
Following is the command to unzip the rpm package.

********:temp% rpm2cpio libunilog-1.3.0-r422.aarch64.rpm | cpio -idmv
./usr/include/UniLog
./usr/include/UniLog/ErrorCode.hpp
./usr/include/UniLog/UniLog.hpp
./usr/lib/libunilog.so
./usr/lib/libunilog.so.1
./usr/lib/libunilog.so.1.3.0
./usr/share/cmake
./usr/share/cmake/unilog
./usr/share/cmake/unilog/unilog-config-version.cmake
./usr/share/cmake/unilog/unilog-config.cmake
./usr/share/cmake/unilog/unilog-targets-release.cmake
./usr/share/cmake/unilog/unilog-targets.cmake

andyluo7, 2021-02-26:
@dduartefernandes, can we close this issue?

qianglin-xlnx, 2022-02-17:
Hi @dduartefernandes 
Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-02-17 by qianglin-xlnx