Xilinx/Vitis-AI #289: want help with compilation on DPUCADF8H for Alveo u200

daivikbhatia, 2021-02-09:
Hi, Can anyone share path to compilation examples using DPUCADF8H for alveo U200?

wilderfield, 2021-02-09:
https://github.com/Xilinx/Vitis-AI/tree/master/examples/DPUCADF8H/tf_resnet

daivikbhatia, 2021-02-16:
Hi, thank-you so much for the example link.
Unfortunately, while running the compiled file, i.e resnet50_v1_tf_compiled.xmodel,
 via command- ./resnet_example ./resnet50_v1_tf_compiled.xmodel ./images/,
I'm getting the following error Error: xclbinPath is not set, please consider setting XLNX_VART_FIRMWARE.
I have already run the overlaybins setup as told in the example, Still I'm getting this error, can someone please help me out?

liuhongduosc, 2021-04-28:
Hi, @daivikbhatia 

I met the same problem, do you know how to solve it now?

daivikbhatia, 2021-04-28:
Hi @liuhongduosc 
For me, reinstalling the butler to one version back solved the problem.
the command for reinstalling is given in readme file of this link, 
https://github.com/Xilinx/Vitis-AI/tree/master/setup/alveo/u200_u250/packages

liuhongduosc, 2021-04-29:
Hi @daivikbhatia 
Thanks for your suggestion, I tried your solution, but it did not work. For me,
 ```sudo service xbutler Restart```
 could work.

qianglin-xlnx, 2022-08-23:
Closing since no activity for more than 1 year, please reopen if you still have question, thanks

closed 2022-08-23 by qianglin-xlnx