Xilinx/Vitis-AI #290: ONNX runtime support for MPSoC devices

zackdilan, 2021-02-09:
Hello all, 
[https://github.com/Xilinx/Vitis-AI/tree/master/external/onnxruntime](url), Here, as you have mentioned, the Vitis ai execution provider in ONNX runtime supports DPUCADX8G(u200/250) devices. 

So my question is, do we need to wait for future releases for targeting other DPU architectures or is it possible to do the same on a user level. If that's the case, it would be really helpful, if someone can provide me some resources.

Thanks in Advance.

jtuyls, 2021-02-26:
Hi @zackdilan , we currently have an open PR on the ONNX Runtime repo to add support for other DPU's: https://github.com/microsoft/onnxruntime/pull/6690. As stated in the PR, it's targeted at adding support for DPUCZDX8G (Zynq devices), but as the PR allows passing the DPU identifier as a provider option to the ONNX Runtime inference session, it will make it more easy to add support for other DPU's too. For example, DPUCAHX8H (u50, u280) support is upcoming. Did you have any specific DPU in mind?

zackdilan, 2021-02-26:
Thanks @jtuyls . No I was actually looking for implementations for DPUCZDX8G (Zynq devices). When we can expect this?

jtuyls, 2021-02-26:
@zackdilan Ah, I missed the MPSoC in the title. The DPUCZDX8G target will be enabled when the PR gets merged. The timeline depends on when ONNX Runtime people can review and approve it. If you want to try it out you can use the PR though. Documentation is work in progress but I will do a separate PR for that shortly.

zackdilan, 2021-02-26:
@jtuyls thanks again ... I will keep an eye on the PR then. So in the meantime, I will keep this issue open for few weeks, I m not sure when I have time to explore it.  let's say 2 weeks, otherwise, we can close this issue.

Elonaever, 2021-03-17:
Also want to use onnxruntime on the ZCU106 with dpu.
And could the onnxruntime will be supported in the future device such as AIE board?

TUT-jiayou, 2021-09-09:
Hi,
About the soc Zynq UltraScale+ MPSoCs EV XCZU4EV-1SFVC784I supporting for onnx?

hossam-snke, 2021-09-16:
Hi,
it is written in documentation that  DPUCZDX8G (Zynq) is supported for onnxruntime. But I can not find any installation instructions
Will you have any updates soon?
Thanks

jeinstei, 2022-01-05:
I'm seeing some interesting notes in the PR as well, as the PR references the vitis-ai docker (x86), but also adds the cross-compilation tools. What it does not seem to have is a way actually actually push and RUN the model on the edge device, even if the provider field now supports this. I'm not sure how onnxruntime would work on a Zynq device without some other layer in there, even if it does support building and optimizing a model for the DPU

closed 2022-03-03 by qianglin-xlnx

jeinstei, 2022-03-03:
I'm sorry, but why was this issue closed with no comment? Does Xilinx have a statement on this at all? I'd love to know if an ONNXruntime with an xrt backend will be available for Zynq DPUs, and the toolflow is definitely tied to Vitis-AI,.

If there was a recent push to Vitis-AI, can you at least reference the commit?