Xilinx/Vitis-AI #291: AssertionError: 'bias' should not be None.

hill19850213, 2021-02-09:
Hi,
I quantized our custom resent50 model, but it had big accuracy loss when doing evaluation. Therefore, I referred to the user guide to do Quantize Finetuning and the final model had a little accuracy loss. However, there was an error during model compiling. The error message is as below:
```
 **************************************************
* VITIS_AI Compilation - Xilinx Inc.
**************************************************
[INFO] Namespace(inputs_shape=None, layout='NHWC', model_files=['./compal_model/quantized/quantized.h5'], model_type='tensorflow2', out_filename='./compal_model/vai_c_output_ZCU102//compal_tf2_resnet50_org.xmodel', proto=None)
[INFO] tensorflow2 model: compal_model/quantized/quantized.h5
/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/xnnc/translator/tensorflow_translator.py:1752: H5pyDeprecationWarning: dataset.value has been deprecated. Use dataset[()] instead.
  value = param.get(group).get(ds).value
[INFO] parse raw model     : 99%|▉| 124/125 [00:00<00:00, 45718.50it/s]         
Traceback (most recent call last):
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/bin/xnnc-run", line 33, in <module>
    sys.exit(load_entry_point('xnnc==1.3.0', 'console_scripts', 'xnnc-run')())
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/xnnc/__main__.py", line 194, in main
    normal_run(args)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/xnnc/__main__.py", line 178, in normal_run
    in_shapes=in_shapes if len(in_shapes) > 0 else None,
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/xnnc/xconverter.py", line 131, in run
    xmodel = CORE.make_xmodel(model_files, model_type, _layout, in_shapes)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/xnnc/core.py", line 104, in make_xmodel
    model_files, layout, in_shapes=in_shapes, model_type=model_t
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/xnnc/translator/tensorflow_translator.py", line 97, in to_xmodel
    model_name, raw_nodes, layout, in_shapes, model_fmt, model_type
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/xnnc/translator/tensorflow_translator.py", line 163, in create_xmodel
    xmodel = cls.__create_xmodel_from_tf2(name, layers, layout, in_shapes)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/xnnc/translator/tensorflow_translator.py", line 1121, in __create_xmodel_from_tf2
    xnode.bias = param.get(wrapper_name).get("bias:0")
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/xnnc/ir/xnode.py", line 2974, in bias
    assert bias is not None, "'bias' should not be None."
AssertionError: 'bias' should not be None.
```
How to solve it?

Thanks

qianglin-xlnx, 2021-02-24:
Hi @hill19850213 
Thank you for your feedback. It probably be a bug. 
Could you please share the quantized model so that we can test it and find a work round for it. Thank you.

hill19850213, 2021-02-24:
Hi @qianglin-xlnx ,

Can I send private message to you?

Thanks
Sitting

qianglin-xlnx, 2021-02-24:
> 
> 
> Hi @qianglin-xlnx ,
> 
> Can I send private message to you?
> 
> Thanks
> Sitting

Sure. Your email is?

hill19850213, 2021-02-24:
Hi @qianglin-xlnx ,
My mail is hill19850213@gmail.com

Thanks
Sitting

foolmarks, 2021-02-24:
I've already reported this error internally - its scheduled to be fixed in 1.4

hill19850213, 2021-02-24:
Hi @foolmarks ,

When do Xilinx release vitis ai 1.4?

Hi @qianglin-xlnx ,
I hope you can find the workaround.

closed 2021-03-04 by hill19850213

zackdilan, 2021-03-23:
Hello, Sorry for reopening the issue!!
Since we are not sure about Vitis 1.4 release, is there any workaround to solve this problem!

Thanks in advance

foolmarks, 2021-03-26:
@zackdilan 

I believe something is in the works - pls ping me again mid next week.