Xilinx/Vitis-AI #293: tensorflow2 xmodel compilation error

waldenou, 2021-02-15:
Hi,
I was trying to compile the xmodel for resnet50, but the following error was encountered during model compilation.

```
(vitis-ai-tensorflow2) Vitis-AI /workspace/test/tf2 > vai_c_tensorflow2 -m resnet50.h5 -a /opt/vitis_ai/compiler/arch/DPUCADF8H/U250/arch.json -o compiled_dir -n resnet50
**************************************************
* VITIS_AI Compilation - Xilinx Inc.
**************************************************
[INFO] Namespace(inputs_shape=None, layout='NHWC', model_files=['resnet50.h5'], model_type='tensorflow2', out_filename='compiled_dir/resnet50_org.xmodel', proto=None)
[INFO] tensorflow2 model: resnet50.h5
/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/xnnc/translator/tensorflow_translator.py:1752: H5pyDeprecationWarning: dataset.value has been deprecated. Use dataset[()] instead.
  value = param.get(group).get(ds).value
[INFO] parse raw model     :100%|████████| 125/125 [00:00<00:00, 20254.51it/s]             
[INFO] infer shape (NHWC)  :100%|████████| 202/202 [00:00<00:00, 6135.31it/s]              
[OPT] No optimization method available for xir-level optimization.
[INFO] generate xmodel     :100%|████████| 202/202 [00:00<00:00, 642.19it/s]               
[INFO] generate xmodel: /workspace/test/tf2/compiled_dir/resnet50_org.xmodel
[UNILOG][INFO] The compiler log will be dumped at "/tmp/vitis-ai-user/log/xcompiler-20210214-163115-203"
[UNILOG][INFO] Compiling subgraph : subgraph_quant_avg_pool
terminate called after throwing an instance of 'std::runtime_error'
  what():  Error during initialization: multiple incompatible definitions with name "argv"
Aborted (core dumped)
```
The quantized model is generated using the code snippet below:
```
model = tf.keras.applications.ResNet50(include_top=True, weights='imagenet', classes=1000)
tf.keras.backend.set_learning_phase(0)
quantizer = vitis_quantize.VitisQuantizer(model) 
data_dir = "/home/vitis-ai-user/CK-TOOLS/dataset-imagenet-ilsvrc2012-val-min"
listimage = os.listdir(data_dir)
img = []
runTotal = len(listimage)
for i in range(runTotal):
    path = os.path.join(data_dir, listimage[i])
    if path[-4:] == 'JPEG':
        img.append(input_fn.preprocess_fn(path))

img = np.reshape(img, (len(img), 224,224,3))

quantized_model = quantizer.quantize_model(calib_dataset=img)      
quantized_model.save('resnet50.h5')
```
Please let me know if any additional information is needed. Any help is appreciated.

qianglin-xlnx, 2021-02-26:
Hi @waldenou 
I've already reported this error internally. 
At the same time, could you try to compiled it for the other DPU target, such as U50.  Run the following command. If it compiled successfully, then it means the model is ok, just that it not supported on U250 target.
vai_c_tensorflow2 -m resnet50.h5 -a /opt/vitis_ai/compiler/arch/DPUCAHX8H/U50/arch.json -o compiled_dir -n resnet50

waldenou, 2021-02-27:
Hi @qianglin-xlnx,

Thanks for your reply. I tried to compiled it for U50 DPU target, and it did compile successfully.

rubende, 2021-05-04:
Hi,

I am having the same problem, is there any update on its fixing?

@waldenou, have you tried it with other models?

qianglin-xlnx, 2022-03-28:
Hi @wilderfield 
Any suggestion on this? Is tf2 model supported by U250 in VAI2.0?

paolodalberto, 2022-08-13:
tf2 should be supported ... the only layer that is not completely supported is avgpool when there are paddings ... but resnet 50

please, could you share the h5  (tensorflow2) model ?