Xilinx/Vitis-AI #297: Hourglass model training

d01trhs, 2021-02-17:
Hello,

Hourglass model (cf_hourglass_pe_mpii_256_256_10.2 G_1.3.zip) added in Vitis AI v1.3 did not have the files needed for training.
Where can I get trainval.prototxt and solver.prototxt for training Hourglass models?

qianglin-xlnx, 2021-02-22:
Hi @d01trhs 
As far as I know, this model is converted using other framework training model, so there is no training code for you to train it in Caffe framework.

d01trhs, 2021-02-23:
@qianglin-xlnx Thanks for your reply.
In what framework is this Hourglass model trained?
README listed https://github.com/bearpaw/pytorch-pose as a reference. Did you convert from this model?
I would like to try it from the training of the hourglass model, could you tell me how to convert it?

qianglin-xlnx, 2021-02-24:
> 
> 
> @qianglin-xlnx Thanks for your reply.
> In what framework is this Hourglass model trained?
> README listed https://github.com/bearpaw/pytorch-pose as a reference. Did you convert from this model?
> I would like to try it from the training of the hourglass model, could you tell me how to convert it?

Yes. The origin model is from Pytorch framework, as you mentioned. For the pytorch2caffe tool, it's not public released. You can contact local Xilinx FAE to get more resource. Thank you.

d01trhs, 2021-03-01:
Hi @qianglin-xlnx 
Thank you for your reply.
One thing I would like to ask is that while Vitis AI v1.3 supports Pytorch, is there any reason to convert Pytorch's Hourglass model to the Caffe model?

qianglin-xlnx, 2021-03-01:
Hi @d01trhs 
Although this model was released in VAI1.3, we deployed this model very early and the pytorch related tool was not mature enough at that time. 
Currently, I don't think there is reason to convert Pytorch's Hourglass model to the Caffe model.  As for whether we will release support for the Pytorch's Hourglass model, it depends on the market requirement. Thank you again for your feedback.

qianglin-xlnx, 2022-02-20:
Hi @d01trhs 
Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-02-20 by qianglin-xlnx