Xilinx/Vitis-AI #298: Segmentation fault when running tiny_yolov3_416 with DpuTask (Vitis-AI 1.3)

dextroza, 2021-02-18:
Hi,

I get segmentation fault  in 'yolov3_post_process' function when I try to run tiny_yolov3_416 for 2 classes (compiled in docker vitis-ai 1.3.411)

I use this script 'tools/Vitis-AI-Library/overview/demo/yolov3/demo_yolov3.cpp' for running model on zcu104 board with zcu104-2020.2 image.

Here is the tiny_yolov3_416 postprocessing params:

const string yolov3_config  =
{
  "   name: \"tiny_yolov3_416\" \n"
  "   model_type : YOLOv3 \n"
  "   yolo_v3_param { \n"
  "     num_classes: 2 \n"
  "     anchorCnt: 3 \n"
  "     conf_threshold: 0.3 \n"
  "     nms_threshold: 0.45 \n"
  "     layer_name: \"15\" \n"
  "     layer_name: \"22\" \n"
  "     biases: 10 \n"
  "     biases: 14 \n"
  "     biases: 23 \n"
  "     biases: 27 \n"
  "     biases: 37 \n"
  "     biases: 58 \n"
  "     biases: 81 \n"
  "     biases: 82 \n"
  "     biases: 135 \n"
  "     biases: 169 \n"
  "     biases: 344 \n"
  "     biases: 319 \n"
  "     test_mAP: false \n"
  "   } \n"
};

I found out that 'task->getOutputTensor(0u);' returns 'layer8-conv_fixed' and 'layer10-act_fix' which seem to be wrong. I expected 'layer-15' and 'layer-22'.

Any help would be appreciated.

Regards,
dextroza

jimheaton, 2021-02-18:
I have seen something like this before, when the layer_names in the post processing params do not match what's in the model.

Double check the outputs of your model. 

If you were using tensorflow you can run the via_inspect command the list model outputs. If you used Caffe you can check the prototxt file. 
The output layers should also match the output layers you specified when you ran the quantizer.

dextroza, 2021-02-18:
Thak you for your reply. 
I use caffe version of tiny_yolov3_416.
I double checked layer names and I got 'layer-8' and 'layer 12' for subgraph[0], and 'layer-15', 'layer-22' for subgraph[1].

I think I need something like this:
1. auto task = DpuTask::create(kernelName); // creates both subgraphs
2. task->run(0u); // runs subgraph[0]
3. outputTensors_0 = task->getOutputTensor(0u) // gets outputs layer8, layer12
4. **// Take outputTensors_0 and set them as input to subgraph[1]?**
5. task->run(1u); // runs subgraph[1] with outputTensors_0 as input
6. task->getOutputTensor(1u); // gets outputs layer15, layer22
7. run yolov3_post_process with the layer15, layer22

I need help with the 4. step. I am not sure if this is even possible with DpuTask.

Thanks!

dextroza, 2021-03-23:
Hi,
has anyone tried to compile and run tiny_yolov3 on Vitis AI 1.3? I still haven't found solution.

Thanks!

qianglin-xlnx, 2022-02-23:
Hi @dextroza 
Has your issue been solved?

dextroza, 2022-02-23:
Hi @qianglin-xlnx,

yes, it has been solved, but I don't remember how.

qianglin-xlnx, 2022-02-23:
@dextroza 
Thank you. I will close this issue. If you have any other questions, please open a new issue.

closed 2022-02-23 by qianglin-xlnx