Xilinx/Vitis-AI #299: VART API for the softmax core and DNNDK_XRT error in using DPU core

JerrySciangula, 2021-02-18:
Hi everyone,
I'm a master graduating student at Scuola Superiore Sant'Anna in Pisa.

For my thesis, I'm trying to accelerate a custom network model on a ZCU102 ultrascale+ using the new Vitis-AI tools v1.3. Initially I accelerated the network using VART APIs, without any issues, but I had to implement a softmax layer in software. Since it is costly in terms of execution time, I would like to use the Softmax IP core to accelerate its operations. Since I couldn't find any API for the new VART APIs concerning the usage of the Softmax core, I decided to take a look to the old Legacy DNNDK interface.

As indicated in the guide, there exists the **_dpuRunSoftmax()_** function of the **n2cube** library of the Legacy DNNDK. Therefore, I firstly tried to run the DNNDK demo examples of _ResNet50_ in https://github.com/Xilinx/Vitis-AI/tree/master/demo/DNNDK/resnet50. I can successfully compile the application, using the related SDK, but when I try to run the application an error is arised, reporting:

**_[DNNDK_XRT] Cannot find device, index: 0_**.

It seems like the board or the runtime environment cannot recognize the DPU peripheral. 

What does this error specifically mean?
How can I resolve this problem?
Is there any similar VART API to that of DNNDK that enable me to use the softmax core?

qianglin-xlnx, 2021-02-22:
Hi @JerrySciangula 
Do you use the official VAI1.3 ZCU102 system image? If yes, you can directly compile the demo/DNNDK/resnet50 on the target and run it successfully.
For other interface of softmax, you can use Vitis-AI-Library xnnpp interface to use the softmax. Please refer to the following code.
https://github.com/Xilinx/Vitis-AI/blob/master/tools/Vitis-AI-Library/math/src/softmax.cpp
https://github.com/Xilinx/Vitis-AI/blob/master/tools/Vitis-AI-Library/xnnpp/src/classification/classification.cpp#L94

JerrySciangula, 2021-02-22:
Hi @qianglin-xlnx, yes I'm using the official VAI 1.3 ZCU102 system image. I just can successfully compile the demo/DNNDK/resnet50 both on the target and on my machine thorugh the use of the sdk, but i can't run it on the board because of the error reported in this issue. What does the **_[DNNDK_XRT] Cannot find device, index: 0_** error mean?

Anyway, thank you for the other softmax interfaces!

wanghy-xlnx, 2021-02-23:
Hi @JerrySciangula , **_[DNNDK_XRT] Cannot find device, index: 0_** means that the runtime failed to correctly read the value of the DPU registers. Therefore, do you flow [README.md](https://github.com/Xilinx/Vitis-AI/blob/master/demo/DNNDK/README.md) operation? And used xilinx-zcu102-dpu-v2020.2-v1.3.0.img.gz, installed vitis-ai_v1.3_dnndk.tar.gz on the target board, compiled resnet50 use **_./build zcu102_** on the target board, finally run the **_./resnet50_**. If this issue still exists, please provide the information printed by the following command:

1. uname -a
2. dexplorer -v

JerrySciangula, 2021-03-02:
Hi @wanghy-xlnx, thank you for your reply. I tried again with the official VAI 1.3 image and the example runs successfully. But I calculated the execution time of the hardware softmax and it is quite similar to the software one. Having no improving performance I think the new n2cube library masks the hardware API for the softmax. Am I right?

Therefore, in order to use the softmax hardware core, I tried the other interface of softmax, as suggested by @qianglin-xlnx, but an other error is arised when calling the run() method:

**_F0302 13:37:36.878149  1160 sfm_control_xrt_edge.cpp:73] Check failed: supported(scale, cls, group) scale=1 cls=1000 group=1 num_of_cus: 1 ENV_PARAM(XLNX_ENABLE_HW_SMFC) = 1_** 

Do you have any idea on what is happening? Could you help me?

wanghy-xlnx, 2021-03-03:
Hi @JerrySciangula , the hardware softmax is not masked. You can observe the number of interrupts through the **_cat /proc/interrupts | grep cu_** command, which will print the number of interrupts for 3 DPU IPs and 1 Softmax IP. Or use the **_dexplorer -s_** command to observe whether SM_LEN_X, SRC and other registers in the SMFC Registers are non-zero after the sample is executed. In addition, can you give me the performance data result of your test? So we can confirm it from our side.

JerrySciangula, 2021-03-04:
Hi @wanghy-xlnx, I just tried the DNNDK **_resnet50_** demo with the **_dpuRunSoftmax()_** function. Using the command **_dexplorer -s_**, we can actually observe that the registers are set and have non-zero values after the execution. However, the hardware softmax shows more or less the same execution time of the software one, as you can see from this picture:
[image]

Here, I report the code:
[main.txt](https://github.com/Xilinx/Vitis-AI/files/6083156/main.txt)

In previous version images of the board, such as the **_DNNDK of the 2019_**, we tried the same **_resnet50_** sample and the hardware softmax had **_30x improvement_** with respect to the software one. Therefore, I cannot understand why in **_Vitis AI 1.3_** the performances between them are so similar.

Moreover, with a Petalinux images containing custom hardware, we found that no instruction are triggered into the softmax core 
master AXI port.

Could you help me understand what is happening?

wanghy-xlnx, 2021-03-08:
Hi @JerrySciangula , 
According to confirmation, the Softmax IP can calculate 4 floating-point numbers in 1 clock cycle. When the clock frequency is 300MHz, the calculation time for 4 floating-point numbers is 3.3 ns. In this way, the theoretical time of softmax API in resnet50 is (1000 * 1) / 4 * 3.3ns = 825ns = 0.825us.
However, the actual API execution time also includes read/write registers time, startup time, memory bandwidth competition, etc. Therefore, if the amount of calculated data is too small, there will be a situation where the hardware performance is not as good as the software performance. For example, in this Vitis AI 1.3 flow, there are more registers to read and write than in traditional Vivado flow, and each operation takes longer time.
In addition, there are two parameters that affect the performance of hardware softmax, numClasses and batchSize. Their usage has the following restrictions:
1. The value range of numClasses is 1-1023, and the value range of batchSize is 1-65535. 
2. If numClasses % 4 == 0, then numClasses * batchSize < 1048575 is required.
3. If numClasses % 4 != 0, then (numClasses - (numClasses% 4) + 4) * batchSize < 1048575 is required.
In summary, if you want to use the performance of hardware, you can increase the amount of data processed each time. For example, provide multiple batchSize data in one calculation.

JerrySciangula, 2021-03-08:
Hi @wanghy-xlnx, thank you very much for your exhaustive reply. I have some questions about what you stated. 

1) What do you mean by using multiple batchSize data? For example, do you mean using an higher number of images for inference and then apply the softmax? 

2) How the softmax calculation is carried out with a batchSize greater than 1?

wanghy-xlnx, 2021-03-09:
Hi @JerrySciangula ,

For the performance of the following declaration function:
int dpuRunSoftmax(int8_t *input, float *output, int numClasses, int batchSize, float scale);

I think it depends on the specific performance requirements of the application. For example, now the softmax of 1000 data is calculated, if the performance of about 60us per image is enough, then the software method can be used. If you want to reach about 2us per image, you can put the data of every 100 or more images in the **_input_** one by one, and set **_numClasses_** to 1000 and **_batchSize_** to 100. Then the Softmax values of 100 images will be obtained at a time, and get them from the **_output_** one by one.

The data in **_input_** and **_output_** are arranged in the order of batch by batch. For example, if the **_input_** size is 1000, we can treat it as 1 batch with 1000 data; or as 100 batches with 10 data in each batch. The arrangement of **_output_** is the same as that of **_input_**.

closed 2021-03-10 by JerrySciangula