Xilinx/Vitis-AI #301: Build python XIR for conda

DaniilNelubin, 2021-02-19:
I have some questions about XIR deploying for conda environment. Unilog,xir, target factory and vart pakeges building and installing correctly. My main question goes like this: “How to install compiled XIR library in my conda environment?”

I have compiled XIR library with the following keys:
~/Vitis-AI/tools/Vitis-AI-Runtime/VART/xir$sudo ./cmake.sh --conda --clean --build-python --pack deb –install-prefix /home/nelubin/anaconda3/envs/enotWork/

All keys with which cmake.sh is run:
cmake -DBUILD_CONTRIB=OFF -DBUILD_DOC=OFF -DCPACK_GENERATOR=DEB -DCMAKE_PREFIX_PATH=/home/nelubin/anaconda3/envs/enotWork/ -DCMAKE_BUILD_TYPE=Debug -DCMAKE_EXPORT_COMPILE_COMMANDS=ON -DBUILD_PYTHON=ON -DPYTHON_EXECUTABLE:FILEPATH=/home/nelubin/anaconda3/envs/enotWork/bin/python3.7 -DBUILD_TEST=OFF -DINSTALL_HOME=OFF -DINSTALL_USER=OFF -DCMAKE_INSTALL_PREFIX=/home/nelubin/anaconda3/envs/enotWork/ /home/nelubin/Vitis-AI/tools/Vitis-AI-Runtime/VART/xir

This script is installed correctly in the following path:

Install the project...
-- Install configuration: "Debug"
-- Installing: /home/nelubin/anaconda3/envs/enotWork/share/cmake/xir/xir-config.cmake
-- Installing: /home/nelubin/anaconda3/envs/enotWork/share/cmake/xir/xir-config-version.cmake
-- Installing: /home/nelubin/anaconda3/envs/enotWork/lib/libxir.so.1.3.0
-- Up-to-date: /home/nelubin/anaconda3/envs/enotWork/lib/libxir.so.1
-- Set runtime path of "/home/nelubin/anaconda3/envs/enotWork/lib/libxir.so.1.3.0" to "/usr/local/lib:/home/nelubin/anaconda3/envs/enotWork/lib"
-- Up-to-date: /home/nelubin/anaconda3/envs/enotWork/lib/libxir.so
-- Up-to-date: /home/nelubin/anaconda3/envs/enotWork/include/xir/xir.h
-- Up-to-date: /home/nelubin/anaconda3/envs/enotWork/include/xir/op/op.hpp
-- Up-to-date: /home/nelubin/anaconda3/envs/enotWork/include/xir/op/op_def.hpp
-- Up-to-date: /home/nelubin/anaconda3/envs/enotWork/include/xir/attrs/attrs.hpp
-- Up-to-date: /home/nelubin/anaconda3/envs/enotWork/include/xir/attrs/attr_def.hpp
-- Up-to-date: /home/nelubin/anaconda3/envs/enotWork/include/xir/attrs/attr_expander.hpp
-- Up-to-date: /home/nelubin/anaconda3/envs/enotWork/include/xir/util/any.hpp
-- Up-to-date: /home/nelubin/anaconda3/envs/enotWork/include/xir/util/tool_function.hpp
-- Up-to-date: /home/nelubin/anaconda3/envs/enotWork/include/xir/util/data_type.hpp
-- Up-to-date: /home/nelubin/anaconda3/envs/enotWork/include/xir/graph/graph.hpp
-- Up-to-date: /home/nelubin/anaconda3/envs/enotWork/include/xir/graph/graph_template.hpp
-- Up-to-date: /home/nelubin/anaconda3/envs/enotWork/include/xir/graph/subgraph.hpp
-- Up-to-date: /home/nelubin/anaconda3/envs/enotWork/include/xir/tensor/tensor.hpp
-- Installing: /home/nelubin/anaconda3/envs/enotWork/share/cmake/xir/xir-targets.cmake
-- Installing: /home/nelubin/anaconda3/envs/enotWork/share/cmake/xir/xir-targets-debug.cmake
-- Installing: /home/nelubin/anaconda3/envs/enotWork/lib/python3.7/site-packages/xir.cpython-37m-x86_64-linux-gnu.so
-- Set runtime path of "/home/nelubin/anaconda3/envs/enotWork/lib/python3.7/site-packages/xir.cpython-37m-x86_64-linux-gnu.so" to "/home/nelubin/anaconda3/envs/enotWork/lib"
-- Installing: /home/nelubin/anaconda3/envs/enotWork/bin/xir
-- Set runtime path of "/home/nelubin/anaconda3/envs/enotWork/bin/xir" to "/usr/local/lib:/home/nelubin/anaconda3/envs/enotWork/lib"

But when I check my packages in the conda env ("conda list"), I make sure that conda doesn't see the installed package.

I understand that this question is more related to the conda environment, but I'm wondering how you did it, because in your docker xir the package is installed in anaconda.

DaniilNelubin, 2021-02-19:
Hmm, pytorch-nndct didn't no longer requires XIR to be installed, that is, he sees it. But XIR still doesn't show up in conda.

wangxd-xlnx, 2021-03-03:
Hi @DaniilNelubin, ./cmake.sh will install the xir into the conda environment, but not use the conda install way, so, there's some information for conda missing, which will make the conda list ignoring the xir and doesn't show it. But the python will search the conda environment for xir when user try to import xir, so, the nndct can import it properly, and that's the same for other user defined python scripts.

DaniilNelubin, 2021-03-03:
Okay, thanks for your explanation. But how can I create a conda recipe to use conda install (or whatever to install conda)? I understand that I don't need it right now, but I'm interested.

wangxd-xlnx, 2021-03-03:
Hi @DaniilNelubin 
You could active conda like this. 
[image]
At first, you need build and run docker. For steps, please refer to here: https://github.com/Xilinx/Vitis-AI#installation

qianglin-xlnx, 2022-02-18:
Hi @DaniilNelubin 
Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-02-18 by qianglin-xlnx