Xilinx/Vitis-AI #302: Quantization own pointpillar model

vasdvczxc, 2021-02-22:
I'm trying to quantize my custom pointpillar model.

However, I am struggling with no guides other than quantization of the resnet model.

Anyone somebody know about this?

Thank you.

wangxd-xlnx, 2021-02-22:
Hi @vasdvczxc,
You could refer to this page at first. 
https://www.xilinx.com/html_docs/vitis_ai/1_3/pytorch.html#nvh1592318322520
In addition, you could describe the specific problems in detail. Then we will reply for it.

vasdvczxc, 2021-02-22:
Thank you for your reply. 
I already saw URL that you said. 
In resnet quantization, input data is image using dataloader.
However, in pointpillar, I don't know how to consist calibration dataloader.

vasdvczxc, 2021-02-23:
Thank you for your reply.

When I was trying to quantize my custom model, I got an error like below. 

Segmentation fault (core dumped)

Can you guide about this?

vasdvczxc, 2021-02-23:
Hi, @wangxd-xlnx,
Thank you for your reply.
I'm trying quantization using bash file in pointpillar.
However, I'm getting some error like below.

[NNDCT_NOTE]: Loading NNDCT kernels...
Traceback (most recent call last):
  File "code/test/test.py", line 396, in <module>
    main(args)
  File "code/test/test.py", line 288, in main
    text_format.Merge(proto_str, config)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/google/protobuf/text_format.py", line 734, in Merge
    allow_unknown_field=allow_unknown_field)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/google/protobuf/text_format.py", line 802, in MergeLines
    return parser.MergeLines(lines, message)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/google/protobuf/text_format.py", line 827, in MergeLines
    self._ParseOrMerge(lines, message)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/google/protobuf/text_format.py", line 849, in _ParseOrMerge
    self._MergeField(tokenizer, message)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/google/protobuf/text_format.py", line 974, in _MergeField
    merger(tokenizer, message, field)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/google/protobuf/text_format.py", line 1048, in _MergeMessageField
    self._MergeField(tokenizer, sub_message)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/google/protobuf/text_format.py", line 974, in _MergeField
    merger(tokenizer, message, field)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/google/protobuf/text_format.py", line 1048, in _MergeMessageField
    self._MergeField(tokenizer, sub_message)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/google/protobuf/text_format.py", line 941, in _MergeField
    (message_descriptor.full_name, name))
google.protobuf.text_format.ParseError: 8:5 : Message type "second.protos.VoxelNet" has no field named "num_class".

Can you let me know about this problem?

Treemann, 2021-02-25:
Hi @vasdvczxc ,

It's recommended to use our released training code (which is based on [https://github.com/nutonomy/second.pytorch.git](https://github.com/nutonomy/second.pytorch.git)) to train with your configuration and use our code for quantization with VAI-1.3.
Since PointPillars is more complex than simple Resnet for classification and it's our first time to support this kind of models, it requires some efforts to modify your code. Beside the official guidance, let me give you some hint:
1) Basic usage of torch_quantizer, refer to L330-L341 at "pointpillars/code/test/test.py"
```python
    if quant_mode != 'float':
        max_voxel_num = config.eval_input_reader.max_number_of_voxels
        max_point_num_per_voxel = model_cfg.voxel_generator.max_number_of_points_per_voxel
        aug_voxels = torch.randn((1, 4, max_voxel_num, max_point_num_per_voxel)).to(device)
        coors = torch.randn((max_voxel_num, 4)).to(device)
        quantizer = torch_quantizer(quant_mode=quant_mode,
                                    module=net, 
                                    input_args=(aug_voxels, coors), 
                                    output_dir=output_dir,
                                    device=device,
                                    )
        net = quantizer.quant_model
```
2) The pre-/post-processing parts (which are not run on DPU) should be picked out of the model's  forward() function, and make sure your modified model could pass jit.trace() ([Vitis AI User Documentation](https://www.xilinx.com/html_docs/vitis_ai/1_3/pytorch.html#nvh1592318322520)), refer to L302 (`net, preprocessor, postprocessor = second_builder.build(model_cfg, voxel_generator, target_assigner)`) at "pointpillars/code/test/test.py"
3) "QuantStub" and "DeQuantStub" should be used to mark the start & end of each sub-network (For PointPillars, the sub-networks is PointNet and RPN, the middle layer - scatter is not a network and should not be included in any pair of "QuantStub" and "DeQuantStub"), refer to L509-L525 at "pointpillars/code/test/models/voxelnet.py"
```python
        if os.environ["W_QUANT"]=='1':
            aug_voxels = self.quant(aug_voxels)
        voxel_features = self.voxel_feature_extractor(aug_voxels)
        if os.environ["W_QUANT"]=='1':
            voxel_features = self.dequant(voxel_features)

        spatial_features = self.middle_feature_extractor(
            voxel_features, coors, batch_size_dev)

        if os.environ["W_QUANT"]=='1':
            spatial_features = self.quant(spatial_features)
        if self._use_bev:
            preds_dict = self.rpn(spatial_features, example["bev_map"])
        else:
            preds_dict = self.rpn(spatial_features)
        if os.environ["W_QUANT"]=='1':
            preds_dict = self.dequant(preds_dict)
```
4) The original class definition of PointNet should be replaced with equivalent form (due to the original definition includes  `permute` operation which is not supported by DPU), refer to `class PFNLayer` in "pointpillars/code/test/models/pointpillars.py"

You could try with the main points here with the official documentation.

wangxd-xlnx, 2022-02-18:
Hi @vasdvczxc 

Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you.

closed 2022-02-18 by qianglin-xlnx