Xilinx/Vitis-AI #304: Core dumped on executing any tensorflow1.15

hitesharora1997, 2021-02-23:
When I am trying to run any `tensorflow1.15`, the core is being dumped every time while it runs fine on `tensorflow2.3`

``` Docker Image Version:  1.3.593 
Build Date: 2021-02-03
VAI_ROOT: /opt/vitis_ai

For TensorFlow 1.15 Workflows do:
     conda activate vitis-ai-tensorflow 
For Caffe Workflows do:
     conda activate vitis-ai-caffe 
For Neptune Workflows do:
     conda activate vitis-ai-neptune 
For PyTorch Workflows do:
     conda activate vitis-ai-pytorch 
For TensorFlow 2.3 Workflows do:
     conda activate vitis-ai-tensorflow2 
Vitis-AI /workspace > conda activate vitis-ai-tensorflow
(vitis-ai-tensorflow) Vitis-AI /workspace > vai_q_tensorflow quantize
Illegal instruction (core dumped) 
```

```
user@user-To-be-filled-by-O-E-M:~/vitis/Vitis-AI$ sudo lspci -v
01:00.0 Processing accelerators: Xilinx Corporation Device 5000
	Subsystem: Xilinx Corporation Device 000e
	Flags: bus master, fast devsel, latency 0
	Memory at f2000000 (64-bit, prefetchable) [size=32M]
	Memory at f4000000 (64-bit, prefetchable) [size=128K]
	Capabilities: [40] Power Management version 3
	Capabilities: [60] MSI-X: Enable+ Count=33 Masked-
	Capabilities: [70] Express Endpoint, MSI 00
	Capabilities: [100] Advanced Error Reporting
	Capabilities: [1c0] #19
	Capabilities: [400] Access Control Services
	Capabilities: [410] #15
	Kernel driver in use: xclmgmt
	Kernel modules: xclmgmt

01:00.1 Processing accelerators: Xilinx Corporation Device 5001
	Subsystem: Xilinx Corporation Device 000e
	Flags: bus master, fast devsel, latency 0, IRQ 16
	Memory at f0000000 (64-bit, prefetchable) [size=32M]
	Memory at f4020000 (64-bit, prefetchable) [size=64K]
	Memory at e0000000 (64-bit, prefetchable) [size=256M]
	Capabilities: [40] Power Management version 3
	Capabilities: [60] MSI-X: Enable+ Count=33 Masked-
	Capabilities: [70] Express Endpoint, MSI 00
	Capabilities: [100] Advanced Error Reporting
	Capabilities: [400] Access Control Services
	Capabilities: [410] #15
	Kernel driver in use: xocl
	Kernel modules: xocl

```
```tensorboard               1.15.0             pyhb230dea_0  
tensorflow                1.15.0          mkl_py36h4920b83_0  
tensorflow-base           1.15.0          mkl_py36he1670d9_0  
tensorflow-estimator      1.15.1             pyh2649769_0  
```

qianglin-xlnx, 2021-02-26:
Hi @hitesharora1997 
It probably be the host environment issue. We can't reproduce the issue in our server.
Could you provide more details about your host machine and the docker you used. 
If possible, could you use another new machine and try it again. Thank you.

hitesharora1997, 2021-03-04:
So, these are my host machine details as mentioned below

``` user-to-be-filled-by-o-e-m  
    description: Computer
    width: 64 bits
    capabilities: smp vsyscall32
  *-core
       description: Motherboard
       physical id: 0
     *-memory
          description: System memory
          physical id: 0
          size: 15GiB
     *-cpu
          product: Intel(R) Core(TM) i7-3770 CPU @ 3.40GHz
          vendor: Intel Corp.
          physical id: 1
          bus info: cpu@0
          size: 3719MHz
          capacity: 3900MHz
          width: 64 bits
          capabilities: fpu fpu_exception wp vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx rdtscp x86-64 constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx smx est tm2 ssse3 cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm cpuid_fault epb pti ssbd ibrs ibpb stibp tpr_shadow vnmi flexpriority ept vpid fsgsbase smep erms xsaveopt dtherm ida arat pln pts md_clear flush_l1d cpufreq
     *-pci
          description: Host bridge
          product: Xeon E3-1200 v2/3rd Gen Core processor DRAM Controller
          vendor: Intel Corporation
          physical id: 100
          bus info: pci@0000:00:00.0
          version: 09
          width: 32 bits
          clock: 33MHz
          configuration: driver=ivb_uncore
          resources: irq:0
        *-pci:0
             description: PCI bridge
             product: Xeon E3-1200 v2/3rd Gen Core processor PCI Express Root Port
             vendor: Intel Corporation
             physical id: 1
             bus info: pci@0000:00:01.0
             version: 09
             width: 32 bits
             clock: 33MHz
             capabilities: pci normal_decode bus_master cap_list
             configuration: driver=pcieport
             resources: irq:26 ioport:e0000000(size=336592896)
           *-generic:0
                description: Processing accelerators
                product: Xilinx Corporation
                vendor: Xilinx Corporation
                physical id: 0
                bus info: pci@0000:01:00.0
                version: 00
                width: 64 bits
                clock: 33MHz
                capabilities: bus_master cap_list
                configuration: driver=xclmgmt latency=0
                resources: irq:0 memory:f2000000-f3ffffff memory:f4000000-f401ffff
           *-generic:1
                description: Processing accelerators
                product: Xilinx Corporation
                vendor: Xilinx Corporation
                physical id: 0.1
                bus info: pci@0000:01:00.1
                version: 00
                width: 64 bits
                clock: 33MHz
                capabilities: bus_master cap_list
                configuration: driver=xocl latency=0
                resources: irq:16 memory:f0000000-f1ffffff memory:f4020000-f402ffff memory:e0000000-efffffff
        *-display
[... 154 lines cut ...]
 Built:             Tue Mar  2 20:18:05 2021
 OS/Arch:           linux/amd64
 Context:           default
 Experimental:      true

Server: Docker Engine - Community
 Engine:
  Version:          20.10.5
  API version:      1.41 (minimum version 1.12)
  Go version:       go1.13.15
  Git commit:       363e9a8
  Built:            Tue Mar  2 20:16:00 2021
  OS/Arch:          linux/amd64
  Experimental:     false
 containerd:
  Version:          1.4.3
  GitCommit:        269548fa27e0089a8b8278fc4fc781d7f65a939b
 runc:
  Version:          1.0.0-rc92
  GitCommit:        ff819c7e9184c13b7c2607fe6c30ae19403a7aff
 docker-init:
  Version:          0.19.0
  GitCommit:        de40ad0
```
When i am trying running some commands on this machine it says ``` Illegal instruction (core dumped) ``` but when i tried running on some other machine it runs fine .

qianglin-xlnx, 2021-03-04:
HI @hitesharora1997 
Can you run the following command to show the march? Thank you.
gcc -march=native -Q --help=target|grep march

hitesharora1997, 2021-03-04:
hello Xilinx/Vitis-AI
after running the command i got i this output

```  -march=                     ivybridge
```

hitesharora1997, 2021-03-04:
hello Xilinx/Vitis-AI
after running the command i got i this output

  -march=                     ivybridge

tres-vite, 2022-04-22:
I am still facing the same issue.. @qianglin-xlnx @hitesharora1997 

the steps i did is, 

1) First i tried to build the docker from source
cd Vitis-AI/setup/docker/
./docker_build_cpu.sh

2) i closed it in between the process and tried to pull the docker 
3) Now i am not able to import tensorflow from vitis ai docker having error like
     Illegal instruction (core dumped) 

Please help me out you guys have any solution.

qianglin-xlnx, 2022-04-22:
Hi @tres-vite 
Could you please provide the full log? Thanks.

tres-vite, 2022-04-22:
@qianglin-xlnx :Please check logs

==========================================

__      ___ _   _                   _____
\ \    / (_) | (_)            /\   |_   _|
 \ \  / / _| |_ _ ___ ______ /  \    | |
  \ \/ / | | __| / __|______/ /\ \   | |
   \  /  | | |_| \__ \     / ____ \ _| |_
    \/   |_|\__|_|___/    /_/    \_\_____|

==========================================

Docker Image Version:  1.4.1.978
Vitis AI Git Hash:  9f3d6db
Build Date: 2021-10-08

For TensorFlow 1.15 Workflows do:
     conda activate vitis-ai-tensorflow
For Caffe Workflows do:
     conda activate vitis-ai-caffe
For PyTorch Workflows do:
     conda activate vitis-ai-pytorch
For TensorFlow 2.3 Workflows do:
     conda activate vitis-ai-tensorflow2
Vitis-AI /workspace > conda activate vitis-ai-tensorflow
(vitis-ai-tensorflow) Vitis-AI /workspace >
(vitis-ai-tensorflow) Vitis-AI /workspace > python
Python 3.6.13 | packaged by conda-forge | (default, Sep 23 2021, 07:56:31)
[GCC 9.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import tensorflow
Illegal instruction (core dumped)
(vitis-ai-tensorflow) Vitis-AI /workspace >

[image]

qianglin-xlnx, 2022-04-22:
Hi @tres-vite 
I can't reproduce your error on my side, as shown below.
[image]

I think the "Illegal instruction (core dumped)" means your host CPU doesn't support AVX or some other instruction.  Maybe you can try to compile the tensorflow from source to resolved this issue.

You can refer to: https://github.com/tensorflow/tensorflow/issues/17411

tres-vite, 2022-04-22:
@qianglin-xlnx 

I believe all the dependencies should come along with docker image. because we are using docker here.

tres-vite, 2022-04-22:
@qianglin-xlnx @hitesharora1997 

One of the alternative.

1) Activate environment
conda activate vitis-ai-tensorflow

2) uninstall tensorflow 
pip uninstall tensorflow 

3) install tf again
pip uninstall tensorflow==1.15.2

closed 2022-07-29 by qianglin-xlnx