Xilinx/Vitis-AI #309: ZCU104: adas_detection not working with prebuild SD_card image

NorbertBeck, 2021-02-26:
When I try to run  the VART adas detection demo on the prebuild SD card image I get the following error:

root@xilinx-zcu104-2020_2:~/Vitis-AI/demo/VART/adas_detection# ./adas_detection adas.webm /usr/share/vitis_ai_library/models/yolov3_adas_pruned_0_9/yolov3_adas_pruned_0_9.xmodel
WARNING: Logging before InitGoogleLogging() is written to STDERR
I1226 12:43:46.814553  4316 main.cc:379] create running for subgraph: subgraph_layer0-conv
W1226 12:43:57.409380  4322 xrt_cu.cpp:195] cu timeout! device_core_idx 1  handle=0xaaaaefeecea0 ENV_PARAM(XLNX_DPU_TIMEOUT) 10000 state 1 ERT_CMD_STATE_COMPLETED 4 ms 10042  bo=1 is_done 0
I1226 12:43:57.409515  4322 xrt_cu.cpp:105] Total: 10042876us   ToDriver: 15us  ToCU: 18446738924291202us       Complete: 0us Done: 5159461210us
F1226 12:43:57.409552  4322 dpu_control_xrt_edge.cpp:177] dpu timeout! core_idx = 1
 LSTART 18  LEND 16  CSTART 1  CEND 1  SSTART 0  SEND 0  MSTART 0  MEND 0  CYCLE_L 2614271509  CYCLE_H 322

I found similar reports on the Xilinx forum, but no reply.

qianglin-xlnx, 2021-02-26:
Hi @NorbertBeck 
The log shows DPU timeout. It's abnormal. Since we have tested it many times and can't reproduce this issue on our side, we need more information.
1 Could you please reboot the board and run the other samples to see if the DPU works normally, such as resnet50 sample. Execute the following command
 ./resnet50 /usr/share/vitis_ai_library/models/resnet50/resnet50.xmodel
2 Provide the detail steps after you download the zcu104 image. (md5: 308f91cd95f88d0c08fcbccd4272ad00  xilinx-zcu104-dpu-v2020.2-v1.3.0.img.gz)

NorbertBeck, 2021-03-01:
Hello,

  I can confirm that resnet50 is working means the following output is generated

WARNING: Logging before InitGoogleLogging() is written to STDERR

I1229 07:13:56.149976  1201 main.cc:285] create running for subgraph: subgraph_conv1

Image : 001.jpg

top[0] prob = 0.980779  name = brain coral

top[1] prob = 0.008485  name = coral reef

top[2] prob = 0.008485  name = jackfruit, jak, jack

top[3] prob = 0.000542  name = sea urchin

top[4] prob = 0.000422  name = puffer, pufferfish, blowfish, globefish

I verified that the md5sum of the image which I write to SD Card using etcher is correct.

After writing the image I copy the videos/images tarball and extract it in the VART directory.

I do nothing else before launching the commands

I found a similar report here: https://forums.xilinx.com/t5/AI-and-Vitis-AI/CU-timeout-DPU-aborts-while-running-demo/td-p/1203567

Thanks in advance and Best Regards,

Norbert

qianglin-xlnx, 2021-03-01:
Hi @NorbertBeck 
I'm sorry, I still can't reproduce the issue on my board. 
It looks like a hardware issue, such as power consumption, memory, etc. I'm not sure. Are there any other peripherals connected to the board.  If so, can you temporarily remove them.

Beside, can you run the other video vart samples, such as:

 ./video_analysis video/structure.webm /usr/share/vitis_ai_library/models/ssd_traffic_pruned_0_9/ssd_traffic_pruned_0_9.xmodel

./segmentation video/traffic.webm /usr/share/vitis_ai_library/models/fpn/fpn.xmodel

Or you can try the following: 
1. change another SD card and tyr again;
2.  Reduce DPU frequency, such as: dpu_clk 80

NorbertBeck, 2021-03-01:
Hello

video_analysis and segmentation also fail with the cu timeout! Device_core_idx 1

Regarding your proposals:
Re 1.: Currently I do not have another SD card but I will bring one tomorrow
Re 2.: How can I change the dpu_clk when using the prebuild SD image?

Best Regards,

Norbert

qianglin-xlnx, 2021-03-02:
@NorbertBeck 
Run the following command to set the DPU clock to 80% 
`root@xilinx-zcu104-2020_2:~# dpu_clk 80
root@xilinx-zcu104-2020_2:~# dpu_clk
Real PL0_CLK 83333333
DPU Performance 83.3%
root@xilinx-zcu104-2020_2:~#
`

NorbertBeck, 2021-03-03:
re1.: Changing the SD card has no effect 
re2: Reducing the DPU frequency has no effect

I suspect that there is something wrong with the display as I get a lot of drm related logs in dmesg after I run the command:
[  791.115077] [drm] Pid 1188 opened device
[  791.115104] [drm] Pid 1188 closed device
[  791.115192] [drm] Pid 1188 opened device
[  791.115202] [drm] Pid 1188 closed device
[  791.118175] [drm] Pid 1188 opened device
[  791.118187] [drm] Pid 1188 closed device
[  791.118201] [drm] Pid 1188 opened device
[  791.118211] [drm] Pid 1188 closed device
[  791.124391] [drm] Pid 1188 opened device
[  791.124525] [drm] Pid 1188 closed device
[  791.142282] [drm] Pid 1188 opened device
[  791.142297] [drm] Pid 1188 closed device
[  791.142324] [drm] Pid 1188 opened device
[  791.151494] [drm] get section DEBUG_IP_LAYOUT err: -22
[  791.151503] [drm] get section AIE_METADATA err: -22
[  791.151529] [drm] zocl_xclbin_read_axlf f0bb3934-5df0-4c3d-ba52-852e904e4245 ret: 0
[  791.155500] [drm] bitstream f0bb3934-5df0-4c3d-ba52-852e904e4245 locked, ref=1
[  791.155534] [drm] No ERT scheduler on MPSoC, using KDS
[  791.155640] [drm] scheduler config ert(0)
[  791.155642] [drm]   cus(2)
[  791.155644] [drm]   slots(16)
[  791.155646] [drm]   num_cu_masks(1)
[  791.155648] [drm]   cu_shift(16)
[  791.155650] [drm]   cu_base(0x80000000)
[  791.155652] [drm]   polling(0)
[  791.155679] [drm] bitstream f0bb3934-5df0-4c3d-ba52-852e904e4245 unlocked, ref=0
[  791.155828] [drm] Pid 1188 opened device
[  791.155842] [drm] Pid 1188 closed device
[  791.155945] [drm] Pid 1188 opened device
[  791.156190] [drm] bitstream f0bb3934-5df0-4c3d-ba52-852e904e4245 locked, ref=1
[  791.156210] [drm] Pid 1188 opened device
[  791.156220] [drm] Pid 1188 closed device
[  791.156461] [drm] Pid 1188 opened device
[  791.156595] [drm] bitstream f0bb3934-5df0-4c3d-ba52-852e904e4245 locked, ref=2
[  791.156616] [drm] Pid 1188 opened device
[  791.156626] [drm] Pid 1188 closed device
[  791.156647] [drm] Pid 1188 opened device
[  791.156702] [drm] bitstream f0bb3934-5df0-4c3d-ba52-852e904e4245 locked, ref=3
[  791.157255] [drm] Pid 1188 opened device
[  791.157265] [drm] Pid 1188 closed device
[  791.157287] [drm] Pid 1188 opened device
[  791.157329] [drm] bitstream f0bb3934-5df0-4c3d-ba52-852e904e4245 locked, ref=4
[  791.237379] [drm] Pid 1188 opened device
[  791.237410] [drm] Pid 1188 closed device
[  791.237454] [drm] Pid 1188 opened device
[  791.237569] [drm] bitstream f0bb3934-5df0-4c3d-ba52-852e904e4245 locked, ref=5
[  791.238128] [drm] Pid 1188 opened device
[  791.238139] [drm] Pid 1188 closed device
[  791.238161] [drm] Pid 1188 opened device
[  791.238205] [drm] bitstream f0bb3934-5df0-4c3d-ba52-852e904e4245 locked, ref=6
[  791.288493] [drm] Pid 1188 opened device
[  791.288525] [drm] Pid 1188 closed device
[  791.288572] [drm] Pid 1188 opened device
[  791.288684] [drm] bitstream f0bb3934-5df0-4c3d-ba52-852e904e4245 locked, ref=7
[  791.289243] [drm] Pid 1188 opened device
[  791.289253] [drm] Pid 1188 closed device
[  791.289275] [drm] Pid 1188 opened device
[  791.289319] [drm] bitstream f0bb3934-5df0-4c3d-ba52-852e904e4245 locked, ref=8
[  791.339270] [drm] Pid 1188 opened device
[  791.339304] [drm] Pid 1188 closed device
[  791.339348] [drm] Pid 1188 opened device
[  791.339466] [drm] bitstream f0bb3934-5df0-4c3d-ba52-852e904e4245 locked, ref=9
[  791.340020] [drm] Pid 1188 opened device
[  791.340031] [drm] Pid 1188 closed device
[  791.340054] [drm] Pid 1188 opened device
[  791.340101] [drm] bitstream f0bb3934-5df0-4c3d-ba52-852e904e4245 locked, ref=10
[... 21 lines cut ...]
[  812.288195] [drm] addr 0x80010000, status 0x1
[  812.288198] [drm] addr 0x80000000, status 0x4
[  812.288205] [drm] bitstream f0bb3934-5df0-4c3d-ba52-852e904e4245 unlocked, ref=0
[  812.288210] [drm] Pid 1188 closed device
[  812.288240] [drm] bitstream f0bb3934-5df0-4c3d-ba52-852e904e4245 unlocked, ref=0
[  812.288242] [drm] Pid 1188 closed device
[  812.288257] [drm] bitstream f0bb3934-5df0-4c3d-ba52-852e904e4245 unlocked, ref=0
[  812.288259] [drm] Pid 1188 closed device
[  812.288275] [drm] bitstream f0bb3934-5df0-4c3d-ba52-852e904e4245 unlocked, ref=0
[  812.288277] [drm] Pid 1188 closed device
[  812.288291] [drm] bitstream f0bb3934-5df0-4c3d-ba52-852e904e4245 unlocked, ref=0
[  812.288293] [drm] Pid 1188 closed device
[  812.288306] [drm] bitstream f0bb3934-5df0-4c3d-ba52-852e904e4245 unlocked, ref=0
[  812.288308] [drm] Pid 1188 closed device
[  812.288322] [drm] bitstream f0bb3934-5df0-4c3d-ba52-852e904e4245 unlocked, ref=0
[  812.288324] [drm] Pid 1188 closed device
[  812.288337] [drm] bitstream f0bb3934-5df0-4c3d-ba52-852e904e4245 unlocked, ref=0
[  812.288340] [drm] Pid 1188 closed device
[  812.288346] [drm] bitstream f0bb3934-5df0-4c3d-ba52-852e904e4245 unlocked, ref=0
[  812.288348] [drm] Pid 1188 closed device
[  812.293688] [drm] bitstream f0bb3934-5df0-4c3d-ba52-852e904e4245 unlocked, ref=0
[  812.293692] [drm] Pid 1188 closed device
[  812.293721] [drm] Pid 1188 closed device

This looks very similar to the report, that I linked earlier:  https://forums.xilinx.com/t5/AI-and-Vitis-AI/CU-timeout-DPU-aborts-while-running-demo/td-p/1203567

tianfang-fafafa, 2022-02-24:
hi @NorbertBeck ， could you please try the same command with zcu102 or zcu104 v2021.2 ?
[image]

qianglin-xlnx, 2022-03-03:
Hi @NorbertBeck 
Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-03-03 by qianglin-xlnx