Xilinx/Vitis-AI #311: NNDCT_ERROR following pytorch tutorial

Graeme22, 2021-03-01:
Hello,

While following the [pytorch tutorial](https://github.com/Xilinx/Vitis-In-Depth-Tutorial/tree/master/Machine_Learning/Design_Tutorials/09-mnist_pyt-master) I came across the following error in the quantization step:

```shell
python -u quantize.py --quant_mode calib
[NNDCT_ERROR]: /opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/nn/_kernels.cpython-36m-x86_64-linux-gnu.so: undefined symbol: _ZN6caffe28TypeMeta21_typeMetaDataInstanceIN3c108BFloat16EEEPKNS_6detail12TypeMetaDataEv
```

Seems like nndct is broken, any solutions?

Graeme22, 2021-03-02:
Looks like there was a version mismatch between pip and conda, and pip's torch 1.7 was overriding conda's torch 1.4. Uninstalling the pip version solved the issue.

Graeme22, 2021-03-02:
Edit: Should've tested more first. Reopening as torch 1.4 appears to be too old to run the training script; however, if the training is done with a newer torch version (e.g. 1.7.1 as is default on the docker image) then we get the error above.

foolmarks, 2021-03-03:
Just tried the flow again using the Vitis-AI 1.3 GPU Docker and all went well - PyTorch version is shown as being 1.4 which matches the version supported according to the UG1414 User Guide.

niuxjxlnx, 2022-02-22:
Solution to change pytorch versions among the supported version range is released since Vitis-AI 1.4.1, please refer to the related part about script replace_pytorch.sh in user guide.

closed 2022-02-22 by niuxjxlnx