Xilinx/Vitis-AI #312: ValueError: substring not found

DaniilNelubin, 2021-03-02:
I'm trying to quantize **HRnet**. This **model passes torch.jit** test and jit test in the pytorch_nndct (I had to make an effort to get the model to pass the test. [For this](https://github.com/pytorch/pytorch/issues/30459), I replaced all None in the model with toch.nn.Identity ). After the changes made, the model works correctly.
But when I trying to do quantizing (with `qat_porc=True`) I have the next error:

-------- Start quantization --------
/home/nelubin/Documents/mseg_test/mseg-semantic/mseg_semantic/model/init_model.py:515: TracerWarning: Converting a tensor to a Python boolean might cause the trace to be incorrect. We can't record the data flow of Python values, so this value will be treated as a constant in the future. This means that the trace might not generalize to other inputs!
  assert (x_size[2] - 1) % 8 == 0 and (x_size[3] - 1) % 8 == 0
/home/nelubin/anaconda3/envs/enotWork/lib/python3.7/site-packages/torch/nn/functional.py:2506: UserWarning: Default upsampling behavior when mode=bilinear is changed to align_corners=False since 0.4.0. Please specify align_corners=True if the old behavior is desired. See the documentation of nn.Upsample for details.
  "See the documentation of nn.Upsample for details.".format(mode))
/home/nelubin/anaconda3/envs/enotWork/lib/python3.7/site-packages/torch/nn/functional.py:2416: UserWarning: nn.functional.upsample is deprecated. Use nn.functional.interpolate instead.
  warnings.warn("nn.functional.upsample is deprecated. Use nn.functional.interpolate instead.")
/home/nelubin/anaconda3/envs/enotWork/lib/python3.7/site-packages/torch/jit/__init__.py:1044: TracerWarning: Output nr 1. of the traced function does not match the corresponding output of the Python function. Detailed error:
Not within tolerance rtol=1e-05 atol=1e-05 at input[1, 191, 240, 92] (1.6136640310287476 vs. 1.6137433052062988) and 2438119 other locations (5.00%)
  check_tolerance, _force_outplace, True, _module_class)

[NNDCT_NOTE]: Quantization calibration process start up...

[NNDCT_NOTE]: =>Quant Module is in 'cpu'.

[NNDCT_NOTE]: =>Parsing HighResolutionNet...

[NNDCT_NOTE]: =>Quantizable module is generated.(quantize_result/HighResolutionNet.py)

[NNDCT_NOTE]: =>Exporting quant config.(quantize_result/quant_info.json)

File "/home/nelubin/Documents/mseg_test/mseg-semantic/mseg_semantic/tool/inference_task.py", line 684, in quantize_to_xilinx
    self.args.quant_mode, self.model, input_args=input_tensor, device=self.device,qat_proc=True)
  File "/home/nelubin/anaconda3/envs/enotWork/lib/python3.7/site-packages/pytorch_nndct/apis.py", line 50, in __init__
    mix_bit = mix_bit)
  File "/home/nelubin/anaconda3/envs/enotWork/lib/python3.7/site-packages/pytorch_nndct/quantization/quant_aware_training.py", line 293, in __init__
    name = _topo_node_name(node)
  File "/home/nelubin/anaconda3/envs/enotWork/lib/python3.7/site-packages/pytorch_nndct/quantization/quant_aware_training.py", line 104, in _topo_node_name
    module_name = mod_util.module_name_from_node(node)
  File "/home/nelubin/anaconda3/envs/enotWork/lib/python3.7/site-packages/pytorch_nndct/utils/module_util.py", line 36, in module_name_from_node
    left_bracket = part.index('[')
ValueError: substring not found

yuwang-xlnx, 2021-03-03:
Hi DaniilNelubin,

The reason is that some quantized operations are not torch.nn.Module.
Can you please help print out the name of the node ?

Open /home/nelubin/anaconda3/envs/enotWork/lib/python3.7/site-packages/pytorch_nndct/utils/module_util.py",
In line 32, add a print statement: 
```
parts = node_name.split(_NODE_NAME_SEPERATOR)[1:-1]
print(node_name)
```

DaniilNelubin, 2021-03-03:
Hi @yuwang-xlnx
I got the following result:
```

HighResolutionNet::HighResolutionNet/1841
HighResolutionNet::HighResolutionNet/1843
HighResolutionNet::HighResolutionNet/Conv2d[conv1]/input.2
HighResolutionNet::HighResolutionNet/BatchNorm2d[bn1]/input.3
HighResolutionNet::HighResolutionNet/Conv2d[conv1]/input.2
HighResolutionNet::HighResolutionNet/ReLU[relu]/input.4
HighResolutionNet::HighResolutionNet/BatchNorm2d[bn1]/input.3
HighResolutionNet::HighResolutionNet/Conv2d[conv2]/input.5
HighResolutionNet::HighResolutionNet/ReLU[relu]/input.4
HighResolutionNet::HighResolutionNet/BatchNorm2d[bn2]/input.6
HighResolutionNet::HighResolutionNet/Conv2d[conv2]/input.5
HighResolutionNet::HighResolutionNet/ReLU[relu]/input.7
HighResolutionNet::HighResolutionNet/BatchNorm2d[bn2]/input.6
HighResolutionNet::HighResolutionNet/Sequential[layer1]/Bottleneck[0]/AdaptiveBottleneck[block]/Conv2d[conv1]/input.8
HighResolutionNet::HighResolutionNet/ReLU[relu]/input.7
HighResolutionNet::HighResolutionNet/Sequential[layer1]/Bottleneck[0]/AdaptiveBottleneck[block]/BatchNorm2d[bn1]/input.9
HighResolutionNet::HighResolutionNet/Sequential[layer1]/Bottleneck[0]/AdaptiveBottleneck[block]/Conv2d[conv1]/input.8
HighResolutionNet::HighResolutionNet/Sequential[layer1]/Bottleneck[0]/AdaptiveBottleneck[block]/ReLU[relu]/input.10
HighResolutionNet::HighResolutionNet/Sequential[layer1]/Bottleneck[0]/AdaptiveBottleneck[block]/BatchNorm2d[bn1]/input.9
HighResolutionNet::HighResolutionNet/Sequential[layer1]/Bottleneck[0]/AdaptiveBottleneck[block]/Conv2d[conv2]/input.11
HighResolutionNet::HighResolutionNet/Sequential[layer1]/Bottleneck[0]/AdaptiveBottleneck[block]/ReLU[relu]/input.10
HighResolutionNet::HighResolutionNet/Sequential[layer1]/Bottleneck[0]/AdaptiveBottleneck[block]/BatchNorm2d[bn2]/input.12
HighResolutionNet::HighResolutionNet/Sequential[layer1]/Bottleneck[0]/AdaptiveBottleneck[block]/Conv2d[conv2]/input.11
HighResolutionNet::HighResolutionNet/Sequential[layer1]/Bottleneck[0]/AdaptiveBottleneck[block]/ReLU[relu]/input.13
HighResolutionNet::HighResolutionNet/Sequential[layer1]/Bottleneck[0]/AdaptiveBottleneck[block]/BatchNorm2d[bn2]/input.12
HighResolutionNet::HighResolutionNet/Sequential[layer1]/Bottleneck[0]/AdaptiveBottleneck[block]/Conv2d[conv3]/input.14
HighResolutionNet::HighResolutionNet/Sequential[layer1]/Bottleneck[0]/AdaptiveBottleneck[block]/ReLU[relu]/input.13
HighResolutionNet::HighResolutionNet/Sequential[layer1]/Bottleneck[0]/AdaptiveBottleneck[block]/BatchNorm2d[bn3]/1927
HighResolutionNet::HighResolutionNet/Sequential[layer1]/Bottleneck[0]/AdaptiveBottleneck[block]/Conv2d[conv3]/input.14
HighResolutionNet::HighResolutionNet/Sequential[layer1]/Bottleneck[0]/Sequential[downsample]/Conv2d[0]/input.15
HighResolutionNet::HighResolutionNet/ReLU[relu]/input.7
HighResolutionNet::HighResolutionNet/Sequential[layer1]/Bottleneck[0]/Sequential[downsample]/BatchNorm2d[1]/1943
HighResolutionNet::HighResolutionNet/Sequential[layer1]/Bottleneck[0]/Sequential[downsample]/Conv2d[0]/input.15
HighResolutionNet::HighResolutionNet/Sequential[layer1]/Bottleneck[0]/input.16
HighResolutionNet::HighResolutionNet/Sequential[layer1]/Bottleneck[0]/AdaptiveBottleneck[block]/BatchNorm2d[bn3]/1927
HighResolutionNet::HighResolutionNet/Sequential[layer1]/Bottleneck[0]/Sequential[downsample]/BatchNorm2d[1]/1943
HighResolutionNet::HighResolutionNet/Sequential[layer1]/Bottleneck[0]/ReLU[relu]/input.17
HighResolutionNet::HighResolutionNet/Sequential[layer1]/Bottleneck[0]/input.16
HighResolutionNet::HighResolutionNet/Sequential[layer1]/Bottleneck[1]/AdaptiveBottleneck[block]/Conv2d[conv1]/input.18
HighResolutionNet::HighResolutionNet/Sequential[layer1]/Bottleneck[0]/ReLU[relu]/input.17
HighResolutionNet::HighResolutionNet/Sequential[layer1]/Bottleneck[1]/AdaptiveBottleneck[block]/BatchNorm2d[bn1]/input.19
HighResolutionNet::HighResolutionNet/Sequential[layer1]/Bottleneck[1]/AdaptiveBottleneck[block]/Conv2d[conv1]/input.18
HighResolutionNet::HighResolutionNet/Sequential[layer1]/Bottleneck[1]/AdaptiveBottleneck[block]/ReLU[relu]/input.20
HighResolutionNet::HighResolutionNet/Sequential[layer1]/Bottleneck[1]/AdaptiveBottleneck[block]/BatchNorm2d[bn1]/input.19
HighResolutionNet::HighResolutionNet/Sequential[layer1]/Bottleneck[1]/AdaptiveBottleneck[block]/Conv2d[conv2]/input.21
HighResolutionNet::HighResolutionNet/Sequential[layer1]/Bottleneck[1]/AdaptiveBottleneck[block]/ReLU[relu]/input.20
HighResolutionNet::HighResolutionNet/Sequential[layer1]/Bottleneck[1]/AdaptiveBottleneck[block]/BatchNorm2d[bn2]/input.22
HighResolutionNet::HighResolutionNet/Sequential[layer1]/Bottleneck[1]/AdaptiveBottleneck[block]/Conv2d[conv2]/input.21
HighResolutionNet::HighResolutionNet/Sequential[layer1]/Bottleneck[1]/AdaptiveBottleneck[block]/ReLU[relu]/input.23
HighResolutionNet::HighResolutionNet/Sequential[layer1]/Bottleneck[1]/AdaptiveBottleneck[block]/BatchNorm2d[bn2]/input.22
HighResolutionNet::HighResolutionNet/Sequential[layer1]/Bottleneck[1]/AdaptiveBottleneck[block]/Conv2d[conv3]/input.24
HighResolutionNet::HighResolutionNet/Sequential[layer1]/Bottleneck[1]/AdaptiveBottleneck[block]/ReLU[relu]/input.23
HighResolutionNet::HighResolutionNet/Sequential[layer1]/Bottleneck[1]/AdaptiveBottleneck[block]/BatchNorm2d[bn3]/1996
HighResolutionNet::HighResolutionNet/Sequential[layer1]/Bottleneck[1]/AdaptiveBottleneck[block]/Conv2d[conv3]/input.24
HighResolutionNet::HighResolutionNet/Sequential[layer1]/Bottleneck[1]/input.25
[... cut ...]
HighResolutionNet::HighResolutionNet/Sequential[layer1]/Bottleneck[3]/input.43
HighResolutionNet::HighResolutionNet/Sequential[layer1]/Bottleneck[3]/AdaptiveBottleneck[block]/BatchNorm2d[bn3]/2102
HighResolutionNet::HighResolutionNet/Sequential[layer1]/Bottleneck[2]/ReLU[relu]/input.35
HighResolutionNet::HighResolutionNet/Sequential[layer1]/Bottleneck[3]/ReLU[relu]/input.44
HighResolutionNet::HighResolutionNet/Sequential[layer1]/Bottleneck[3]/input.43
HighResolutionNet::HighResolutionNet/Sequential/Conv2d[0]/input.45
```

yuwang-xlnx, 2021-03-08:
Hi @DaniilNelubin,

I have figured out the cause of the problem and it will be fixed in the next release.

DaniilNelubin, 2021-03-09:
Thanks for your reply @yuwang-xlnx!  
I understand that this may be inside information, but when approximately can we get a new release from your team?
Quantization and porting of HRnet to the edge device are also very important to me. Can I fix this problem myself?

yuwang-xlnx, 2022-02-21:
Hi @DaniilNelubin,

The issue has been fixed. 
Have you tried the latest version?

closed 2022-03-02 by qianglin-xlnx