Xilinx/Vitis-AI #313: Is it possible to connect DPU directly to DDR instead of connecting to PS?

nimaomidsajedi, 2021-03-02:
Hi everyone,

I have a design issue. I am using the DPU TRD example (https://github.com/Xilinx/Vitis-AI/blob/master/dsa/DPU-TRD/prj/Vivado/README.md). I am wondering if it is possible to connect the DPU directly to the PS side DDR memory (with a DMA IP) or to the PL side DDR memory and read the weights and activations directly instead of connecting DPU to the PS and from there to the PS side DDR memory. I think this can boost the system performance but since I am not aware of Vitis AI side of the project, I am not sure if this is possible.

Question summary:

The current control path of DPU:
DPU => Connect to PS => Connect to PS side DDR memory

Is it possible to change the control path of DPU to the following without problem?
DPU => Connect directly to DDR memory of PS side
or
DPU => Connect directly to DDR memory of PL side

I appreciate any feedback on the issue.

davidxudong, 2021-03-05:
Hi @nimaomidsajedi 

It is possible to connect the DPU directly to DDR memory of PL side, but you should also evaluate the DDR bandwith of PL side on your board, whether DDR of PL side is better than PS side.

Thanks

qianglin-xlnx, 2022-02-25:
Hi @nimaomidsajedi 
Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-02-25 by qianglin-xlnx