Xilinx/Vitis-AI #314: inceptionv3

dingguodong-826, 2021-03-03:
Hello, thank you very much for your work. When I was using the PyTorch version of Inceptionv3, I wanted to use my own data set to carry out training quantification and other operations, but I didn't find the following file you mentioned. Could you please provide me with it?thank you
[image]

wangxd-xlnx, 2021-03-03:
Hi @dingguodong-826 ,
The PyTorch Inceptionv3 you mentioned is from torchvision.models. You could refer to this link and find it. So we didn't provide this model in AI-Model-Zoo similar to other models.
https://pytorch.org/vision/stable/models.html

dingguodong-826, 2021-03-03:
Will you release the training, quantization, and compilation parts of this version in the future? Because I would like to refer to my improved InceptionV3 network based on PyTorch after quantifying compilation, if you can share it. Thank you very much!

wangxd-xlnx, 2022-02-22:
Hi @dingguodong-826 

We have released this model in Vitis-AI 2.0 model zoo, with quantization and compiled xmodel. You could refer to this link.
https://github.com/Xilinx/Vitis-AI/tree/master/models/AI-Model-Zoo

closed 2022-02-22 by qianglin-xlnx