Xilinx/Vitis-AI #315: ZCU102: Vivado flow not working

ttymrz, 2021-03-04:
Hi,

I'm trying [DPU TRD Vivado Flow](https://github.com/Xilinx/Vitis-AI/tree/master/dsa/DPU-TRD/prj/Vivado) with ZCU102.
But the inference result is wrong. And it seems DPU is not running.

```console
root@xilinx-zcu102-2020_2:~# insmod dpu.ko
[   83.102694] dpu: loading out-of-tree module taints kernel.
[   83.109475] [DPU][816]Found DPU signature addr = 0x8f000000 in device-tree
[   83.116359] [DPU][816]Checking DPU signature at addr = 0x8ff00000,
[   83.122875] [DPU][816]Init softmax IP...
[   83.126837] [DPU][816]Request softmax IRQ 51 successful.
[   83.126841] [DPU][816]Init softmax IP done
root@xilinx-zcu102-2020_2:~# insmod dpcma.ko
root@xilinx-zcu102-2020_2:~# show_dpu
device_core_id=0 device= 0 core = 0 fingerprint = 0x1000020f6014407 batch = 1 full_cu_name=unknown:dpu0
root@xilinx-zcu102-2020_2:~# samples/bin/resnet50 bellpeppe-994958.JPEG
score[3]    =  0.001        text: tiger shark, Galeocerdo cuvieri,
score[4]    =  0.001        text: hammerhead, hammerhead shark,
score[1]    =  0.001        text: goldfish, Carassius auratus,
score[0]    =  0.001        text: tench, Tinca tinca,
score[2]    =  0.001        text: great white shark, white shark, man-eater, man-eating shark, Carcharodon carcharias,
root@xilinx-zcu102-2020_2:~# export DEBUG_DPU_RUNNER=1
root@xilinx-zcu102-2020_2:~# samples/bin/resnet50 bellpeppe-994958.JPEG
WARNING: Logging before InitGoogleLogging() is written to STDERR
I0109 05:26:51.832034   861 dpu_session_imp.cpp:67] create dpu session @0x556a96b940 device_core_id_ 0 device_id 0 is_ddr 1 dpu_name unknown
I0109 05:26:53.111438   861 dpu_kernel.cpp:37] filename resnet50.xmodel kernel resnet50_0 ret.get() 0x556a987890
I0109 05:26:53.111538   861 dpu_kernel.cpp:170] get workspace sizes for deploy
I0109 05:26:53.111577   861 dpu_kernel.cpp:193] total workspace size = 2610152
I0109 05:26:53.111600   861 dpu_kernel.cpp:50] create dpu kernel. graph deploy;sub graph subgraph_conv1(ReplaceConv2d) @0x556eac9b60
I0109 05:26:53.111626   861 dpu_kernel_ddr.cpp:40]  create dpu kernel @0x556a989900 cu=unknown:dpu0 device_id=0 device_core_id=0
I0109 05:26:53.111649   861 dpu_kernel.cpp:88] loading parameter for deploy
I0109 05:26:53.167703   861 dpu_kernel.cpp:127] loading release code for deploy
I0109 05:26:53.172976   861 dpu_kernel_ddr.cpp:105] loading release code  1697580 bytes to 0x60100000
I0109 05:26:53.188490   861 dpu_session_base_imp.cpp:111] session is created.subgraph: subgraph_conv1(ReplaceConv2d)
I0109 05:26:53.188556   861 dpu_session_base_imp.cpp:115] input tensor:mytensor{data_fixed(float2fix)(fix):(1,224,224,3), fixpos=-1}
I0109 05:26:53.188596   861 dpu_session_base_imp.cpp:118] input tensor:mytensor{fc1000_fixed(float2fix)(ReplaceReshape):(1,1000), fixpos=2}
I0109 05:26:53.188627   861 dpu_session_imp.cpp:140]  create dpu runner @ 0x556a96b940 device_id= 0 device_core_id=0
I0109 05:26:53.253299   861 dpu_runner_base_imp.cpp:63] create  dpu runner 0x556c79ad30 device_core_id 0
I0109 05:26:53.273422   861 dpu_runner_ddr.cpp:79] copy input:TensorBuffer{@0x556c7d83f0,tensor=xir::Tensor{name = data_fixed(float2fix)(fix), type = XINT8, shape = {1, 224, 224, 3}},location=HOST_PHY,data=[(Virt=0x7f9e2abbfe, 150528,Phy=0x61c00bfe, 150528)]}
I0109 05:26:53.273605   861 dpu_runner_base_imp.cpp:631] @0x556c79ad30 device_core_id=0 DPU: unknown:dpu0:0 running dpu code  0x60100000 gen_reg.size() 8 gen_reg
        0x60300000      0x61c00000      0xffffffffffffffff      0xffffffffffffffff      0xffffffffffffffff      0xffffffffffffffff      0xffffffffffffffff      0xffffffffffffffff
I0109 05:26:53.273679   861 dpu_runner_ddr.cpp:132] copy_output:TensorBuffer{@0x55732c9300,tensor=xir::Tensor{name = fc1000_fixed(float2fix)(ReplaceReshape), type = XINT8, shape = {1, 1000}},location=HOST_PHY,data=[(Virt=0x7f9e2ab000, 1000,Phy=0x61c00000, 1000)]}
score[3]    =  0.001        text: tiger shark, Galeocerdo cuvieri,
score[4]    =  0.001        text: hammerhead, hammerhead shark,
score[1]    =  0.001        text: goldfish, Carassius auratus,
score[0]    =  0.001        text: tench, Tinca tinca,
score[2]    =  0.001        text: great white shark, white shark, man-eater, man-eating shark, Carcharodon carcharias,
I0109 05:26:53.274003   861 dpu_runner_base_imp.cpp:70] destroying dpu runner 0x556c79ad30 device_core_id 0
I0109 05:26:53.274025   861 dpu_session_imp.cpp:125]  destroy dpu session @0x556a96b940
I0109 05:26:53.279440   861 dpu_kernel_ddr.cpp:78] kernel destoryed
I0109 05:26:53.279798   861 dpu_kernel.cpp:71] kernel destoryed. graph deploy;sub graph subgraph_conv1(ReplaceConv2d) @0x556eac9b60
root@xilinx-zcu102-2020_2:~#
```

I had to change `dpu.ko` from [xilinx-zcu102-trd.bsp](https://github.com/Xilinx/Vitis-AI/tree/master/dsa/DPU-TRD/prj/Vivado/dpu_petalinux_bsp) to [Vitis-AI-Runtime](https://github.com/Xilinx/Vitis-AI/tree/master/tools/Vitis-AI-Runtime/DNNDK/driver).
And I had to change devicetree.

```
dpu {
	compatible = "xilinx,dpu";
	base-addr = <0x8f000000>;

	dpucore {
		compatible = "xilinx,dpucore";
		interrupt-parent = <&gic>;
		interrupts = <0 91 1 0 92 1>;
		core-num = <2>;
	};
	softmax {
		compatible = "xilinx,smfc";
		interrupt-parent = <&gic>;
		interrupts = <0 95 1>;
		core-num = <1>;
	};
};
dpcma: dpcma {
	compatible = "deephi,cma";
};
```

Thanks

qianglin-xlnx, 2021-03-04:
HI @ttymrz 
Do you install the VART1.3.0 package? If not, please run the following command to test the sample.
env LD_LIBRARY_PATH=samples/lib samples/bin/resnet50 img/bellpeppe-994958.JPEG

ttymrz, 2021-03-05:
Hi @qianglin-xlnx 
I am using VART1.3.0 package.

```console
root@xilinx-zcu102-2020_2:~# env LD_LIBRARY_PATH=samples/lib  samples/bin/resnet50 bellpeppe-994958.JPEG
score[3]    =  0.001        text: tiger shark, Galeocerdo cuvieri,
score[4]    =  0.001        text: hammerhead, hammerhead shark,
score[1]    =  0.001        text: goldfish, Carassius auratus,
score[0]    =  0.001        text: tench, Tinca tinca,
score[2]    =  0.001        text: great white shark, white shark, man-eater, man-eating shark, Carcharodon carcharias,
```

```console
root@xilinx-zcu102-2020_2:~# env LD_LIBRARY_PATH=samples/lib  vart_version
Symbol xilinx_version not found in libunilog.so.1: samples/lib/libunilog.so.1: undefined symbol: xilinx_version
Symbol xilinx_version not found in libtarget-factory.so.1: /usr/lib/libtarget-factory.so.1: undefined symbol: xilinx_version
libxir.so.1: Xilinx xir Version: xir-a57405ab47fbe8f07e6e022558947c2ebb285386 2021-01-11-15:14:48
libvart-buffer-object.so.1: Xilinx vart-buffer-object Version: 1.3.0-ee9521bac72865a6dbb1f215e4e0f83f9e8fe74d  2021-01-11-15:14:53
libvart-dpu-controller.so.1: Xilinx vart-dpu-controller Version: 1.3.0-ee9521bac72865a6dbb1f215e4e0f83f9e8fe74d  2021-01-11-15:14:53
libvart-dpu-runner.so.1: Xilinx vart-dpu-runner Version: 1.3.0-ee9521bac72865a6dbb1f215e4e0f83f9e8fe74d  2021-01-11-15:14:53
libvart-mem-manager.so.1: Xilinx vart-mem-manager Version: 1.3.0-ee9521bac72865a6dbb1f215e4e0f83f9e8fe74d  2021-01-11-15:14:53
libvart-runner.so.1: Xilinx vart-runner Version: 1.3.0-ee9521bac72865a6dbb1f215e4e0f83f9e8fe74d  2021-01-11-15:14:53
libvart-util.so.1: Xilinx vart-util Version: 1.3.0-ee9521bac72865a6dbb1f215e4e0f83f9e8fe74d  2021-01-11-15:14:53
libvart-xrt-device-handle.so.1: Xilinx vart-xrt-device-handle Version: 1.3.0-ee9521bac72865a6dbb1f215e4e0f83f9e8fe74d  2021-01-11-15:14:53

```

andrew0917, 2021-03-05:
Hi @ttymrz,
     As for your devicetree configuration and shell log, I want to konw which version of xilinx-zcu102-trd.bsp do your use? Beacuse for the recent released xilinx-zcu102-trd.bsp(VAI1.1,VAI1.2,VAI1.3), the reference devicetree(xilinx-zcu102-trd/project-spec/meta-user/recipes-bsp/device-tree/files/system-user.dtsi) as below.
```
/{
        dpu: dpu@8f000000 {
                compatible = "deephi, dpu";
                reg = <0x0 0x8f000000 0x0 0x700>;
                interrupts = <0x0 106 0x1 0x0 107 0x1 0x0 108 0x1 0x0 109 0x1>;
                interrupt-parent = <&gic>;
                core-num = <0x4>;

                softmax@800001000 {
                        compatible = "deephi,smfc";
                        interrupt-parent = <&gic>;
                        interrupts = <0x0 0x6e 0x1>;
                        core-num = <0x1>;
                };
        };
        dpcma: dpcma {
                compatible = "deephi,cma";
        };

};
```
    In addtion, the dpu.ko & dpcma.ko are intergrated into the rootfs in released xilinx-zcu102-trd.bsp, it can be loaded to system automatically after linux boot, so I'm strange that why you load them manually.

ttymrz, 2021-03-05:
Hi @andrew0917

I replaced dpu.ko and devicetree with [xilinx-zcu102-trd.bsp](https://github.com/Xilinx/Vitis-AI/blob/master/dsa/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/download_bsp.sh).
Then I got correct result.

Could you tell me which dpu.ko is latest version.
https://github.com/Xilinx/Vitis-AI/tree/master/tools/Vitis-AI-Runtime/DNNDK/driver
https://github.com/Xilinx/Vitis-AI/blob/master/dsa/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/download_bsp.sh

```console
root@xilinx-zcu102-2020_2:~# env LD_LIBRARY_PATH=samples/lib samples/bin/resnet50 bellpeppe-994958.JPEG
score[945]  =  0.992235     text: bell pepper,
score[941]  =  0.00315807   text: acorn squash,
score[943]  =  0.00191546   text: cucumber, cuke,
score[939]  =  0.000904801  text: zucchini, courgette,
score[949]  =  0.00054879   text: strawberry,
```

Thanks

andrew0917, 2021-03-05:
Hi @ttymrz 
As for Vivado flow BSP release, the dpu driver is only integrated into the BSP(VAI1.1,1.2,1.3). It means that the latest dpu.ko version is https://github.com/Xilinx/Vitis-AI/blob/master/dsa/DPU-TRD/prj/Vivado/dpu_petalinux_bsp/download_bsp.sh.
I'm sorry that I have no idea about https://github.com/Xilinx/Vitis-AI/tree/master/tools/Vitis-AI-Runtime/DNNDK/driver, maybe you can ask @qianglin-xlnx for more details.
Thanks

closed 2021-04-14 by ttymrz