Xilinx/Vitis-AI #319: Vitis-AI-1.2 run command docker_build_gpu.sh dead(last for several hours) at step 20

hopesala, 2021-03-05:
Step 17/31 : RUN cd /tmp && wget http://launchpadlibrarian.net/436533799/libjson-c4_0.13.1+dfsg-4_amd64.deb &&     dpkg -i libjson-c4_0.13.1+dfsg-4_amd64.deb && rm -fr /tmp/*
 ---> Using cache
 ---> 4638e62eba7d
Step 18/31 : ENV PATH="${VAI_ROOT}/utility:${PATH}"
 ---> Using cache
 ---> badfc066ec87
Step 19/31 : ENV MY_CONDA_CHANNEL="file:///scratch/conda-channel"
 ---> Using cache
 ---> a55a12a8a089
Step 20/31 : RUN mkdir -p /scratch && wget -q https://repo.anaconda.com/miniconda/Miniconda3-latest-Linux-x86_64.sh -O /scratch/miniconda.sh &&         /bin/bash /scratch/miniconda.sh -b -p $VAI_ROOT/conda &&         ln -s $VAI_ROOT/conda/etc/profile.d/conda.sh /etc/profile.d/conda.sh &&         echo ". $VAI_ROOT/conda/etc/profile.d/conda.sh" >> ~/.bashrc &&         echo "conda activate base" >> ~/.bashrc
 ---> Running in 8b5ee18547bd

hanxue, 2021-03-08:
Hi @hopesala ,

Could you cancel the command and run again?

If you are still having problem, could you please check if you can download Miniconda, i.e. 

```
wget https://repo.anaconda.com/miniconda/Miniconda3-latest-Linux-x86_64.sh
```

qianglin-xlnx, 2022-02-27:
Hi @hopesala 
Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-02-27 by qianglin-xlnx