Xilinx/Vitis-AI #321: Build deeplabv3 from pytorch in host docker error

abdulazizm, 2021-03-06:
This is my first post and I am new to TVM and Pyxir/Vitis AI, please excuse if my question seems silly.

Trying to build deeplabv3 from pytorch to deploy on top of xilinx EDGE device - zcu104. Facing some issue while annotating. Code snippets here:

```
mod, params = relay.frontend.from_pytorch(scripted_model, shape_list)
mod["main"] = bind_params_by_name(mod["main"], params)
desired_layouts = {'nn.conv2d': ['NHWC', 'default'], 'nn.adaptive_avg_pool2d':['nn.avg_pool2d', 'default']}
seq = tvm.transform.Sequential([relay.transform.RemoveUnusedFunctions(),
                                relay.transform.ConvertLayout(desired_layouts),
                                relay.transform.FoldConstant()])
with tvm.transform.PassContext(opt_level=3):
    mod = seq(mod)
mod = annotation(mod, params, target) #shows error here
```
Two doubtful things in output:

1. /opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/pyxir-0.1.6-py3.6-linux-x86_64.egg/pyxir/frontend/tvm/relay_tools/relay_l10_temporary.py:64: UserWarning: Convert Relay Adaptive Avg pool2d layer into normal average pool2d layer warnings.warn(“Convert Relay Adaptive Avg pool2d layer into normal”
2. File “/opt/vitis_ai/conda/envs/vitis-ai-tensorflow/lib/python3.6/site-packages/pyxir-0.1.6-py3.6-linux-x86_64.egg/pyxir/graph/ops/l1_basic_nn.py”, line 167, in concat assert i == axis or len(check) == 1 AssertionError

How to proceed further? Is there any way to change Adaptive Avg pool2d layer to normal average pool2d layer using relay.transform?

Is there any tutorials any one aware of building deeplabv3 for Xilinx EDGE device (from mxnet or pytorch or any other framework)?

wangxd-xlnx, 2021-03-11:
@abdulazizm Hi, I noticed that the conda you used is vitis-ai-tensorflow. If it's a pytorch model, you could have a try to use vitis-ai-pytorch. 
For more details, you could refer to https://www.xilinx.com/html_docs/vitis_ai/1_3/deploying_running.html#zgy1576168058789

abdulazizm, 2021-03-11:
@wangxd-xlnx OOPS! Thats a good catch. Will give it a try with vitis-ai-pytorch environment and come back. I followed this tutorial -> https://xterra2.avnet.com/xilinx/ml/tvm-on-xilinx-mpsoc/-/tree/master where it builds mxnet framework models in vitis-ai-tensorflow environment (I assumed that the same shoud be working for pytorch too). 

Thanks for your reply!!

abdulazizm, 2021-03-11:
@wangxd-xlnx I tried with vitis-ai-pytorch environment, but getting the same assertion error.

/home/vitis-ai-user/.local/lib/python3.6/site-packages/pyxir-0.1.6-py3.6-linux-x86_64.egg/pyxir/frontend/tvm/relay_tools/relay_l10_temporary.py:64: UserWarning: Convert Relay Adaptive Avg pool2d layer into normal average pool2d layer
  warnings.warn("Convert Relay Adaptive Avg pool2d layer into normal"
Traceback (most recent call last):
  File "compile_pytorch_deeplab.py", line 189, in <module>
    mod = annotation(mod, params, target)
  File "/workspace/python/tvm/relay/op/contrib/vitis_ai.py", line 92, in annotation
    xgraph = pyxir.frontend.tvm.from_relay(mod, params, postprocessing=None)
  File "/home/vitis-ai-user/.local/lib/python3.6/site-packages/pyxir-0.1.6-py3.6-linux-x86_64.egg/pyxir/frontend/tvm/relay.py", line 58, in from_relay
    cvx_preprocessing=cvx_preprocessing
  File "/home/vitis-ai-user/.local/lib/python3.6/site-packages/pyxir-0.1.6-py3.6-linux-x86_64.egg/pyxir/frontend/tvm/relay_tools/relay_2_xgraph_converter.py", line 96, in from_relay_to_xgraph
    cvx_prep=cvx_preprocessing)
  File "/home/vitis-ai-user/.local/lib/python3.6/site-packages/pyxir-0.1.6-py3.6-linux-x86_64.egg/pyxir/frontend/tvm/relay_tools/relay_l0_expr_and_others.py", line 75, in function
    op_idx, RELAY_2_XLAYER, **kwargs)
  File "/home/vitis-ai-user/.local/lib/python3.6/site-packages/pyxir-0.1.6-py3.6-linux-x86_64.egg/pyxir/frontend/tvm/relay_tools/relay_l0_expr_and_others.py", line 184, in tuple_expr
    RELAY_2_XLAYER, **kwargs)
  File "/home/vitis-ai-user/.local/lib/python3.6/site-packages/pyxir-0.1.6-py3.6-linux-x86_64.egg/pyxir/frontend/tvm/relay_tools/relay_l0_expr_and_others.py", line 117, in call
    RELAY_2_XLAYER, **kwargs)
  File "/home/vitis-ai-user/.local/lib/python3.6/site-packages/pyxir-0.1.6-py3.6-linux-x86_64.egg/pyxir/frontend/tvm/relay_tools/relay_2_xlayer_registry.py", line 122, in __base_relay_2_xlayer
    RELAY_2_XLAYER, **kwargs)
  File "/home/vitis-ai-user/.local/lib/python3.6/site-packages/pyxir-0.1.6-py3.6-linux-x86_64.egg/pyxir/frontend/tvm/relay_tools/relay_l0_expr_and_others.py", line 117, in call
    RELAY_2_XLAYER, **kwargs)
  File "/home/vitis-ai-user/.local/lib/python3.6/site-packages/pyxir-0.1.6-py3.6-linux-x86_64.egg/pyxir/frontend/tvm/relay_tools/relay_2_xlayer_registry.py", line 122, in __base_relay_2_xlayer
    RELAY_2_XLAYER, **kwargs)
  File "/home/vitis-ai-user/.local/lib/python3.6/site-packages/pyxir-0.1.6-py3.6-linux-x86_64.egg/pyxir/frontend/tvm/relay_tools/relay_l0_expr_and_others.py", line 117, in call
    RELAY_2_XLAYER, **kwargs)
  File "/home/vitis-ai-user/.local/lib/python3.6/site-packages/pyxir-0.1.6-py3.6-linux-x86_64.egg/pyxir/frontend/tvm/relay_tools/relay_l1_basic.py", line 78, in add
    **kwargs)
  File "/home/vitis-ai-user/.local/lib/python3.6/site-packages/pyxir-0.1.6-py3.6-linux-x86_64.egg/pyxir/frontend/tvm/relay_tools/relay_l0_expr_and_others.py", line 117, in call
    RELAY_2_XLAYER, **kwargs)
  File "/home/vitis-ai-user/.local/lib/python3.6/site-packages/pyxir-0.1.6-py3.6-linux-x86_64.egg/pyxir/frontend/tvm/relay_tools/relay_l2_convolution.py", line 216, in nn_conv2d
    **kwargs)
  File "/home/vitis-ai-user/.local/lib/python3.6/site-packages/pyxir-0.1.6-py3.6-linux-x86_64.egg/pyxir/frontend/tvm/relay_tools/relay_l0_expr_and_others.py", line 117, in call
    RELAY_2_XLAYER, **kwargs)
  File "/home/vitis-ai-user/.local/lib/python3.6/site-packages/pyxir-0.1.6-py3.6-linux-x86_64.egg/pyxir/frontend/tvm/relay_tools/relay_2_xlayer_registry.py", line 122, in __base_relay_2_xlayer
    RELAY_2_XLAYER, **kwargs)
  File "/home/vitis-ai-user/.local/lib/python3.6/site-packages/pyxir-0.1.6-py3.6-linux-x86_64.egg/pyxir/frontend/tvm/relay_tools/relay_l0_expr_and_others.py", line 239, in tuple_get_item
    RELAY_2_XLAYER, **kwargs)
  File "/home/vitis-ai-user/.local/lib/python3.6/site-packages/pyxir-0.1.6-py3.6-linux-x86_64.egg/pyxir/frontend/tvm/relay_tools/relay_l0_expr_and_others.py", line 117, in call
    RELAY_2_XLAYER, **kwargs)
  File "/home/vitis-ai-user/.local/lib/python3.6/site-packages/pyxir-0.1.6-py3.6-linux-x86_64.egg/pyxir/frontend/tvm/relay_tools/relay_l1_basic.py", line 239, in nn_batch_norm
    **kwargs)
  File "/home/vitis-ai-user/.local/lib/python3.6/site-packages/pyxir-0.1.6-py3.6-linux-x86_64.egg/pyxir/frontend/tvm/relay_tools/relay_l0_expr_and_others.py", line 117, in call
    RELAY_2_XLAYER, **kwargs)
  File "/home/vitis-ai-user/.local/lib/python3.6/site-packages/pyxir-0.1.6-py3.6-linux-x86_64.egg/pyxir/frontend/tvm/relay_tools/relay_l2_convolution.py", line 216, in nn_conv2d
    **kwargs)
  File "/home/vitis-ai-user/.local/lib/python3.6/site-packages/pyxir-0.1.6-py3.6-linux-x86_64.egg/pyxir/frontend/tvm/relay_tools/relay_l0_expr_and_others.py", line 239, in tuple_get_item
    RELAY_2_XLAYER, **kwargs)
  File "/home/vitis-ai-user/.local/lib/python3.6/site-packages/pyxir-0.1.6-py3.6-linux-x86_64.egg/pyxir/frontend/tvm/relay_tools/relay_l0_expr_and_others.py", line 117, in call
    RELAY_2_XLAYER, **kwargs)
  File "/home/vitis-ai-user/.local/lib/python3.6/site-packages/pyxir-0.1.6-py3.6-linux-x86_64.egg/pyxir/frontend/tvm/relay_tools/relay_2_xlayer_registry.py", line 122, in __base_relay_2_xlayer
    RELAY_2_XLAYER, **kwargs)
  File "/home/vitis-ai-user/.local/lib/python3.6/site-packages/pyxir-0.1.6-py3.6-linux-x86_64.egg/pyxir/frontend/tvm/relay_tools/relay_l0_expr_and_others.py", line 117, in call
    RELAY_2_XLAYER, **kwargs)
[... cut ...]
    X = px.ops.concat(op_name, data_layers, axis, relay_id=relay_idx)
  **File "/home/vitis-ai-user/.local/lib/python3.6/site-packages/pyxir-0.1.6-py3.6-linux-x86_64.egg/pyxir/graph/ops/l1_basic_nn.py", line 167, in concat
    assert i == axis or len(check) == 1
AssertionError**

Note: Tried replacing `mod= annotation(mod, params, target)` with `mod = relay.transform.AnnotateTarget(target)(mod)` - can able to generate params, so files but not sure whether this helped me with the functionality to run in DPU. Any suggestions?

naughtyrabbit, 2022-09-26:
as far as i know, the tvm package is built only in the tensorflow conda environment if the default dockerfile is used to build the tvm_vitis-ai docker image. So could you tell me did you rebuild the tvm in the pytorch conda environment ? What i did is to simply install torch in the tensorflow conda environment as tvm is hard to compile. but still, i met with some problem to compile a basic pytorch model in the torchvison model zoo(resnet18). i failed with the relay.frontend.from_pytorch api while succeed with relay.frontend.from_onnx api after transfer the pytorch model to a onnx model. i still dont know why the failure happended. did you problem fixed?