Xilinx/Vitis-AI #322: quantization

dingguodong-826, 2021-03-08:
Hello, I encountered this problem while quantifying the PyTorch model, thank you!

[image]

niuxjxlnx, 2021-03-08:
hi, @dingguodong-826:

I noticed you met several problems when you quantize your inception-v3 model(another issue: https://github.com/Xilinx/Vitis-AI/issues/316). But we can quantize offical inception-v3 model successfully.  
So you can paste your file quantize.py here? I can help analysis the root cause.

dingguodong-826, 2021-03-08:
Hello, I have uploaded the relevant files for quantification (weights, common.py; Quant.py), thank you for your guidance
Note: I have made changes to some layers of Inceptionv3. I will email you the weights of the training if needed

[inception3_quant.zip](https://github.com/Xilinx/Vitis-AI/files/6099753/inception3_quant.zip)

niuxjxlnx, 2021-03-08:
hi, @dingguodong-826:

Can I send your quant.py, common.py and weights to me (niuxj@xilinx.com)?

dingguodong-826, 2021-03-08:
Yes, please leave your email address for me

carrotandjzy, 2021-03-11:
hi, @dingguodong-826:
We have found the root cause of your isssue. There are two consecutive slice operators in the generated quantizable module , the quantizer can't handle this situation. This issue will be solved in the next release.

hoadv-qh, 2021-03-21:
> hi, @dingguodong-826:
> We have found the root cause of your isssue. There are two consecutive slice operators in the generated quantizable module , the quantizer can't handle this situation. This issue will be solved in the next release.

Hi @carrotandjzy:
I got same issue when trying to quantize [route] layer of yolov4-tiny with attribute as below.
<pre>
[route]
layers=-1
groups=2
group_id=1
</pre> 

* Above [route] layer requires to slice output of previous layer to feed to next layer
* I checked model class (output by Vitis AI Quantizer for Pytorch). It shows there are 2 consecutive slices.
<pre>
self.module_54 = py_nndct.nn.Module('shape')
self.module_55 = py_nndct.nn.Module('tensor')
self.module_56 = py_nndct.nn.Module('const')
self.module_58 = py_nndct.nn.Int()
self.module_59 = py_nndct.nn.strided_slice() <----
self.module_60 = py_nndct.nn.strided_slice() <----
</pre>

My question: Is there any workaround to treat 2 consecutive slice operators while waiting for next release of Vitis AI?

YanxianChen, 2021-04-09:
Hi @hoadv-qh , have you found some ways to replace the slice operator in caffe or pytorch?

niuxjxlnx, 2022-02-21:
The problem has been fixed in VAI 1.4/1.4.1 and 2.0 version. Will close it if no more response.

closed 2022-02-22 by niuxjxlnx