Xilinx/Vitis-AI #323: Pointpillar Quantization Error

yongwon2, 2021-03-08:
Hello, 

I'm trying to quantize my own pointpillar model.
My first error is 
RuntimeError: version_ <= kMaxSupportedFileFormatVersion INTERNAL ASSERT FAILED at /opt/conda/conda-bld/pytorch_1579027003190/work/caffe2/serialize/inline_container.cc:132, please report a bug to PyTorch. Attempted to read a PyTorch file with version 3, but the maximum supported version for reading is 2. Your PyTorch installation may be too old. (init at /opt/conda/conda-bld/pytorch_1579027003190/work/caffe2/serialize/inline_container.cc:132)
frame #0: c10::Error::Error(c10::SourceLocation, std::string const&) + 0x47 (0x7f92ac730627 in /opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/torch/lib/libc10.so)

So, I changed my pytorch version from 1.4.0 to 1.7.1.

And next, I got an error like below.
[NNDCT_ERROR]: /opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/nn/_kernels.cpython-36m-x86_64-linux-gnu.so: undefined symbol: _ZN6caffe28TypeMeta21_typeMetaDataInstanceIN3c108BFloat16EEEPKNS_6detail12TypeMetaDataEv

Anyone know about this problem?

qianglin-xlnx, 2021-03-08:
HI @reddittoob 
You can refer to https://github.com/Xilinx/Vitis-AI/issues/302 to quantize your own pointpillar model.

reddittoob, 2021-03-09:
I already saw repo that you said.
However, in #302, there isn't about my issue.
I think my issue is occurred because pytorch version is different with my training environment. So I tried to train model in pytorch 1.4, but I got a same error.
Can you give me some solution?

niuxjxlnx, 2021-03-09:
@reddittoob 

It sames that pytorch in this directory has been changed: /opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/.
Solution to solve the problem is to re-run the docker image and activate vitis-ai-pytorch conda env again. 

Other hints:
1) Pytorch 1.7.1 is not supported in latest released vitis-ai. It will be supported in next release.
2) Make sure the model is can be ran by pytorch 1.4 version before its deployment.

niuxjxlnx, 2022-02-21:
Hi, @reddittoob:

pytorch 1.7 version has been supported in Vitis-AI 1.4,  and pytorch 1.8, 1.9 version have been supported in Vitis-AI 2.0.

closed 2022-02-22 by niuxjxlnx