Xilinx/Vitis-AI #329: Vitis_AI 1.3 caffe compilation output 0 DPU subgraph

romaintha, 2021-03-12:
Hi,

I am trying to compile the dk_yolov4_coco_416_416_60.1G_1.3 model (downloaded [here](https://www.xilinx.com/bin/public/openDownload?filename=dk_yolov4_coco_416_416_60.1G_1.3.zip) )using Vitis-AI 1.3 and caffe conda environment inside the docker like this:
`vai_c_caffe --prototxt quantized/DNNC/deploy.prototxt --caffemodel quantized/deploy.caffemodel --arch /opt/vitis_ai/compiler/arch/DPUCZDX8G/ULTRA96/arch.json --output_dir ./compiled --net_name yolov4_coco --options "{'save_kernel':''}"`

and the output is:
```
**************************************************
* VITIS_AI Compilation - Xilinx Inc.
**************************************************
[INFO] Namespace(inputs_shape=None, layout='NCHW', model_files=['quantized/deploy.caffemodel'], model_type='caffe', out_filename='./compiled/yolov4_coco_org.xmodel', proto='quantized/DNNC/deploy.prototxt')
[INFO] caffe model: quantized/deploy.caffemodel
[INFO] caffe model: quantized/DNNC/deploy.prototxt
[INFO] parse raw model     :100%|███████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 256/256 [00:19<00:00, 13.33it/s]                
[INFO] infer shape (NCHW)  :100%|███████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 256/256 [00:00<00:00, 4814.29it/s]              
[INFO] infer shape (NHWC)  :100%|███████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 256/256 [00:00<00:00, 3730.42it/s]              
[INFO] generate xmodel     :100%|███████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 256/256 [00:00<00:00, 336.49it/s]               
[INFO] generate xmodel: /workspace/yolov4_coco/compiled/yolov4_coco_org.xmodel
[UNILOG][INFO] The compiler log will be dumped at "/tmp/vitis-ai-user/log/xcompiler-20210312-014819-506"
[UNILOG][INFO] Compile mode: dpu
[UNILOG][INFO] Debug mode: function
[UNILOG][INFO] Target architecture: DPUCZDX8G_ISA0_B2304_MIN
[UNILOG][INFO] Graph name: deploy, with op num: 696
[UNILOG][INFO] Begin to compile...
[UNILOG][INFO] Total device subgraph number 2, DPU subgraph number 0
[UNILOG][INFO] Compile done.
[UNILOG][INFO] The meta json is saved to "/workspace/yolov4_coco/./compiled/meta.json"
[UNILOG][INFO] The compiled xmodel is saved to "/workspace/yolov4_coco/./compiled/yolov4_coco.xmodel"
[UNILOG][INFO] The compiled xmodel's md5sum is 23969698c768c5d88a2b15846a927561, and been saved to "/workspace/yolov4_coco/./compiled/md5sum.txt"
```
From this log I can see that the DPU subgraph number is 0, which is kinda wrong no?

Am I doing anything wrong?

qianglin-xlnx, 2021-03-12:
Hi @romaintha 
Please use quantized/deploy.prototxt instead of quantized/DNNC/deploy.prototxt

romaintha, 2021-03-12:
Oh my bad... Thanks

closed 2021-03-12 by romaintha