Xilinx/Vitis-AI #331: VART instalation error

DaniilNelubin, 2021-03-15:
Thank you for the new release 1.3.1.
When I tried to build vart 1.3.1 (after unilog, xir and target_factory of course) I get the following error. This error occurs when I ask --install-prefix (/home/nelubin/anaconda3/envs/enotWork/). And without this parameter, everything goes well. With what it can be connected?

Native-platform building...
cd /home/nelubin/build/build.Ubuntu.18.04.x86_64.Debug/vart
cmake -DBUILD_TEST=ON -DENABLE_DPU_RUNNER=ON -DCMAKE_PREFIX_PATH=/home/nelubin/anaconda3/envs/enotWork/ -DENABLE_DPU_RUNNER=ON -DENABLE_CPU_RUNNER=OFF -DENABLE_SIM_RUNNER=OFF -DCMAKE_BUILD_TYPE=Debug -DCMAKE_EXPORT_COMPILE_COMMANDS=ON -DBUILD_PYTHON=OFF -DCMAKE_INSTALL_PREFIX=/home/nelubin/anaconda3/envs/enotWork/ -DCMAKE_INSTALL_PREFIX=/home/nelubin/anaconda3/envs/enotWork/ /home/nelubin/Vitis-AI/tools/Vitis-AI-Runtime/VART/vart
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Protobuf: /usr/local/lib/libprotobuf.so;-lpthread (found version "3.14.0") 
-- Found importable target unilog::unilog: /home/nelubin/anaconda3/envs/enotWork/lib/libunilog.so.1.3.1
-- Found importable target xir::xir: /home/nelubin/anaconda3/envs/enotWork/lib/libxir.so.1.3.1
-- Found importable target target-factory::target-factory: /home/nelubin/anaconda3/envs/enotWork/lib/libtarget-factory.so.1.3.1
-- Found Xrt: /opt/xilinx/xrt/include (found version "2.3.1301") 
-- Found Boost: /usr/include (found version "1.65.1") found components: filesystem system 
-- Found OpenSSL: /home/nelubin/anaconda3/envs/enotWork/lib/libcrypto.so (found version "1.1.1j")  
-- Performing Test HAS_xclIPName2Index_3
-- Performing Test HAS_xclIPName2Index_3 - Failed
-- Performing Test HAS_xclIPName2Index_2
-- Performing Test HAS_xclIPName2Index_2 - Success
+ cat /home/nelubin/Vitis-AI/tools/Vitis-AI-Runtime/VART/vart/dpu-runner/test/word_list.txt
+ sed 's/\(.*\)/"\1",/g'
-- Found OpenCV: /usr/local (found version "4.5.1") 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/nelubin/build/build.Ubuntu.18.04.x86_64.Debug/vart
Scanning dependencies of target vart_version
Scanning dependencies of target test_batch_tensor_buffer
Scanning dependencies of target show_kernel
Scanning dependencies of target util
[  1%] Building C object util/CMakeFiles/vart_version.dir/src/vart_version.c.o
[  2%] Building CXX object dpu-controller/runner-assistant/CMakeFiles/test_batch_tensor_buffer.dir/test/test_batch_tensor_buffer.cpp.o
[  2%] Building CXX object dpu-runner/test/CMakeFiles/show_kernel.dir/show_kernel.cpp.o
[  2%] Building C object util/CMakeFiles/util.dir/version.c.o
[  3%] Building CXX object util/CMakeFiles/util.dir/src/simple_config.cpp.o
[  3%] Building CXX object util/CMakeFiles/util.dir/src/dim_calc.cpp.o
[  3%] Building CXX object util/CMakeFiles/util.dir/src/util_4bit.cpp.o
[  4%] Building CXX object util/CMakeFiles/util.dir/src/tracepoint.cpp.o
[  4%] Building CXX object util/CMakeFiles/util.dir/src/time_measure.cpp.o
[  4%] Building CXX object util/CMakeFiles/util.dir/src/file_lock.cpp.o
[  5%] Building CXX object util/CMakeFiles/util.dir/src/xir_helper.cpp.o
[  6%] Building CXX object util/CMakeFiles/util.dir/src/device_scheduler_imp.cpp.o
[  7%] Building CXX object util/CMakeFiles/util.dir/src/variable_bit.cpp.o
[  7%] Building CXX object util/CMakeFiles/util.dir/src/zero_copy_helper.cpp.o
[  7%] Building CXX object util/CMakeFiles/util.dir/src/thread_pool.cpp.o
[  7%] Building CXX object util/CMakeFiles/util.dir/src/device_scheduler.cpp.o
[  7%] Linking C executable vart_version
[  7%] Built target vart_version
[  7%] Linking CXX executable test_batch_tensor_buffer
[  7%] Built target test_batch_tensor_buffer
[  8%] Linking CXX executable show_kernel
[  8%] Built target show_kernel
[  8%] Linking CXX shared library libvart-util.so
[  8%] Built target util
Scanning dependencies of target test_file_lock
Scanning dependencies of target test_injector_lib
Scanning dependencies of target test_env_config
Scanning dependencies of target test_variable_bit
Scanning dependencies of target test_dim_calc
Scanning dependencies of target test_thread_pool
Scanning dependencies of target test_simple_config
[... 331 lines cut ...]
/usr/local/lib/libopencv_imgcodecs.so.4.5.1: undefined reference to `TIFFIsTiled@LIBTIFF_4.0'
/usr/local/lib/libopencv_imgcodecs.so.4.5.1: undefined reference to `TIFFWriteScanline@LIBTIFF_4.0'
/usr/local/lib/libopencv_imgcodecs.so.4.5.1: undefined reference to `TIFFGetField@LIBTIFF_4.0'
/usr/local/lib/libopencv_imgcodecs.so.4.5.1: undefined reference to `TIFFScanlineSize@LIBTIFF_4.0'
/usr/local/lib/libopencv_imgcodecs.so.4.5.1: undefined reference to `TIFFWriteDirectory@LIBTIFF_4.0'
/usr/local/lib/libopencv_imgcodecs.so.4.5.1: undefined reference to `TIFFReadEncodedTile@LIBTIFF_4.0'
/usr/local/lib/libopencv_imgcodecs.so.4.5.1: undefined reference to `TIFFReadRGBATile@LIBTIFF_4.0'
/usr/local/lib/libopencv_imgcodecs.so.4.5.1: undefined reference to `TIFFClose@LIBTIFF_4.0'
/usr/local/lib/libopencv_imgcodecs.so.4.5.1: undefined reference to `TIFFClientOpen@LIBTIFF_4.0'
/usr/local/lib/libopencv_imgcodecs.so.4.5.1: undefined reference to `TIFFRGBAImageOK@LIBTIFF_4.0'
/usr/local/lib/libopencv_imgcodecs.so.4.5.1: undefined reference to `TIFFOpen@LIBTIFF_4.0'
/usr/local/lib/libopencv_imgcodecs.so.4.5.1: undefined reference to `TIFFReadEncodedStrip@LIBTIFF_4.0'
/usr/local/lib/libopencv_imgcodecs.so.4.5.1: undefined reference to `TIFFSetField@LIBTIFF_4.0'
/usr/local/lib/libopencv_imgcodecs.so.4.5.1: undefined reference to `TIFFSetWarningHandler@LIBTIFF_4.0'
/usr/local/lib/libopencv_imgcodecs.so.4.5.1: undefined reference to `TIFFSetErrorHandler@LIBTIFF_4.0'
collect2: error: ld returned 1 exit status
dpu-runner/test/CMakeFiles/resnet50.dir/build.make:122: recipe for target 'dpu-runner/test/resnet50' failed
make[2]: *** [dpu-runner/test/resnet50] Error 1
CMakeFiles/Makefile2:1831: recipe for target 'dpu-runner/test/CMakeFiles/resnet50.dir/all' failed
make[1]: *** [dpu-runner/test/CMakeFiles/resnet50.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[100%] Linking CXX executable test_dpu_runner_mt
[100%] Built target test_dpu_runner_mt
Makefile:170: recipe for target 'all' failed
make: *** [all] Error 2

DaniilNelubin, 2021-03-15:
I dive into this problem. I can solve this by removing (libtiff through the `conda remove libtiff`) [link](https://github.com/BVLC/caffe/issues/1559). After this project build successfully. But `conda remove` uninstall seaborn, numpy and other important libraries.

DaniilNelubin, 2021-03-15:
1) I remove libtiff
2) Build VART
3) Install pytorch_nndct
4) Installing pillow and numpy 1.20 (It's installing libtiff again)
But it didn't work. So I have the next error:
  File "/home/nelubin/anaconda3/envs/enotWork/lib/python3.7/site-packages/torch/utils/cpp_extension.py", line 1066, in _build_extension_module
    check=True)
  File "/home/nelubin/anaconda3/envs/enotWork/lib/python3.7/subprocess.py", line 512, in run
    output=stdout, stderr=stderr)
subprocess.CalledProcessError: Command '['ninja', '-v']' returned non-zero exit status 1.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "tool/model_quantazition.py", line 24, in <module>
    from pytorch_nndct.apis import torch_quantizer, dump_xmodel
  File "/home/nelubin/anaconda3/envs/enotWork/lib/python3.7/site-packages/pytorch_nndct/__init__.py", line 12, in <module>
    from .apis import *
  File "/home/nelubin/anaconda3/envs/enotWork/lib/python3.7/site-packages/pytorch_nndct/apis.py", line 24, in <module>
    from .qproc import TorchQuantProcessor
  File "/home/nelubin/anaconda3/envs/enotWork/lib/python3.7/site-packages/pytorch_nndct/qproc/__init__.py", line 1, in <module>
    from .base import TorchQuantProcessor, dump_xmodel
  File "/home/nelubin/anaconda3/envs/enotWork/lib/python3.7/site-packages/pytorch_nndct/qproc/base.py", line 30, in <module>
    from pytorch_nndct.quantization import TORCHQuantizer
  File "/home/nelubin/anaconda3/envs/enotWork/lib/python3.7/site-packages/pytorch_nndct/quantization/__init__.py", line 2, in <module>
    from .quant_aware_training import QatScheduler
  File "/home/nelubin/anaconda3/envs/enotWork/lib/python3.7/site-packages/pytorch_nndct/quantization/quant_aware_training.py", line 28, in <module>
    from pytorch_nndct.nn.qat.modules import conv_fused
  File "/home/nelubin/anaconda3/envs/enotWork/lib/python3.7/site-packages/pytorch_nndct/nn/__init__.py", line 1, in <module>
    from .modules import *
  File "/home/nelubin/anaconda3/envs/enotWork/lib/python3.7/site-packages/pytorch_nndct/nn/modules/__init__.py", line 2, in <module>
    from .linear import *
  File "/home/nelubin/anaconda3/envs/enotWork/lib/python3.7/site-packages/pytorch_nndct/nn/modules/linear.py", line 26, in <module>
    from .fix_ops import NndctScale
  File "/home/nelubin/anaconda3/envs/enotWork/lib/python3.7/site-packages/pytorch_nndct/nn/modules/fix_ops.py", line 22, in <module>
    from ..load_kernels import nndct_kernels
  File "/home/nelubin/anaconda3/envs/enotWork/lib/python3.7/site-packages/pytorch_nndct/nn/load_kernels.py", line 93, in <module>
    with_cuda=with_cuda)
  File "/home/nelubin/anaconda3/envs/enotWork/lib/python3.7/site-packages/torch/utils/cpp_extension.py", line 680, in load
    is_python_module)
  File "/home/nelubin/anaconda3/envs/enotWork/lib/python3.7/site-packages/torch/utils/cpp_extension.py", line 866, in _jit_compile
    with_cuda=with_cuda)
  File "/home/nelubin/anaconda3/envs/enotWork/lib/python3.7/site-packages/torch/utils/cpp_extension.py", line 919, in _write_ninja_file_and_build
    _build_extension_module(name, build_directory, verbose)
  File "/home/nelubin/anaconda3/envs/enotWork/lib/python3.7/site-packages/torch/utils/cpp_extension.py", line 1079, in _build_extension_module
    raise RuntimeError(message)
RuntimeError: Error building extension 'nndct_kernels': [1/4] /usr/local/cuda/bin/nvcc -DTORCH_EXTENSION_NAME=nndct_kernels -DTORCH_API_INCLUDE_EXTENSION_H -I/home/nelubin/Vitis-AI/tools/Vitis-AI-Quantizer/vai_q_pytorch/include/cpu -I/home/nelubin/Vitis-AI/tools/Vitis-AI-Quantizer/vai_q_pytorch/pytorch_binding/pytorch_nndct/nn/include -I/home/nelubin/Vitis-AI/tools/Vitis-AI-Quantizer/vai_q_pytorch/include/cuda -isystem /home/nelubin/anaconda3/envs/enotWork/lib/python3.7/site-packages/torch/include -isystem /home/nelubin/anaconda3/envs/enotWork/lib/python3.7/site-packages/torch/include/torch/csrc/api/include -isystem /home/nelubin/anaconda3/envs/enotWork/lib/python3.7/site-packages/torch/include/TH -isystem /home/nelubin/anaconda3/envs/enotWork/lib/python3.7/site-packages/torch/include/THC -isystem /usr/local/cuda/include -isystem /home/nelubin/anaconda3/envs/enotWork/include/python3.7m -D_GLIBCXX_USE_CXX11_ABI=0 -D__CUDA_NO_HALF_OPERATORS__ -D__CUDA_NO_HALF_CONVERSIONS__ -D__CUDA_NO_HALF2_OPERATORS__ --expt-relaxed-constexpr -gencode=arch=compute_75,code=sm_75 --compiler-options '-fPIC' -std=c++11 -c /home/nelubin/Vitis-AI/tools/Vitis-AI-Quantizer/vai_q_pytorch/csrc/cuda/nndct_cuda_math.cu -o nndct_cuda_math.cuda.o
FAILED: nndct_cuda_math.cuda.o 
/usr/local/cuda/bin/nvcc -DTORCH_EXTENSION_NAME=nndct_kernels -DTORCH_API_INCLUDE_EXTENSION_H -I/home/nelubin/Vitis-AI/tools/Vitis-AI-Quantizer/vai_q_pytorch/include/cpu -I/home/nelubin/Vitis-AI/tools/Vitis-AI-Quantizer/vai_q_pytorch/pytorch_binding/pytorch_nndct/nn/include -I/home/nelubin/Vitis-AI/tools/Vitis-AI-Quantizer/vai_q_pytorch/include/cuda -isystem /home/nelubin/anaconda3/envs/enotWork/lib/python3.7/site-packages/torch/include -isystem /home/nelubin/anaconda3/envs/enotWork/lib/python3.7/site-packages/torch/include/torch/csrc/api/include -isystem /home/nelubin/anaconda3/envs/enotWork/lib/python3.7/site-packages/torch/include/TH -isystem /home/nelubin/anaconda3/envs/enotWork/lib/python3.7/site-packages/torch/include/THC -isystem /usr/local/cuda/include -isystem /home/nelubin/anaconda3/envs/enotWork/include/python3.7m -D_GLIBCXX_USE_CXX11_ABI=0 -D__CUDA_NO_HALF_OPERATORS__ -D__C [... cut ...]
  File "/home/nelubin/anaconda3/envs/enotWork/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/home/nelubin/anaconda3/envs/enotWork/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/nelubin/anaconda3/envs/enotWork/lib/python3.7/site-packages/torch/distributed/launch.py", line 263, in <module>
    main()
  File "/home/nelubin/anaconda3/envs/enotWork/lib/python3.7/site-packages/torch/distributed/launch.py", line 259, in main
    cmd=cmd)

xihuazhanghui, 2022-02-17:
I think you'd better use the latest 2.0 release docker and have a try, because your conda envirionment will lead the vart found not proper library and make the build failed.

qianglin-xlnx, 2022-03-01:
Hi @DaniilNelubin 
Since we haven't received your reply for a long time, we assume you have solved this issue and I'm going to close it. If you still have any questions, please feel free to reopen it. Thank you very much.

closed 2022-03-01 by qianglin-xlnx