Xilinx/Vitis-AI #332: Deploying vai_q_pytorch U-net architecture

jonassundseth, 2021-03-16:
Hi, 

I am trying to deploy a quantized U-net architecture, Both the calib and test stage runs as expected, but when trying to deploy this error occurs:

```
File "models.py", line 193, in quantization
    quantizer.export_xmodel(deploy_check=False)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/apis.py", line 92, in export_xmodel
    self.processor.export_xmodel(output_dir, deploy_check)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/base.py", line 161, in export_xmodel
    dump_xmodel(output_dir, deploy_check)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/base.py", line 184, in dump_xmodel
    deploy_graphs = get_deploy_graph_list(quantizer.quant_model, quantizer.Nndctgraph)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/pytorch_nndct/qproc/utils.py", line 218, in get_deploy_graph_list
    g_optmizer.constant_folding()
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/nndct_shared/compile/deploy_optimizer.py", line 79, in constant_folding
    node.set_node_attr(attr_name, attr_val)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/nndct_shared/nndct_graph/base_node.py", line 202, in set_node_attr
    self._op.set_attr(key, value)
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/nndct_shared/nndct_graph/base_operator.py", line 271, in set_attr
    self._attrs[attr_name].value = attr_value
  File "/opt/vitis_ai/conda/envs/vitis-ai-pytorch/lib/python3.6/site-packages/nndct_shared/nndct_graph/base_operator.py", line 124, in value
    f"The type of attr '{self._name.value}' should be {self._type} instead of {type(value[0])}"
TypeError: The type of attr 'input' should be (<class 'int'>, <class 'float'>, <class 'nndct_shared.nndct_graph.base_tensor.Tensor'>) instead of <class `'numpy.ndarray'>`
```

If anyone knows what might be the issue please let me know

carrotandjzy, 2021-03-17:
This issue is related with constant folding for upsample2d,  it will be solved in the next version. You can refer https://github.com/Xilinx/Vitis-AI/issues/216 to walk around it.

jonassundseth, 2021-03-17:
Thank you! Constant padding did the trick.

closed 2021-03-17 by jonassundseth