Xilinx/Vitis-AI #334: No models available in Model zoo for Alveo U200 under tensorflow 2

daivikbhatia, 2021-03-16:
Hi, 
I want to test tensorflow 2 for Alveo U200 and I want to start with a pre built model. Unfortunately the .xmodel file is missing for alveo U200 under tensorflow 2 in the .ymal file. Can someone please tell when the examples will be ready for alveo U200?

sumitn-xilinx, 2021-03-16:
Check this area: https://github.com/Xilinx/Vitis-AI/tree/master/examples/DPUCADX8G/tensorflow

daivikbhatia, 2021-03-25:
this is for tensorflow 1. I need examples for tensorflow 2

bryanloz-xilinx, 2022-03-10:
@paolodalberto, @woinck ... is it true we don't have tf2 support for u200 / u250 in the model zoo?

For instance: 
https://github.com/Xilinx/Vitis-AI/blob/master/models/AI-Model-Zoo/model-list/tf2_resnet50_imagenet_224_224_7.76G_2.0/model.yaml

I notice this doesn't have an entry for u200 / u250.