Xilinx/Vitis-AI #335: Operators Supported by PyTorch about torch.nn.Linear()

sunshine411, 2021-03-17:
Hi
   I learned from Table27 in <[ug1414-vitis-ai-1.3](https://www.xilinx.com/support/documentation/sw_manuals/vitis_ai/1_3/ug1414-vitis-ai.pdf)> that Pytorch does not support torch.nn.Linear() layer. Is there any plan to support the fully connected layer in the next version?

jimheaton, 2021-03-17:
Fully connected layers are supported in Pytorch. They get converted to the matmul operation under the hood.

sunshine411, 2021-03-20:
Thanks for your answer

      I want know if Is it OK if I define like the following code.
      Is this the best way？

```
class Net(nn.Module):
    def __init__(self):
        super(Net, self).__init__()
        self.conv1 = nn.Conv2d(1, 10, kernel_size=5)
    
        ......
        self.weight = Parameter(torch.Tensor(out_size, in_size))
        self.bias = Parameter(torch.Tensor(out_size))

    def forward(self, x):
        x = F.relu(F.max_pool2d(self.conv1(x), 2))
        ......
        x = torch.add(torch.matmul(x, self.weight.t()), self.bias)
        return F.log_softmax(x, dim=1)
```

wanghong4compiler, 2022-02-19:
Hi @sunshine411 ,
yes, it is ok.

closed 2022-02-25 by qianglin-xlnx