Xilinx/Vitis-AI #336: Too many subgraphs. Teach me HOW to compile an own made model into an xmodel with only one subgraph.

Hikaru-Furuta, 2021-03-18:
I'm trying to compile an own customized xmodel for ZCU104  in a conda env, vitis-ai-pytorch, using Vitis-ai docker.
I know we need a quantized xmodel and arch.json for ZCU104 to generate an xmodel for inference on ZCU104, but they were converted into an xmodel, which has 52 root subgraphs.
I just want them to be only one subgraph.
The methods described on Page 86~88 in the Vitis ai User guide(UG1414 v1.3 December 17, 2020) will solve this?
If there's any other way out to compile the model into an xmodel with only one root subgraph, please teach me.

qianglin-xlnx, 2021-03-26:
Hi @Hikaru-Furuta 
Which pytorch model do you use?

Hikaru-Furuta, 2021-03-26:
@qianglin-xlnx 
thanks for replying
The one I've created by myself.
after extracting Leakyrelu from the network, the number of subgraphs became 4.
I think Leakyrelus cannot be converted by this compiler right?

Fabioni, 2021-07-03:
Hey @Hikaru-Furuta and @qianglin-xlnx,

how did you solve the problem? I have the same problem when compiling my own pytorch model. I have 152 subgraphs. Each one look like this:

```
in:  ["{name: 'Darknet__Darknet_Sequential_LeakyReLU_activation__input_298_fix', shape: [1, 32, 32, 256], type: 'xint8', attrs: {'ddr_addr': 0, 'fix_point': 4, 'location': 1, 'round_mode': 'DPU_ROUND', 'bit_width': 8, 'if_signed': True, 'reg_id': 1}}[1, 32, 32, 256]"]
out:  ["{name: 'Darknet__Darknet_Sequential_Conv2d_Conv2d__input_302_fix', shape: [1, 16, 16, 512], type: 'xint8', attrs: {'ddr_addr': 262144, 'fix_point': 4, 'location': 1, 'round_mode': 'DPU_ROUND', 'bit_width': 8, 'if_signed': True, 'reg_id': 1}}[1, 16, 16, 512]"]
```

I only want to have one subgraph.

Hikaru-Furuta, 2021-07-04:
@Fabioni  
I couldn't solve, but think these problems are caused by unsupported Layers like leaky relu.
In order to make it in one graph, we may need to use only supported Layers(padding, concat, activation functions, etc.)

Fabioni, 2021-07-05:
Hey, I also found the problem today. In Vitis ai user guide, they say LeakyRelu is supported but obviously this was the problem. Exchanging LeakyRelu to Relu solved the problem with the subgraphs. Upsampeling was also a problem but disappeared somehow, I really don't know why :D
Now I have 5 subgraphs and one on the DPU. First I thought thats a problem but actually that is how it should be. One is the input, three are the output and one is the actual subgraph we want.

For the moment I will just go with that (so Relu) but would really be interested in why LeakyRelu doesn't work here although it is mentioned as support in the user guide.

Greetings,
Fabian

Fabioni, 2021-07-05:
Maybe it is just with pytorch and it works if we go the detour ober caffee or tensorflow.

Hikaru-Furuta, 2021-07-05:
@Fabioni 
Yes, exactly. I've seen that leaky relu is mentioned as supported function in the user guide too.
As for the outputs, I've got the same problem as well.
I meant it to be in only one output and wanted to make only one subgraph but it separated into three elements when compiled.
anyway, thank you for the  information!

Fabioni, 2021-07-05:
The output is no problem, my model outputs a tuple of three tensors so that is really what I want. It's just a bit confusing to call this a subgraph.

Hikaru-Furuta, 2021-07-05:
@Fabioni 
and I only have one more question. What DPU do you use for your DNN?

Fabioni, 2021-07-05:
I use the pre build image of the zcu102

BraunPhilipp, 2021-07-14:
I highly recommend looking into the xnnc compiler code. All supported ops are clearly stated. Caffe has the best support in my opinion. Otherwise, you will need to fold your batch norm etc. into a conv layer.

* https://github.com/Xilinx/Vitis-AI/blob/dd383346871339e5ee1ffbdec3bfa665210fcfd2/docs/quick-start/install/Vitis%20AI%201.3.2%20April%202021%20Patch.md

jackliu-bili, 2021-10-19:
Hi,everyone
I have same problem.
I met a AssertionError because "assert len(subgraphs) == 1".
Now, I want to solve the problem but I havn't see a perfect solution.There‘s a possibility that these problems are caused by unsupported Layers. But I don't know which layers are unsupported layers.
So, I pull the image here. Hoping that you can help me.
Thanks very mach.
[image]

levent9402, 2022-02-17:
> 

Hi，I find your model has too many subgraphs, but the demo code only support 1 subgraphs. It is possible that there are unsupported layers in the model.

guohaot-Xlnx, 2022-02-23:
Hi @Hikaru-Furuta ,
This issue is about to close. If you have any questions, please repost to the issues

closed 2022-02-23 by guohaot-Xlnx