Xilinx/Vitis-AI #337: () is missing in parse_log.py

satoru-i, 2021-03-18:
When I use 'Vitis-AI/models/AI-Model-Zoo/caffe-xilinx/tools/extra/parse_log.py', an errors shows,

  File "/workspace/tutorials/caffe-xilinx/tools/extra/parse_log.py", line 166
    print 'Wrote %s' % output_filename

I think () is missing in the sentence.

print ('Wrote %s' % output_filename) is correct.

lji72, 2021-03-23:
Wich version of python do you use?

satoru-i, 2021-03-23:
Thank you for your reply.

I'm using caffe-xilinx in docker (vitis-ai-gpu:latest), so python version is 3.6.12.
I copied the caffe-xilinx directory under the tutorials directory and use it.

lji72, 2021-03-23:
Ok, I see. Actually, The caffe-xilinx you copied can be only used with python 2 and there is a caffe-xilinx of python 3 in docker.

satoru-i, 2021-03-23:
I thought AI-Model-Zoo/caffe-xilinx would work in docker. Thank you.

closed 2021-03-23 by satoru-i