Xilinx/Vitis-AI #339: Kernel panic happen when install vitis-ai-runtime 1.3

Xinjie-Qiu, 2021-03-18:
Hi everyone,
I am following the tutoral of waa-trd project at zcu104 board
and has been success generated SD card files 
```
% export TRD_HOME=< Vitis-AI-path >/dsa/WAA-TRD
% source < vitis-install-directory >/Vitis/2020.2/settings64.sh
% source < path-to-XRT-installation-directory >/setup.sh
% gunzip < mpsoc-common-system >/xilinx-zynqmp-common-v2020.2/rootfs.tar.gz
% export EDGE_COMMON_SW=< mpsoc-common-system >/xilinx-zynqmp-common-v2020.2 
% export SDX_PLATFORM=< zcu102-base-platform-path >/xilinx_zcu102_base_202020_1/xilinx_zcu102_base_202020_1.xpfm
% cd \$TRD_HOME/proj/build/detection-pre_DPUv2
% ./build_detection_pre.sh 
```
then use the balenaEtcher tool to generate the SD card
while installing Vitis AI Runtime on the evaluation board
I following the step copy the centos folder into board
then execute command to install
```
cd ~/centos
bash setup.sh
```
and the result is shown below
```
root@zynqmp-common-2020_2:~/centos# ./setup.sh 
Verifying...                          ################################# [100%]
Preparing...                          ################################# [100%]
%prein(libunilog-1.3.0-r422.aarch64): scriptlet start
%prein(libunilog-1.3.0-r422.aarch64): execv(/bin/sh) pid 1306
%prein(libunilog-1.3.0-r422.aarch64): waitpid(1306) rc 1306 status 0
Updating / installing...
   1:libunilog-1.3.0-r422             ################################# [100%]
%post(libunilog-1.3.0-r422.aarch64): scriptlet start
%post(libunilog-1.3.0-r422.aarch64): execv(/bin/sh) pid 1307
%post(libunilog-1.3.0-r422.aarch64): waitpid(1307) rc 1307 status 0
Verifying...                          ################################# [100%]
Preparing...                          ################################# [100%]
%prein(libxir-1.3.0-r422.aarch64): scriptlet start
%prein(libxir-1.3.0-r422.aarch64): execv(/bin/sh) pid 1309
%prein(libxir-1.3.0-r422.aarch64): waitpid(1309) rc 1309 status 0
Updating / installing...
   1:libxir-1.3.0-r422                
```
the system will crash at this step

after that I watch the log at serial port
and is shown below
```
[  432.209430] Kernel panic - not syncing: Fatal exception in interrupt         
[  432.215766] SMP: stopping secondary CPUs                                     
[  432.219673] Kernel Offset: disabled                                          
[  432.223144] CPU features: 0x0002,20002004                                    
[  432.227136] Memory Limit: none                                               
[  432.230177] ---[ end Kernel panic - not syncing: Fatal exception in interrup-
```
Is there any solution to solve this problem?
Thanks in Advance

qianglin-xlnx, 2021-03-18:
Hi @Xinjie-Qiu 
Installing VART itself will not cause any system crash problem. 
So could you please check the memory space before you install the VART? Just make sure it has enough memory space to install it.

Xinjie-Qiu, 2021-03-18:
I watch the whole memory space is 4GB
and before install is 3935.2 total,   3586.9 free,     77.2 used
while crash I check is 3935.2 total,   3541.6 free,     93.5 used

qianglin-xlnx, 2021-03-18:
What about the remaining space of the SD card?  Try the following command.

root@xilinx-zcu102-2020_2:~# df
Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/root       13573640 5746104   7185664  45% /
devtmpfs         1488284       4   1488280   1% /dev
tmpfs            2013308     148   2013160   1% /run
tmpfs            2013308     128   2013180   1% /var/volatile
/dev/mmcblk0p1   1040083   74249    965834   8% /media/sd-mmcblk0p1

Xinjie-Qiu, 2021-03-19:
I think the remaining space is enough for install 
root@zynqmp-common-2020_2:~# df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root       109G  1.3G  103G   2% /
devtmpfs        1.7G  4.0K  1.7G   1% /dev
tmpfs           2.0G  128K  2.0G   1% /run
tmpfs           2.0G  108K  2.0G   1% /var/volatile
/dev/mmcblk0p1 1016M   56M  961M   6% /media/sd-mmcblk0p1

qianglin-xlnx, 2021-03-19:
Hi @Xinjie-Qiu 
I just noticed that the tutoral of waa-trd project uses ZCU102 board instead of ZCU104 board.  I'm not sure if it can be used for ZCU104 board. At least, you have to use xilinx_zcu104_base_202020_1 instead of xilinx_zcu102_base_202020_1. 
If you have zcu102 board, it's better to use it for test.

tianfang-fafafa, 2022-03-14:
hi @sumitn-xilinx , could you pay your attention please?  this issue is related to WAA-TRD.

sumitn-xilinx, 2022-03-14:
Trying to reproduce. Will revert back.

sumitn-xilinx, 2022-03-14:
It seems like a 1 year old issue. Can @Xinjie-Qiu try with the latest version?
https://github.com/Xilinx/Vitis-AI/tree/master/Whole-App-Acceleration/apps/resnet50/build_flow/DPUCZDX8G_zcu102

BTW, these steps are for ZCU102. For running this TRD on ZCU104, you will have to create ZCU104 DPU XO by following the steps in dsa/DPU-TRD folder and do small modifications in the run.sh and Makefile to use that DPU XO in WAA-TRD.

haipnh, 2022-04-08:
Hi @sumitn-xilinx , I am trying to make dsa/WAA-TRD/proj/build/detection-pre_DPUv2 running on my ZCU104.

My environment:

- Host: Ubuntu 18.04.4 / Vitis 2020.2 / Vitis-AI 1.4.1
- Target: ZCU104 / xilinx_zcu104_base_202020_1.xpfm / xilinx-zynqmp-common-v2020.2

I followed the instruction and make the changes as following:

``` shell
diff --git a/dsa/WAA-TRD/proj/build/detection-pre_DPUv2/build_detection_pre.sh b/dsa/WAA-TRD/proj/build/detection-pre_DPUv2/build_detection_pre.sh
index e3652b946..6eb9ea60f 100755
--- a/dsa/WAA-TRD/proj/build/detection-pre_DPUv2/build_detection_pre.sh
+++ b/dsa/WAA-TRD/proj/build/detection-pre_DPUv2/build_detection_pre.sh
@@ -1,9 +1,12 @@
+export EDGE_COMMON_SW=/opt/xilinx/images/xilinx-zynqmp-common-v2020.2
+export SDX_PLATFORM=/opt/xilinx/platforms/xilinx_zcu104_base_202020_1/xilinx_zcu104_base_202020_1.xpfm
+
 #Generate Pre-processor pipeline xo and so file
 export CUR_DIR=$PWD
 cd $CUR_DIR/../../../accel/detection-pre
-make cleanall
+#make cleanall
 make xo TARGET=hw BOARD=Zynq ARCH=aarch64
 
 #Generate SD card files
 cd $CUR_DIR
-make KERNEL=DPU_PP PP_ACCEL=../../../accel/detection-pre DPU_IP=../../../../DPU-TRD/dpu_ip/ DEVICE=zcu102
+make KERNEL=DPU_PP PP_ACCEL=../../../accel/detection-pre DPU_IP=../../../../DPU-TRD/dpu_ip/ DEVICE=zcu104
diff --git a/dsa/WAA-TRD/proj/build/detection-pre_DPUv2/config_file/prj_config b/dsa/WAA-TRD/proj/build/detection-pre_DPUv2/config_file/prj_config
index 9083fd5da..29700e452 100755
--- a/dsa/WAA-TRD/proj/build/detection-pre_DPUv2/config_file/prj_config
+++ b/dsa/WAA-TRD/proj/build/detection-pre_DPUv2/config_file/prj_config
@@ -19,20 +19,20 @@
 
 freqHz=300000000:DPUCZDX8G_1.aclk
 freqHz=600000000:DPUCZDX8G_1.ap_clk_2
-freqHz=300000000:DPUCZDX8G_2.aclk
-freqHz=600000000:DPUCZDX8G_2.ap_clk_2
+#freqHz=300000000:DPUCZDX8G_2.aclk
+#freqHz=600000000:DPUCZDX8G_2.ap_clk_2
 
 [connectivity]
 
 sp=DPUCZDX8G_1.M_AXI_GP0:HPC0
 sp=DPUCZDX8G_1.M_AXI_HP0:HP0
 sp=DPUCZDX8G_1.M_AXI_HP2:HP1
-sp=DPUCZDX8G_2.M_AXI_GP0:HPC0
-sp=DPUCZDX8G_2.M_AXI_HP0:HP2
-sp=DPUCZDX8G_2.M_AXI_HP2:HP3
+#sp=DPUCZDX8G_2.M_AXI_GP0:HPC0
+#sp=DPUCZDX8G_2.M_AXI_HP0:HP2
+#sp=DPUCZDX8G_2.M_AXI_HP2:HP3
 
 
-nk=DPUCZDX8G:2
+nk=DPUCZDX8G:1
 
 [advanced]
 misc=:solution_name=link
@@ -41,6 +41,6 @@ misc=:solution_name=link
 #param=compiler.skipTimingCheckAndFrequencyScaling=1
 
 [vivado]
-prop=run.impl_1.strategy=Performance_Explore
+prop=run.impl_1.strategy=Performance_ExploreWithRemap
 #param=place.runPartPlacer=0
```

After flashing the image onto a SD card and installing required packages, I ran the example by running commands:
```
root@zynqmp-common-2020_2:/media/sd-mmcblk0p1/adas_detection# ./adas_detection model_zcu104/yolov3_adas_pruned_0_9/yolov3_adas_pruned_0_9.xmodel 1
```

The Serial terminal dumped these information and system hanged:
```
root@zynqmp-common-2020_2:~# [ 4995.854297] Internal error: Oops: 96000144 [#1] SMP
[ 4995.859166] Modules linked in: zocl(O) dmaproxy(O) mali(O) [last unloaded: zocl]
[ 4995.866556] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G           O      5.4.0-xilinx-v2020.2 #1
[ 4995.875067] Hardware name: ZynqMP ZCU104 RevC (DT)
[ 4995.879843] pstate: 40000005 (nZcv daif -PAN -UAO)
[ 4995.884622] pc : __dma_inv_area+0x40/0x58
[ 4995.888618] lr : arch_sync_dma_for_cpu+0x24/0x30
[ 4995.893216] sp : ffff800010003d10
[ 4995.896515] x29: ffff800010003d10 x28: ffff000878f34c00 
[ 4995.901810] x27: 0000000080c0c03c x26: ffff800011469508 
[ 4995.907104] x25: ffff000878ca0800 x24: ffff000878ca07c0 
[ 4995.912399] x23: 0000000000000002 x22: 0000000000000002 
[ 4995.917694] x21: 0000000000000600 x20: 5f74666965dabcc0 
[ 4995.922989] x19: ffff00087ab2f410 x18: 000000000000000e 
[ 4995.928283] x17: 0000000000000001 x16: 0000000000000019 
[ 4995.933578] x15: 0000000000000033 x14: 000000000000004c 
[ 4995.938873] x13: 0000437fd3591cd0 x12: 0000000000000000 
[ 4995.944168] x11: 0000000000000003 x10: 0000000000000000 
[ 4995.949463] x9 : 0000000000000000 x8 : ffff00087f781200 
[ 4995.954757] x7 : 0000000000000000 x6 : ffff000865dabcc0 
[ 4995.960052] x5 : 000000000000003c x4 : 5f74666965dabcc0 
[ 4995.965347] x3 : 000000000000003f x2 : 0000000000000040 
[ 4995.970642] x1 : 5f73666965dac2c0 x0 : 5f73666965dabcc0 
[ 4995.975937] Call trace:
[ 4995.978370]  __dma_inv_area+0x40/0x58
[ 4995.982016]  dma_direct_sync_single_for_cpu+0x7c/0x80
[ 4995.987057]  dma_direct_unmap_page+0x84/0x88
[ 4995.991313]  gem_rx+0xf8/0x2b8
[ 4995.994357]  macb_poll+0x58/0x100
[ 4995.997658]  net_rx_action+0x114/0x340
[ 4996.001397]  __do_softirq+0x118/0x22c
[ 4996.005043]  irq_exit+0x98/0xc0
[ 4996.008169]  __handle_domain_irq+0x64/0xb8
[ 4996.012246]  gic_handle_irq+0x5c/0xb8
[ 4996.015893]  el1_irq+0xb8/0x140
[ 4996.019018]  arch_cpu_idle+0x10/0x18
[ 4996.022577]  do_idle+0x200/0x280
[ 4996.025788]  cpu_startup_entry+0x20/0x28
[ 4996.029695]  rest_init+0xb0/0xbc
[ 4996.032906]  arch_call_rest_init+0xc/0x14
[ 4996.036898]  start_kernel+0x400/0x42c
[ 4996.040547] Code: 8a230000 54000060 d50b7e20 14000002 (d5087620) 
[ 4996.046634] ---[ end trace b57b578f9e38743e ]---
[ 4996.051237] Kernel panic - not syncing: Fatal exception in interrupt
[ 4996.057574] SMP: stopping secondary CPUs
[ 4996.061480] Kernel Offset: disabled
[ 4996.064951] CPU features: 0x0002,20002004
[ 4996.068943] Memory Limit: none
[ 4996.071984] ---[ end Kernel panic - not syncing: Fatal exception in interrupt ]---
```
I have no idea with this situation. Could you please help?

qianglin-xlnx, 2022-09-09:
Closing since no activity for more than 5 months, please reopen if you still have question, thanks