Xilinx/Vitis-AI #340: pytorch quantization export_xmodel report errror?

evilidol, 2021-03-18:
I am trying to quantize a custom model using torch_quantizer, but it report error.
during export xmodel  the error occurs.
[image]
I have no idea how to solve it?
thanks

evilidol, 2021-03-19:
I have no idea about this error,  completely no！！
why there is no one answer this question
any clue is helpful, thanks.

carrotandjzy, 2021-03-22:
Can you show me your vitis-ai version ?

evilidol, 2021-03-22:
> Can you show me your vitis-ai version ?

I don't know how to check my vitis-ai version, but probably the version is V1.3

carrotandjzy, 2021-03-22:
You can update to v1.3.1. May it can solve your problem.

niuxjxlnx, 2022-02-21:
Hi, @evilidol:
Suppose the issue has been solved in vitis-ai 1.3.1 and later versions. Did you have a try on your model?

qianglin-xlnx, 2022-08-24:
Hi @evilidol 
Closing since no activity for more than one year, please reopen if you still have question, thanks

closed 2022-08-24 by qianglin-xlnx