Xilinx/Vitis-AI #341: A library for running model on CPU "libvart-cpu-runner.so" cannot found.

lllkiller-jlll, 2021-03-18:
Hi.

I'm trying to using Alveo U50 by Vitis AI.

My neural network model is divided four dpu-subgraph and the others are for cpu runner.
I created runner object for cpu using VART API like below:
`auto runner_cpu = vart::Runner::create_runner(subgraph_cpu[0], "ref");`
But VART API said that:
`F0318 07:46:37.057397  3576 dpu_runner.cpp:158] [UNILOG][FATAL][VART_RUNNER_CONSTRUCTION_FAIL][Cannot create runner] cannot open library! lib=libvart-cpu-runner.so, error=libvart-cpu-runner.so: cannot open shared object file: No such file or directory`

I could not find "libvart-cpu-runner.so". I only could find "libvart-dpu-runner.so" and "libvart-runner.so" in /usr/lib path.
How can I solve this problem?

qianglin-xlnx, 2021-03-22:
Hi @lllkiller-jlll 
cpu-runner is not released in VAI1.3. Currently, only dpu-runner is supported.
You need to implement the CPU op yourself. You can refer to this https://github.com/Xilinx/Vitis-AI/issues/320

lllkiller-jlll, 2021-03-24:
Hi @qianglin-xlnx .

I understand that cpu-runner is not supported currently.
I'll close this issue.

closed 2021-03-24 by lllkiller-jlll