Xilinx/Vitis-AI #342: "CRITICAL WARNING: [IP_Flow 19-5097] Unable to determine VLNV from IP file" is fired while trying to run the synthesis step

ahmedabdelgaber, 2021-03-18:
Hello Dears,

I am getting such messages while trying to synthesis after adding the DPU module:
WARNING: [IP_Flow 19-395] Problem validating against XML schema: see 'xilinx:componentInstanceExtensions' : Bad end of element
CRITICAL WARNING: [IP_Flow 19-5097] Unable to determine VLNV from IP file; verify it has the correct syntax: d:/Product/repos/ahs_15_dpu/ahs/HW/fpga/example_design/example_design.srcs/sources_1/bd/Base_Zynq_MPSoC/ip/Base_Zynq_MPSoC_DPUCZDX8G_0_0/xci/sfm/fp_exp/fp_exp.xci
CRITICAL WARNING: [IP_Flow 19-3577] Failed to recreate IP instance 'Base_Zynq_MPSoC_DPUCZDX8G_0_0'. Error during subcore creation.

Would you please help me solving this issue?
Thanks,
Ahmed

davidxudong, 2021-03-22:
Hi @ahmedabdelgaber 

Could you tell me the detail about how to reproduce this?
Which flow do you use? Vivado flow or Vitis flow?

ahmedabdelgaber, 2021-03-22:
Hello David,

I think it's same issue reported in this link:
https://forums.xilinx.com/t5/AI-and-Vitis-AI/DPU%E7%9A%84IP%E6%A0%B8/td-p/1189510

It's a compatibility issue and fixed once I used v1.2.1

I think I can close this issue.

Thanks and regards,
Ahmed

ahmedabdelgaber, 2021-03-22:
Replaying for your question, I am using the vivado flow.

closed 2021-03-22 by ahmedabdelgaber